leoheck/kiri #4: Plotgitsch is not working fine for nested projects

leoheck, 2021-05-17:
When projects have the same schematic name, gitplottsch does not work fine.

jorlee99, 2021-05-17:
It seems to work fine for both tests
[image]
[image]
I'll show you my file structure
https://gitlab.com/jordan.lee1202/nested_test

leoheck, 2021-05-17:
Thanks.
Plotgitsch has this issue where if it does not find a part, it gives an exception and do not generate the image. I don't know how to solve this. Ocaml is so hard to change something.

leoheck, 2021-05-18:
Thanks. Then I will close this for now.

closed 2021-05-18 by leoheck