leoheck/kiri #5: Branching doesn't track PCB Changes

jorlee99, 2021-05-18:
It seems if I make a branch and make changes it only track changes on the schematic. The PCB doesn't produce an SVG it seems, and there is no output on the actual PCB view. 
[image]

This is the output spit from the verbose
```
    script_dir: /home/jordan/kicad/kdiff

     repo_root: /home/jordan/kicad/high-level-feedback-board
   project_dir: src

     kicad_pcb: src/HLFB_Breakout.kicad_pcb
     sch_files: src/HLFB.sch src/HLFB_Breakout.sch src/IO_Ports.sch
    board_name: HLFB_Breakout

    output_dir: /home/jordan/kicad/high-level-feedback-board/kidiff/src
       web_dir: /home/jordan/kicad/high-level-feedback-board/kidiff/src/web/

git-kdiff-log Jordan_Revision -- src/HLFB_Breakout.kicad_pcb src/HLFB.sch src/HLFB_Breakout.sch src/IO_Ports.sch

Commits list:
3c7c699 | 2021-05-18 13:04:37 -0700 | Jordan Lee | Moved 5V trace down
03a3ddb | 2021-05-18 11:35:56 -0700 | Jordan Lee | Added source files as well as changed README to reflect PCB

Generating artifacts for 2 commits.
Totaling 2 possible comparisons.

Schematic Pages
     1  HLFB_Breakout

Layers
     1  F_Cu
     2  B_Cu
     3  B_Adhes
     4  F_Adhes
     5  B_Paste
     6  F_Paste
     7  B_SilkS
     8  F_SilkS
     9  B_Mask
    10  F_Mask
    11  Dwgs_User
    12  Cmts_User
    13  Eco1_User
    14  Eco2_User
    15  Edge_Cuts
    16  Margin
    17  B_CrtYd
    18  F_CrtYd
    19  B_Fab
    20  F_Fab

Starting webserver at http://127.0.0.1:9092
```
Any idea?

jorlee99, 2021-05-18:
Nevermind it is working fine again

leoheck, 2021-05-19:
Hmm, no ideia yet. I will have to check.
But try this:
1. git pull on kdiff repo to get latest changes
2. git pull on plotgitsh repo to get latest changes
3. reinstall plotgitsch with "./install"

leoheck, 2021-05-19:
Can you share this high-level-feedback-board repo so I can test this it too?

jorlee99, 2021-05-19:
It seems to be working fine again on a separate branch. I believe it comes from this command I run in order for changes not to be recorded for time stamps. It seems to be working fine for now, I'll reopen this if I encounter more problems. The next thing I will test is what happens after merging branches.

closed 2021-05-19 by jorlee99