leoheck/kiri #6: Not working on windows

jorlee99, 2021-05-21:
Hey @leoheck 

I had a colleague install kdiff and it wasn't working for him. So I tried wit ha fresh install for myself and it is no longer working for me. Have you changed any of the installation repos by chance? The schematics work just fine but as soon as I look at a PCB it is blank and this is the error I got.

```
   plotpcb -o "/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/" -q "/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/expansion.kicad_pcb"
/mnt/c/Users/Stanley/Desktop/ELEC/KiCad-Diff/plotpcb: line 2: $'\r': command not found
/mnt/c/Users/Stanley/Desktop/ELEC/KiCad-Diff/plotpcb: line 5: $'\r': command not found
/mnt/c/Users/Stanley/Desktop/ELEC/KiCad-Diff/plotpcb: line 6: syntax error near unexpected token `$'\r''
'mnt/c/Users/Stanley/Desktop/ELEC/KiCad-Diff/plotpcb: line 6: `readlink_osx()
   plotpcb -o "/home/jordan/lcd-conn-expansion/kidiff/src/d9d214e/" -q "/home/jordan/lcd-conn-expansion/kidiff/src/d9d214e/expansion.kicad_pcb"
/mnt/c/Users/Stanley/Desktop/ELEC/KiCad-Diff/plotpcb: line 2: $'\r': command not found
/mnt/c/Users/Stanley/Desktop/ELEC/KiCad-Diff/plotpcb: line 5: $'\r': command not found
/mnt/c/Users/Stanley/Desktop/ELEC/KiCad-Diff/plotpcb: line 6: syntax error near unexpected token `$'\r''
'mnt/c/Users/Stanley/Desktop/ELEC/KiCad-Diff/plotpcb: line 6: `readlink_osx()
```

jorlee99, 2021-05-21:
I also tried pulling older commits as well and it does not seem to work.

leoheck, 2021-05-21:
Is it possible to share the project so I can test it here?
Also, check if Kicad-Diff is the latest version since this is related with the pcb

jorlee99, 2021-05-21:
Did you want to kicad project files?

leoheck, 2021-05-21:
Since msgs are talking about \r character. Try to check if the .kicad_pcb is was created on windows with windows endings character. You could also use dos2unix on these files to make sure they have the Linux line endings. If something like this fixes the issue I have to think how to improve kdiff to be resilient to this.

jorlee99, 2021-05-21:
The .kicad_pcb file was created in windows. It has worked before and was working earlier this morning it just seems however after reinstalling from a fresh install it no longer works.  I'll give your dos2unix thing a try. I will also try cloning it on a Linux operating system to see if it is giving me a problem. I have sent the .zip folder of the kicad files as well. 
[src.zip](https://github.com/leoheck/kdiff/files/6525400/src.zip)

jorlee99, 2021-05-21:
After doing dos2unix the schematic no longer shows up either.

leoheck, 2021-05-21:
Hm, let me check here if I can see the same on windows

leoheck, 2021-05-21:
@jorlee99, I changed the command like to receive the project file `.pro` instead of `.kicad_pcb`
 Please, check if this fixes your issue.

jorlee99, 2021-05-22:
I just tried to command and no luck. I am no longer getting a schematic or a pcb

leoheck, 2021-05-22:
Did you test it on other projects? I tested it on windows it is working too.

leoheck, 2021-05-22:
If you shat your project I can check and fix kdiff if this is the case.

leoheck, 2021-05-22:
Hey, try this branch https://github.com/leoheck/kdiff/tree/windowns-line-endings

I would suggest always use linux line endings when working with git. 
You can check this configuration when installing git for windows.
Git for Windows can convert these file formats while pulling/pushing code.

jorlee99, 2021-05-25:
Sorry I have been hiatus for a couple of days

> Did you test it on other projects? I tested it on windows it is working too.
> […](#)

So I have and it doesn't work on any projects.

> If you shat your project I can check and fix kdiff if this is the case.
> […](#)

Did you want the repo itself? I have linked the files above for you to try

> The .kicad_pcb file was created in windows. It has worked before and was working earlier this morning it just seems however after reinstalling from a fresh install it no longer works. I'll give your [...]

> Hey, try this branch https://github.com/leoheck/kdiff/tree/windowns-line-endings
> 
> I would suggest always use linux line endings when working with git.
> You can check this configuration when installing git for windows.
> Git for Windows can convert these file formats while pulling/pushing code.

I'll give this a try too but let me explain the issue. Firstly here are some new error codes that might shed some light 

```
du: cannot access '/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/high-level-feedback-board/kidiff/src/03a3ddb/HLFB_Breakout.kicad_pcb': No such file or directory
   plotpcb -o "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/high-level-feedback-board/kidiff/src/03a3ddb/" -q "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/high-level-feedback-board/kidiff/src/03a3ddb/HLFB_Breakout.kicad_pcb"
Traceback (most recent call last):
  File "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/kicaddiff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/kicaddiff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/high-level-feedback-board/kidiff/src/03a3ddb/HLFB_Breakout.kicad_pcb" for reading
du: cannot access '/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/high-level-feedback-board/kidiff/src/3c7c699/HLFB_Breakout.kicad_pcb': No such file or directory
   plotpcb -o "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/high-level-feedback-board/kidiff/src/3c7c699/" -q "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/high-level-feedback-board/kidiff/src/3c7c699/HLFB_Breakout.kicad_pcb"
Traceback (most recent call last):
  File "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/kicaddiff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/kicaddiff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/high-level-feedback-board/kidiff/src/3c7c699/HLFB_Breakout.kicad_pcb" for reading
   2.         - 2d5744d
du: cannot access '/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/high-level-feedback-board/kidiff/src/03a3ddb/HLFB_Breakout.kicad_pcb': No such file or directory
   plotpcb -o "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/high-level-feedback-board/kidiff/src/03a3ddb/" -q "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/high-level-feedback-board/kidiff/src/03a3ddb/HLFB_Breakout.kicad_pcb"
Traceback (most recent call last):
  File "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/kicaddiff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/kicaddiff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/high-level-feedback-board/kidiff/src/03a3ddb/HLFB_Breakout.kicad_pcb" for reading
du: cannot access '/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/high-level-feedback-board/kidiff/src/2d5744d/HLFB_Breakout.kicad_pcb': No such file or directory
   plotpcb -o "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/high-level-feedback-board/kidiff/src/2d5744d/" -q "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/high-level-feedback-board/kidiff/src/2d5744d/HLFB_Breakout.kicad_pcb"
Traceback (most recent call last):
  File "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/kicaddiff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/kicaddiff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/mnt/c/Users/Stanley/Desktop/LinuxStuff/kdiffFull/high-level-feedback-board/kidiff/src/2d5744d/HLFB_Breakout.kicad_pcb" for reading
```

So it seems like a running `./install.sh` from the newest clone does not seem to work. It locks until I personally cancel it. 

```
[WARNING] Running as root is not recommended
[WARNING] Running as root is not recommended
[ERROR] No package pinned to this target found, or invalid package name: .
[WARNING] Running as root is not recommended
[ERROR] No package pinned to this target found, or invalid package name: .
[WARNING] Running as root is not recommended
[... cut ...]
So I ran on older clone of plotkicadsch and it seemingly runs fine. These [are the files I use](https://github.com/jorlee99/kdiffFull). So looking at it the schematic shows up fine meaning that pltokicadsch works fine I assume. But it is only the schematic that has a problem and this has only happened on recent builds. I have a copy that has it working from older builds and I can give you it's exported .tar as well. But even importing the tar it doesn't work due to how wsl is laid out.

I apologize as I realize the info I have provided maybe useless but maybe we can figure this out?

jorlee99, 2021-05-25:
With the new branch I still have no luck. These are the following errors 
```
du: cannot access '/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/expansion.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/" -q "/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/expansion.kicad_pcb"
Traceback (most recent call last):
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/expansion.kicad_pcb" for reading
du: cannot access '/home/jordan/lcd-conn-expansion/kidiff/src/d9d214e/expansion.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/lcd-conn-expansion/kidiff/src/d9d214e/" -q "/home/jordan/lcd-conn-expansion/kidiff/src/d9d214e/expansion.kicad_pcb"
Traceback (most recent call last):
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/home/jordan/lcd-conn-expansion/kidiff/src/d9d214e/expansion.kicad_pcb" for reading
   2.         - f9af8cd
du: cannot access '/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/expansion.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/" -q "/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/expansion.kicad_pcb"
Traceback (most recent call last):
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/expansion.kicad_pcb" for reading
du: cannot access '/home/jordan/lcd-conn-expansion/kidiff/src/f9af8cd/expansion.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/lcd-conn-expansion/kidiff/src/f9af8cd/" -q "/home/jordan/lcd-conn-expansion/kidiff/src/f9af8cd/expansion.kicad_pcb"
Traceback (most recent call last):
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/home/jordan/lcd-conn-expansion/kidiff/src/f9af8cd/expansion.kicad_pcb" for reading
Starting webserver at http://127.0.0.1:8000
```

I tried this with a different repo as well

leoheck, 2021-05-25:
[image]
What is the command line you are using? In other words, how you are running kdiff?

jorlee99, 2021-05-25:
> [image]
> What is the command line you are using? In other words, how you are running kdiff?

Oh sorry I forgot about this I just ran it with .pro

This is the output 
```
 kdiff src/expansion.pro
dos2unix: converting file src/expansion.kicad_pcb to Unix format...

Generating artifacts for 3 commits.
Totaling 6 possible comparisons.

   1. 2d62ea3 - d9d214e
du: cannot access '/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/expansion.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/" -q "/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/expansion.kicad_pcb"
Traceback (most recent call last):
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/expansion.kicad_pcb" for reading
du: cannot access '/home/jordan/lcd-conn-expansion/kidiff/src/d9d214e/expansion.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/lcd-conn-expansion/kidiff/src/d9d214e/" -q "/home/jordan/lcd-conn-expansion/kidiff/src/d9d214e/expansion.kicad_pcb"
Traceback (most recent call last):
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/home/jordan/lcd-conn-expansion/kidiff/src/d9d214e/expansion.kicad_pcb" for reading
   2.         - f9af8cd
du: cannot access '/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/expansion.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/" -q "/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/expansion.kicad_pcb"
Traceback (most recent call last):
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/home/jordan/lcd-conn-expansion/kidiff/src/2d62ea3/expansion.kicad_pcb" for reading
du: cannot access '/home/jordan/lcd-conn-expansion/kidiff/src/f9af8cd/expansion.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/lcd-conn-expansion/kidiff/src/f9af8cd/" -q "/home/jordan/lcd-conn-expansion/kidiff/src/f9af8cd/expansion.kicad_pcb"
Traceback (most recent call last):
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/home/jordan/lcd-conn-expansion/kidiff/src/f9af8cd/expansion.kicad_pcb" for reading
Starting webserver at http://127.0.0.1:8000
```

leoheck, 2021-05-25:
The website is still being displayed?

jorlee99, 2021-05-25:
> The website is still being displayed?

Yes 
[image]

leoheck, 2021-05-25:
Do you see any board there?

jorlee99, 2021-05-25:
No I do not

> Do you see any board there?

leoheck, 2021-05-25:
I think you have multiples issues.

The `CRLF` endlines are a thing. 
Check this to try to checkout converting `CRLF` to `LF`.
https://stackoverflow.com/questions/1967370/git-replacing-lf-with-crlf

The second thing you are seeing is that .kicad_pcb boards are missing in your history, I think.
Then a second command is failing, but that is ok. What is not ok is that you are seeing its output. 

Upgrade (git pull) `kdiff` one more time.
Also, try to reinstall my `plotgitsch`. I am not sure you were able to do it.
Inside `pltogitsh` folder do this
```
git pull
opam update
opam upgrade
./install 
```

If `./install` doesn't work, try this instead.
```
make
make install 
```

jorlee99, 2021-05-25:
ok I have done the above and still no luck. I have tried it with a new [repo](https://github.com/jorlee99/newTest) if you want to try yourself. 

I am getting slightly newer error codes 
```
jordan@DESKTOP-BJDDG29:~/newTest$ kdiff newTest/newTest.pro
dos2unix: converting file newTest/newTest.kicad_pcb to Unix format...

Generating artifacts for 2 commits.
Totaling 2 possible comparisons.

grep: .//newTest.sch: No such file or directory
grep: .//newTest.sch: No such file or directory
   1. 9646e03 - 2060795
du: cannot access '/home/jordan/newTest/kidiff/newTest/9646e03/newTest.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/newTest/kidiff/newTest/9646e03/" -q "/home/jordan/newTest/kidiff/newTest/9646e03/newTest.kicad_pcb"
Traceback (most recent call last):
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/home/jordan/newTest/kidiff/newTest/9646e03/newTest.kicad_pcb" for reading
du: cannot access '/home/jordan/newTest/kidiff/newTest/2060795/newTest.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/newTest/kidiff/newTest/2060795/" -q "/home/jordan/newTest/kidiff/newTest/2060795/newTest.kicad_pcb"
Traceback (most recent call last):
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/home/jordan/kdiffFull/kicaddiff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/home/jordan/newTest/kidiff/newTest/2060795/newTest.kicad_pcb" for reading
grep: .//newTest.sch: No such file or directory
grep: .//newTest.sch: No such file or directory
Starting webserver at http://127.0.0.1:8000
^C** Trapped CTRL-C
```

It seems it returns something called a grep? I am not sure

leoheck, 2021-05-25:
Where is this repo, share it with me then I will run it on my Windows VM

You are not using the master/main branch
```
dos2unix: converting file newTest/newTest.kicad_pcb to Unix format...
```

Do this
[image]

jorlee99, 2021-05-25:
> Where is this repo, share it with me then I will run it on my Windows VM
> 
I have linked it here but here is the link again https://github.com/jorlee99/newTest

> You are not using the master/main branch
> 
> ```
> dos2unix: converting file newTest/newTest.kicad_pcb to Unix format...
> ```
> 
> Do this
> [image]
My apologies this is what happens after pulling the main branch from kdiff
``` 
kdiff  newTest/newTest.pro

Generating artifacts for 2 commits.
Totaling 2 possible comparisons.

grep: .//newTest.sch: No such file or directory
grep: .//newTest.sch: No such file or directory
   1. 9646e03 - 2060795
du: cannot access '/home/jordan/newTest/kidiff/newTest/9646e03/newTest.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/newTest/kidiff/newTest/9646e03/" -q "/home/jordan/newTest/kidiff/newTest/9646e03/newTest.kicad_pcb"  &> /dev/null
du: cannot access '/home/jordan/newTest/kidiff/newTest/2060795/newTest.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/newTest/kidiff/newTest/2060795/" -q "/home/jordan/newTest/kidiff/newTest/2060795/newTest.kicad_pcb"  &> /dev/null
ln: failed to create symbolic link '/home/jordan/newTest/kidiff/newTest/9646e03/newTest-/60AD741E'$'\r''.svg': No such file or directory
ln: failed to create symbolic link '/home/jordan/newTest/kidiff/newTest/2060795/newTest-/60AD741E'$'\r''.svg': No such file or directory
ln: failed to create symbolic link '/home/jordan/newTest/kidiff/newTest/9646e03/newTest-${KISYS3DMOD}/Resistor_SMD_3dshapes/R_0603_1608Metric_wrl'$'\r''.svg': No such file or directory
ln: failed to create symbolic link '/home/jordan/newTest/kidiff/newTest/2060795/newTest-${KISYS3DMOD}/Resistor_SMD_3dshapes/R_0603_1608Metric_wrl'$'\r''.svg': No such file or directory
ln: failed to create symbolic link '/home/jordan/newTest/kidiff/newTest/9646e03/newTest-/60AD7907'$'\r''.svg': No such file or directory
ln: failed to create symbolic link '/home/jordan/newTest/kidiff/newTest/2060795/newTest-/60AD7907'$'\r''.svg': No such file or directory
ln: failed to create symbolic link '/home/jordan/newTest/kidiff/newTest/9646e03/newTest-${KISYS3DMOD}/Resistor_SMD_3dshapes/R_0603_1608Metric_wrl'$'\r''.svg': No such file or directory
ln: failed to create symbolic link '/home/jordan/newTest/kidiff/newTest/2060795/newTest-${KISYS3DMOD}/Resistor_SMD_3dshapes/R_0603_1608Metric_wrl'$'\r''.svg': No such file or directory
    Assembling HTML
    Insert commits list
    Insert pages list
grep: .//newTest.sch: No such file or directory
grep: .//newTest.sch: No such file or directory
    - Updating html
    Update board title
    Update board revision
    Update board date
    Update board name
    Update initial schematic images
    Update initial layout images
    Update initial commit legend
    Insert layers list
    - Updating html
Starting webserver at http://127.0.0.1:8000
```

leoheck, 2021-05-25:
Oh, I see. Forgot to say to install `dos2unix`.
Do it with `sudo apt install dos2unix`
Then go back to the  kdiff  repo and change to the `windowns-line-endings` branch to test it.
It should run ok, as it ran here. If it is ok, let me know then I will update the main branch

jorlee99, 2021-05-25:
Unfortunately still no luck I am encountering the exact same errors 
```
kdiff newTest/newTest.pro
dos2unix: converting file newTest/newTest.kicad_pcb to Unix format...

Generating artifacts for 2 commits.
Totaling 2 possible comparisons.

grep: .//newTest.sch: No such file or directory
grep: .//newTest.sch: No such file or directory
   1. 9646e03 - 2060795
du: cannot access '/home/jordan/newTest/kidiff/newTest/9646e03/newTest.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/newTest/kidiff/newTest/9646e03/" -q "/home/jordan/newTest/kidiff/newTest/9646e03/newTest.kicad_pcb"
Traceback (most recent call last):
  File "/home/jordan/KiCad-Diff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/home/jordan/KiCad-Diff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/home/jordan/newTest/kidiff/newTest/9646e03/newTest.kicad_pcb" for reading
du: cannot access '/home/jordan/newTest/kidiff/newTest/2060795/newTest.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/newTest/kidiff/newTest/2060795/" -q "/home/jordan/newTest/kidiff/newTest/2060795/newTest.kicad_pcb"
Traceback (most recent call last):
  File "/home/jordan/KiCad-Diff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/home/jordan/KiCad-Diff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/home/jordan/newTest/kidiff/newTest/2060795/newTest.kicad_pcb" for reading
grep: .//newTest.sch: No such file or directory
grep: .//newTest.sch: No such file or directory
Starting webserver at http://127.0.0.1:8000```

leoheck, 2021-05-25:
My VM is really slow. I pushed one update if you wish to test before of me.
It has this workaround for CRLF in the main branch.

jorlee99, 2021-05-26:
> Try running inside the folder to check if this is the problem.

I have tried that as well. No luck.

> My VM is really slow. I pushed one update if you wish to test before of me.
> It has this workaround for CRLF in the main branch.

It seems the schematics are working now for some of my repos but the pcb's still aren't. However, with the repo I posted it did not work with schematics. I will test with some other repos but I was wondering if you had time to hop into a zoom call or any other sort of communication method to try and figure this out. I am free usually from 9am-5pm PST.

jorlee99, 2021-05-26:
> I have tried that as well. No luck.
>
>
> It seems the schematics are working now for some of my repos but the pcb's still aren't. However, with the repo I posted it did not work with schematics. I will [...]

It seems that the repo itself I linked does not have its schematics working so it seems like kdiff works with schematics which is a great sign. However, the KiCad-Diff does not seem to working as the pcb's still don't show up I will try with and older build of kicad-diff

leoheck, 2021-05-27:
Try to convert the whole repository to unix like line endings. And give it a try.

Plotgitsch cannot handle mixed or or windows line endings. And I don't know how to workaround it from my side this time since it access git history inside of it.

leoheck, 2021-05-28:
Good news. Jnavila pushed a fix for this on plotgitsch, I will try to integrate these changes later, today.

jorlee99, 2021-05-28:
That's great! I'll try it on a fresh install once it is pushed.

jorlee99, 2021-05-28:
So it seems I have maybe fixed the line ending issues. But now it is more the KiCad-Diff may not be working. It just can't access or find the files.
```
Generating artifacts for 2 commits.
Totaling 2 possible comparisons.

   1. 6fb3be5 - d144682
du: cannot access '/home/jordan/TestingKdiff/kidiff/src/6fb3be5/expansion.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/TestingKdiff/kidiff/src/6fb3be5/" -q "/home/jordan/TestingKdiff/kidiff/src/6fb3be5/expansion.kicad_pcb"
Traceback (most recent call last):
  File "/home/jordan/PantheonKdiff/KiCad-Diff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/home/jordan/PantheonKdiff/KiCad-Diff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/home/jordan/TestingKdiff/kidiff/src/6fb3be5/expansion.kicad_pcb" for reading
du: cannot access '/home/jordan/TestingKdiff/kidiff/src/d144682/expansion.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/TestingKdiff/kidiff/src/d144682/" -q "/home/jordan/TestingKdiff/kidiff/src/d144682/expansion.kicad_pcb"
Traceback (most recent call last):
  File "/home/jordan/PantheonKdiff/KiCad-Diff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/home/jordan/PantheonKdiff/KiCad-Diff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/home/jordan/TestingKdiff/kidiff/src/d144682/expansion.kicad_pcb" for reading```

The schematics are showing fine now which is great.

leoheck, 2021-05-31:
I think this is done. I believe I have fixed it in the last commit. Please check if the new changes work for you. You can create new issue after that.

closed 2021-05-31 by leoheck