leoheck/kiri #7: Kicad on Windows + Git

stevaedrum, 2021-05-30:
Hi,

I permit to create an issue. Because I use Kicad in Windows and managing version with git.
When use kdiff, I have see you change dos2unix, but after git detect that the file have change.
The file is not backup correctly, see the correction below:

	# Workaround for windows line endings.
-	if [[ -f "${kicad_pcb}.bkp" ]]; then
+	if [[ ! -f "${kicad_pcb}.bkp" ]]; then
		cp "${kicad_pcb}" "${kicad_pcb}.bkp" 2> /dev/null
	fi
	dos2unix "${kicad_pcb}"

And I have change cp by mv to restore

	# Restore board
	if [[ -f "${kicad_pcb}.bkp" ]]; then
-		cp "${kicad_pcb}.bkp" "${kicad_pcb}" &> /dev/null
+		mv "${kicad_pcb}.bkp" "${kicad_pcb}" &> /dev/null
	fi

Have a nice day

leoheck, 2021-05-31:
Hey, this is a dirty workaround for another user. But this suggestion seems good enough. It is already in the latest version. Could you please check if it still works for you?

stevaedrum, 2021-06-02:
Hi Leandro,

Thanks for your answer and feedback, It’s working for me. Great script and job.

closed 2021-06-02 by stevaedrum