leoheck/kiri #14: Is it possible to produce diffs without -cache.lib files?

edahlseng, 2021-09-14:
The team I'm working with doesn't track `-cache.lib` files in Git. All of our schematics open properly in KiCad, but it looks like `kdiff` expects `-cache.lib` files to exist when calling `plotgitsch`. Can those files be optional?

leoheck, 2021-09-14:
This is a question that can be better addressed on plotgitsch repo.

But, here, I can say that you can try this by your own. The process is simple, open the main script and locate the plotgitsch line(s) an remove the reference to the plotgitsch file. It may work regarding the project (or due to the last features I have asked plotgitsch author). But I would suggest you to have this cache stored in your repo from now on.

Does it make sense?
If you see this working, we could have a command line flag to disable this or to skip this file if it does not exist.

edahlseng, 2021-09-14:
Thank you for that! I will look into this further to see whether the file is needed or not.

leoheck, 2021-11-16:
I am closing this for now due to the lack of activity. Feel free to reopen it if needed.

closed 2021-11-16 by leoheck

edahlseng, 2021-11-19:
Following up on this, it appears that in KiCad 6 schematic files will embed the symbols similar to how layout files embed the footprints. If my understanding of how this works is right, then the cache file won’t be needed anymore at that point, and I’ll revisit this question again then!

leoheck, 2021-11-19:
Yeah but then plotgitsch will have to support it, right now it only works with Kicad 5

leoheck, 2021-11-19:
kiri also has this flag enabled

```
       -m, --allow_missing
           by default, a missing component aborts the comparison. With this
           option, a missing component is skipped and the process continues.
```

Maybe it is possible to remove the `-l lib_filename` form plotgitsch command line.
You could test this. If it works I can add a command-line flag to remove it

leoheck, 2021-11-20:
I added this flag `-s` to ignore `-cache.lib` in the last commit, hope it works for you.