leoheck/kiri #15: SVGs not being output for schematics?

edahlseng, 2021-09-14:
I'm trying to get this up and running on my local machine (macOS), but I just can't seem to get this working. I'm starting with a focus on the schematic diffs... it looks like no SVG files are actually being generated, and so `kdiff` is just using the blank SVG. I notice a text diff output in the console when I run `kdiff --debug --verbose`, but still no SVGs are generated. Do you have any pointers on how to debug this?

leoheck, 2021-09-15:
I would recommend you to try the -r flag to remove old files. And it may help. Also, use the verbose, to see the actual command kidiff is running. Then you can copy just one of those lines and try it manually. You can also post the output of one of those plotgitsch lines here so we can have a better discussion.

edahlseng, 2021-09-15:
Thanks @leoheck! Here's some more information on the steps and output I'm running:

```
git clone https://github.com/jnavila/kicad_playground.git
cd kicad_playground
patch -p1 -i changes.diff
cd schematic
kdiff --verbose --debug --remove kicad_playground.pro
```
At this point, all I see are blank screens in the `kdiff` schematic viewer. Looking through the `kdiff` debug output, one of the lines that was run for generating the schematic files was this one:
```
plotgitsch -r schematic -m -l ./kicad_playground-cache.lib -c FF0000:008FFF:FFFFFF:222222 c0a3b8e cdb8005 # &> /dev/null
```
and it printed out a diff. If I run that command directly in my local shell I see the following diff output on stdout, but no .svg files appear to be created 😕 .
<details>
<summary>diff --git a/schematic/kicad_playground.sch b/schematic/kicad_playground.sch</summary>
index fff8c68..69f2175 100644
--- a/schematic/kicad_playground.sch
+++ b/schematic/kicad_playground.sch
@@ -1,4 +1,125 @@
EESchema Schematic File Version [-2-]{+4+}
EELAYER [-25-]{+26+} 0
EELAYER END
{+$Descr A4 11693 8268+}
{+encoding utf-8+}
{+Sheet 1 2+}
{+Title ""+}
{+Date ""+}
{+Rev ""+}
{+Comp ""+}
{+Comment1 ""+}
{+Comment2 ""+}
{+Comment3 ""+}
{+Comment4 ""+}
{+$EndDescr+}
{+$Comp+}
{+L Timer:LM555 U?+}
{+U 1 1 5BC794C1+}
{+P 3050 2000+}
{+F 0 "U?" H 2750 2350 50  0000 C CNN+}
{+F 1 "LM555" H 3300 2350 50  0000 C CNN+}
{+F 2 "" H 3050 2000 50  0001 C CNN+}
{+F 3 "http://www.ti.com/lit/ds/symlink/lm555.pdf" H 3050 2000 50  0001 C CNN+}
{+	1    3050 2000+}
{+	1    0    0    -1  +}
{+$EndComp+}
{+$Comp+}
{+L Device:R R?+}
{+U 1 1 5BC79558+}
{+P 4000 1250+}
{+F 0 "R?" H 4070 1296 50  0000 L CNN+}
{+F 1 "220K" H 4070 1205 50  0000 L CNN+}
{+F 2 "" V 3930 1250 50  0001 C CNN+}
{+F 3 "~" H 4000 1250 50  0001 C CNN+}
{+	1    4000 1250+}
{+	1    0    0    -1  +}
{+$EndComp+}
{+Wire Wire Line+}
{+	2550 2200 2350 2200+}
{+Wire Wire Line+}
{+	2350 2200 2350 1000+}
{+Wire Wire Line+}
{+	2350 1000 3050 1000+}
{+Wire Wire Line+}
{+	3050 1000 3050 1600+}
{+$Comp+}
{+L Device:R R?+}
{+U 1 1 5BC795B1+}
{+P 4000 2350+}
{+F 0 "R?" H 4070 2396 50  0000 L CNN+}
[... 156 lines cut ...]
{+P 3350 2350+}
{+F 0 "D?" V 3295 2428 50  0000 L CNN+}
{+F 1 "LED" V 3386 2428 50  0000 L CNN+}
{+F 2 "" H 3350 2350 50  0001 C CNN+}
{+F 3 "~" H 3350 2350 50  0001 C CNN+}
{+	1    3350 2350+}
{+	0    -1   -1   0   +}
{+$EndComp+}
{+Wire Wire Line+}
{+	3350 2500 3350 2600+}
{+Wire Wire Line+}
{+	3350 2900 3350 3050+}
{+$Comp+}
{+L power:GND #PWR?+}
{+U 1 1 5BC7CECD+}
{+P 3350 3050+}
{+F 0 "#PWR?" H 3350 2800 50  0001 C CNN+}
{+F 1 "GND" H 3355 2877 50  0000 C CNN+}
{+F 2 "" H 3350 3050 50  0001 C CNN+}
{+F 3 "" H 3350 3050 50  0001 C CNN+}
{+	1    3350 3050+}
{+	1    0    0    -1  +}
{+$EndComp+}
{+$EndSCHEMATC+}
</details>

leoheck, 2021-09-15:
You may have the original plotgitsch file. I forked it and made some changes, those where not added back to the upstream (yet).

If you have my fork of the plotgitsch installed, go inside your `kicad_playground` example and run `plotgitsch` only without arguments. It should generate the following svg files.

[image]

To compare plotgitsch version do this
```
➜ plotgitsch --version
v0.3.0-157-g4603ff9
```

leoheck, 2021-09-15:
@edahlseng look this. To tease you even more, this is a view of this `kicad_playground` repo using my `kdiff`. It may be tricky to setup, but I will help you to put this working in your machine too.

[image]

leoheck, 2021-09-15:
This it the command I have used by the way. 
```
kdiff -V -r kicad_playground.pro
```

leoheck, 2021-09-15:
Ah, this is the `plotgitsch` I use.
https://github.com/leoheck/plotkicadsch

edahlseng, 2021-09-15:
Thank you for your help! It looks like we're getting somewhere... I did notice that you had forked plotkicadsch, and had installed from your fork. Perhaps I did not clean properly in between installing the original and your fork.

Running `plotgitsch` in the directory with no other arguments does not create any files...

```
$ plotgitsch --version
%%VERSION%%
```

I installed using the instructions in the README, rather than the `./install.sh`.

Next I will try uninstalling everything, then installing with the `./install.sh` script, and see how that goes!

edahlseng, 2021-09-15:
@leoheck, after following the build steps in `./install.sh`, I do get a version number now, but SVGs are not output when running `plotgitsch` alone. The output of `plotgitsch --version` is `v0.8.0-41-g4603ff9-dirty`, which matches the commit SHA from your version output (`4603ff9`).

What platform are you running on? Is there any chance that something different is being done on macOS than on other platforms?

leoheck, 2021-09-15:
I am using on linux, but I do test on Windows and on OSX. So it has to work everwhere. But the setup may not be as easy as I put in the README. Maybe it is missing something.

But cool, since you have my plotgitsch installed, go inside your kicad_playground project.
And try to run plotgitsch only

Almos the same way as you did before but instead of using kdiff use plotgitsch

```
#git clone https://github.com/jnavila/kicad_playground.git
cd kicad_playground
#patch -p1 -i changes.diff
cd schematic
#kdiff --verbose --debug --remove kicad_playground.pro
plotgitsch
```
It should output something like this

```
kicad_playground/schematic on  master [!] 
➜ plotgitsch 
use compare (ImageMagick) between bitmaps between Git rev HEAD and file system .

Command line:
/home/lheck/Dropbox/Documents/kdiff/git-imgdiff /tmp/from_LEDsa249a3.svg /tmp/to_LEDs7c61ff.svg

Command line:
/home/lheck/Dropbox/Documents/kdiff/git-imgdiff /tmp/from_kicad_playground2bd164.svg /tmp/to_kicad_playground08bfad.svg

Exporting...
/tmp/from_LEDsa249a3.svg -> sch1-LEDs.svg
/tmp/to_LEDs7c61ff.svg -> sch2-LEDs.svg

Exporting...
/tmp/from_kicad_playground2bd164.svg -> sch1-kicad_playground.svg
/tmp/to_kicad_playground08bfad.svg -> sch2-kicad_playground.svg
```

This output is generated by my version of plotgitsch only.

edahlseng, 2021-09-15:
Hmm, this is what I've got:

```
#git clone https://github.com/jnavila/kicad_playground.git
$ cd kicad_playground
$ patch -p1 -i changes.diff
$ cd schematic
$ plotgitsch --version
v0.8.0-41-g4603ff9
$ plotgitsch
use compare (ImageMagick) between bitmaps between Git rev HEAD and file system .
diff --git a/schematic/kicad_playground.sch b/schematic/kicad_playground.sch
index 69f2175..a950b3e 100644
--- a/schematic/kicad_playground.sch
+++ b/schematic/kicad_playground.sch
@@ -78,7 +78,7 @@ Wire Wire Line
Wire Wire Line
	3050 2550 2450 2550
Wire Wire Line
	2450 2550 2450 [-1800-]{+2300+}
Wire Wire Line
	2450 1800 2550 1800
Connection ~ 3050 2550
@@ -122,4 +122,23 @@ F2 "DRV" I L 4450 1800 50
$EndSheet
Wire Wire Line
	3050 1000 4000 1000
{+$Comp+}
{+L Connector:Screw_Terminal_01x02 J?+}
{+U 1 1 5BC7DEAC+}
{+P 1500 2200+}
{+F 0 "J?" H 1420 2417 50  0000 C CNN+}
{+F 1 "Screw_Terminal_01x02" H 1420 2326 50  0000 C CNN+}
{+F 2 "" H 1500 2200 50  0001 C CNN+}
{+F 3 "~" H 1500 2200 50  0001 C CNN+}
{+	1    1500 2200+}
{+	-1   0    0    -1  +}
{+$EndComp+}
{+Wire Wire Line+}
{+	1700 2200 2350 2200+}
{+Connection ~ 2350 2200+}
{+Wire Wire Line+}
{+	1700 2300 2450 2300+}
{+Connection ~ 2450 2300+}
{+Wire Wire Line+}
{+	2450 2300 2450 1800+}
$EndSCHEMATC
```

`plotgitsch` exists with status code `0`, and no `.svg` files are generated.

edahlseng, 2021-09-16:
I found it! I was building this in an isolated environment and then moving the executables out onto my path. In that process, I was not moving the `git-imgdiff` executable. I've included that in the move step, and the schematic images are now being produced as expected!

Thanks for all of your help debugging!

closed 2021-09-16 by edahlseng