leoheck/kiri #16: Github Action

BenRoe, 2021-09-22:
Hi,
this project looks very promising. I know it's early stage, but a github action would be great and very beginner friendly.
Unfortunately i don't have the time to work on it.

leoheck, 2021-09-22:
Yeah, it would be great indeed. So no one would need to install this tool, and they could use it online directly.

I made a try on this some months ago. I was able to generate everything and all, but I was not able to access the generated web-page or to embed images in the PR page directly. So, that time, I had to download a generated zip file with the results, which is not practical for me. So, I dropped the idea. 

Do you know any a way to create a webserver and share the link so authors can interact with the resulting website?

jacobrosenthal, 2021-11-14:
Github actions can leave comments and comments can have images.
I did a random google and heres an example https://github.com/marketplace/actions/comment-run-actions

leoheck, 2021-11-16:
This is not quite related to this repo, so I am closing this for now. Feel free to reopen it if needed.

closed 2021-11-16 by leoheck