leoheck/kiri #19: empty `title`, `revision`, `date` html fields

maholli, 2021-11-14:
First off, I've been loving this tool! Thank you for streamlining kidiff and plotgitsch, it's a pleasure to use.

Minor error I have is a few html fields aren't populating. See pic below:
[image]

I tried fiddling with the bash, like [here](https://github.com/leoheck/kdiff/blob/01a20b4c8fcf2a66ddc849e44c28851ca3bfd9c6/kdiff#L863), but all I can determine is `$title == ""` is returning `true`, but not sure why. Looks like you're just extracting it from the filename, right?

- using WSL2

Any tips would be appreciated. 😄

leoheck, 2021-11-14:
Hi. This tool started from the layout. This means this data comes from the layout page. Could you check if that filling this information on PC new page solves your problem?

maholli, 2021-11-14:
Ah! I had only set these in the schematic and never knew the layout also had separate definitions. 

That solved it. Thank you!!

closed 2021-11-14 by maholli

leoheck, 2021-11-14:
Glad that it helped. Thanks for the feedback. I will have to check if it is
possible to keep it updated with the schematic too.

leoheck, 2021-11-16:
Hey, @maholli could you test the latest version on master? I improved this info. When showing the layout it shows info from the layout, and when showing the schematics, it shows the info from the schematics.

Hope it works for you.

[image]
[image]