leoheck/kiri #21: build ocaml-base-compiler fails

ckuhtz, 2021-11-14:
This is on impish.  Anyone have an idea what is going on here? I'm not familiar with ocaml and just following directions from [README.md](https://github.com/leoheck/kdiff#readme)

```
----8<---------------------------------------

# gcc -c -O2 -fno-strict-aliasing -fwrapv -Wall -Werror -fno-common -fno-tree-vrp -g -D_FILE_OFFSET_BITS=64 -D_REENTRANT -DCAML_NAME_SPACE  -DOCAML_STDLIB_DIR='"/home/christian/.opam/4.09.1/lib/ocaml"'  -DNATIVE_CODE -DTARGET_amd64 -DMODEL_default -DSYS_linux   -o misc_n.o misc.c
# signals_nat.c:185:13: error: variably modified ‘sig_alt_stack’ at file scope
#   185 | static char sig_alt_stack[SIGSTKSZ];
#       |             ^~~~~~~~~~~~~
# make[3]: *** [Makefile:343: signals_nat_n.o] Error 1
# make[3]: *** Waiting for unfinished jobs....
# make[3]: Leaving directory '/home/christian/.opam/4.09.1/.opam-switch/build/ocaml-base-compiler.4.09.1/runtime'
# make[2]: *** [Makefile:1002: makeruntimeopt] Error 2
# make[2]: Leaving directory '/home/christian/.opam/4.09.1/.opam-switch/build/ocaml-base-compiler.4.09.1'
# make[1]: *** [Makefile:415: opt.opt] Error 2
# make[1]: Leaving directory '/home/christian/.opam/4.09.1/.opam-switch/build/ocaml-base-compiler.4.09.1'
# make: *** [Makefile:466: world.opt] Error 2

<><> Error report <><><><><><><><><><><><><><><><><><><><><><><><><><><><><><><>
┌─ The following actions failed
│ λ build ocaml-base-compiler 4.09.1
└─ 
┌─ The following changes have been performed (the rest was aborted)
│ ∗ install base-bigarray base
│ ∗ install base-threads  base
│ ∗ install base-unix     base
└─ 

<><> ocaml-base-compiler.4.09.1 troubleshooting <><><><><><><><><><><><><><><><>
=> A failure in the middle of the build may be caused by build parallelism
      (enabled by default).

```

leoheck, 2021-11-14:
What is impish?

Also, this is related to plotgitsh. Kidiff uses my fork of it. But I am not modifying it in a way that makes it fail. 
This may be related to the installation of plotgitsch/oCaml stuff. I am also not pretty aware of how to make it work in a way that always works.   

Could you please bring this discussion here https://github.com/jnavila/plotkicadsch ?

ckuhtz, 2021-11-14:
Impish is Ubuntu 21.10.  I just tried 4.13.1 for grins and giggles and it built successfully.  Any issues with 4.13.1 instead of 4.09.1 that you know of?  Maybe we can just change that in the instructions?

ckuhtz, 2021-11-14:
FWIW, I hadn't started plotkicadsch yet, was only building dependencies/prereq.

leoheck, 2021-11-14:
Ah, I see. I don't use ubuntu names they are useless, just numbers. I am also using 21.10.

I am not aware of any issue with 4.13.1, I just did not try it enough since I was following what Jnavila explained on his repo. But sure, test it. Let me know if it works, please. I just have a feeling that I tried this some time ago and something was not working fine.

leoheck, 2021-11-16:
Hi @ckuhtz did you solve your problem?

closed 2021-11-16 by leoheck