leoheck/kiri #24: gsed: -e expression #1, char 33: unterminated `s' command

edahlseng, 2021-12-02:
I just installed the tip of main and am getting the following error when running Kiri:
```
gsed: -e expression #1, char 33: unterminated `s' command
```

I haven't dug in at all, yet. Any chance this looks like an easy workaround or fix?

leoheck, 2021-12-02:
Oh really? What is your OS?

leoheck, 2021-12-02:
Yeah, it may be just a small issue. I just have to check your OS which I believe it is a macOS. And then check where the issue is. This is pretty interesting because I have tested it on all operating systems some days ago.

edahlseng, 2021-12-02:
macOS Monterey. If it looks like it might be something specific with my system, I’d appreciate any pointers with how to debug!

leoheck, 2021-12-02:
Are you trying to fix this right now?

edahlseng, 2021-12-02:
Not at the moment, won’t have a chance to dig for a couple days. Just wanted to flag that if you’ve already tested, it could very well be something specific to my system.

leoheck, 2021-12-02:
I found a macOS with Monterey. Let me see if I have the same issue once I install everything.

leoheck, 2021-12-02:
Sorry to say this, but it ran fine here. But I was wrong. It is the Big Sur. I am going to try to upgrade it (through VNC, so it will take time, and may not work) so... 

I suggest you run these 2 commands again to prepare everything.

```bash
# Installing dependencies
bash -c "$(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_dependencies.sh)"
```

```bash
# Installing KiRI and Kicad Plugin
bash -c "$(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_kiri.sh)"
```

leoheck, 2021-12-02:
Alright. Running it on Monterey and it is working.

leoheck, 2021-12-02:
Let me know if you are still having issues after running 2 lines above.

edahlseng, 2021-12-03:
It looks like this may be related to paths that I have spaces in them. I cloned the repository to a different location on my machine that did not have spaces in the path, and this error does not show up.

leoheck, 2021-12-03:
Alright. I am going to check how I can improve this. It may fail for me too. But common, do yourself a better job. Do not put spaces on paths. Unless you are not a programmer/engineer/hacker.

leoheck, 2021-12-04:
Hey @edahlseng, I could improve the script a bit for paths with spaces. Please, check this for me and close this issue if it is solved.

leoheck, 2021-12-05:
Feel free to open the issue if the problem persists!

closed 2021-12-05 by leoheck

edahlseng, 2021-12-06:
> Do not put spaces on paths.

Technology should not force unnecessary constraints on users! I've used spaces in paths for my entire career and it has worked out just fine. I actually find it helps make my scripts more robust.

It looks like the recent change fixes this issue on my end, too! Thanks!