leoheck/kiri #25: cp: directory /path/to/project/kiri/web does not exist

edahlseng, 2021-12-06:
When running `kiri --remove ./project.pro`, I'm getting the following error message:

```
Assembling the HTML
cp: directory /path/to/project/kiri/web does not exist
< many more error messages much like this >
```

Any ideas on what might be causing this? (I moved into a path that doesn't have spaces, just to be sure! :smile: )

edahlseng, 2021-12-06:
Looks like the `web` directory creation [happens right before](https://github.com/leoheck/kiri/blob/main/bin/kiri#L692-L699) the `--remove` cleanup step! Running with `--remove` seems to work just fine.

leoheck, 2021-12-06:
Remove, removes old folder. not the new one.

leoheck, 2021-12-06:
Alright, I see what you are talking. It is fixed.

closed 2021-12-06 by leoheck

leoheck, 2021-12-06:
Thank you by the way.

edahlseng, 2021-12-21:
You're welcome! Thank you for the fix! Confirmed that it's working as expected on my machine now, too!

leoheck, 2021-12-21:
Awesome