leoheck/kiri #26: Submit plugin to PCM metadata repo

OmegaRogue, 2021-12-11:
Submitting this library to the new Plugin And Content Manager would make it more easy to find and install, though there might be problems with that due to plotkicadsch using OCAML.

Documentation for this is available in:
https://dev-docs.kicad.org/en/addons
The Metadata repo is located on:
https://gitlab.com/kicad/addons/metadata

leoheck, 2021-12-11:
I want to do that but...

It does not work on Kicad >= 5.99 yet.
  - Kicad-Diff kind of works (but still can be improved)
  - Plotgitsh does not work yet
 
Also, this tool requires a lot of external tools to work. So I believe the installer may not work there.

leoheck, 2021-12-14:
Hi. I am closing this right now since it looks like it won't be that easy/good to integrate this tool on Kicad's PCM due to a lot of required dependencies. 

Feel free to open this if needed.

closed 2021-12-14 by leoheck

leoheck, 2021-12-22:
Hey, @OmegaRogue my pretty sad start here, https://gitlab.com/kicad/addons/metadata/-/merge_requests/27#note_784710740

OmegaRogue, 2021-12-24:
> Hey, @OmegaRogue my pretty sad start here, https://gitlab.com/kicad/addons/metadata/-/merge_requests/27#note_784710740
> Do you have any contributions to help me? There is already a way to install the toolbar button on Kicad using my install scripts. But there is nothing on PCM already, it looks people are pretty resistant to accept different things, there. 

If the Official PCM repo is a problem for now, I made an "Unofficial" PCM repo. https://github.com/OmegaRogue/Unofficial-KiCad-Plugins

I can take a look, but I haven't worked with KiCAD APIs yet.