leoheck/kiri #27: Diffs not rendering on mac

apollokit, 2021-12-12:
Just installed per the readme, didn't run into any problems. Ran kiri on a [kicad project 
](https://github.com/nasa-jpl/open-source-rover/blob/master/electrical/pcb/control_board/Control%20Board.pro) and this is what I ended up with:

[image]

Kiri didn't complain at all. When I look at the artifacts it produced, the svg's are all just blank:

[image]

Any advice for how to fix this?

leoheck, 2021-12-13:
Hey @apollokit, thanks for sharing the Kicad project you are using.
This repo is not the easy/standard thing to work on.
It has nested projects. It has folders and files named with spaces in it. I will have to investigate what is going on.

You can check the command line of kiri for more flags to help you to investigate this further. You can also open kiri script investigate closely. 

If you run this command, for instance, 
```
kiri Control\ Board.pro -V -D -r
```

you are going to see that kiri get the commits list with
```
kiri-git-log master -- electrical/pcb/control_board/Control Board.sch  electrical/pcb/control_board/Control Board.kicad_pcb
```

I ran the command above, and could not see a single line.

I changed it a bit to test and I am seeing this
```
open-source-rover/electrical/pcb/control_board on  master [?] 
➜ kiri-git-log master -- *.kicad_pcb 
cbcf3bf | 2020-12-02 19:01:28 -0800 | Kit Kennedy | Rename all the directories%   
```

So, maybe this project was renamed and reorganized multiple times. I haven't checked this further. 

It may be an issue with kiri. But it may be really hard to track files/folders in this repository.

leoheck, 2021-12-13:
If the issue is the history of the git that is pretty chaotic, then I would advise you to find a git command that rewrites the history of git reducing this chaos, and then try to use Kiri again, as a test.

leoheck, 2021-12-14:
I was investigating this one more time today, and it looks like there is nothing on git history that shows other versions/commits of the files.

What I see is that all PCB-related files, for both projects there has a single logline. 
So, this way it is not possible to track the version, unfortunately.

```cbcf3bf Rename all the directories```

Sorry, but since it looks like a bad git usage issue, and not Kiri's issue, I am closing this right now. 

Feel free to open it if you have extra information.

closed 2021-12-14 by leoheck