leoheck/kiri #31: Creation of SVG problem

stevaedrum, 2021-12-28:
Hi,

I have reinstalled "kiri" tools, previously i use "kdiff" (Great tools). Now i have a problem with the "kiri".
When i launch the tool, I have an errror because the svg file aren't not generated.

Image Magick is opening in my desk with the comparaison image !!!

When I launch with several parameters I have this returns.

--------------------------------------------------------------------------------------------------------
[a.stephan1@frtn01s ~]$ kiri -r -l -D -dp -dk

      kicad_pro_path = /mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-Tools-Kiri/TopBoard/TopBoard.pro
           repo_path = /mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-Tools-Kiri/TopBoard
   kicad_pro_subpath = .
           kicad_pro = TopBoard.pro
           kicad_sch = TopBoard.sch
           kicad_pcb = TopBoard.kicad_pcb
       KICAD_VERSION = 5
   KICAD_PCB_VERSION = 20171130
  kicad_pro_filename = TopBoard
 kicad_pro_extension = pro
 kicad_sch_extension = sch
 kicad_pcb_extension = kicad_pcb
          output_dir = /home/a.stephan1/kiri

Output folder: /home/a.stephan1/kiri
Removing leftovers from the previous run

Generating artifacts
25 commits, totaling 600 comparisons

   1. 4315759 - aac2582
   plotgitsch  -l ./TopBoard-cache.lib -k -m -c FF0000:008FFF:FFFFFF:222222 4315759 aac2582
use compare (ImageMagick) between bitmaps between Git rev 4315759 and Git rev aac2582
   Recreating missing sch1-TopBoard.svg
   Recreating missing sch2-TopBoard.svg
   kidiff TopBoard.kicad_pcb --webserver-disable -b 4315759 -a aac2582 -o "/home/a.stephan1/kiri"

      SCM Selected: git
  Kicad Board Path: /mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-Tools-Kiri/TopBoard/TopBoard.kicad_pcb
Kicad Project Path: /mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-Tools-Kiri/TopBoard
         REPO Path: /mnt/Data-Raid1/data/VirtualMachines/ASCIE/Design/Personal_Projects/Test-Tools-Kiri/TopBoard
 Kicad Project Dir: .
   Board File Name: TopBoard.kicad_pcb
        Output Dir: /home/a.stephan1/kiri

Commit 1 (a): aac2582
Commit 2 (b): 4315759
sed: pas de fichier d'entrée
sed: pas de fichier d'entrée
   Recreating missing file: /home/a.stephan1/kiri/4315759/TopBoard-00-F_Cu.svg
   Recreating missing file: /home/a.stephan1/kiri/aac2582/TopBoard-00-F_Cu.svg
   Recreating missing file: /home/a.stephan1/kiri/4315759/TopBoard-31-B_Cu.svg
   Recreating missing file: /home/a.stephan1/kiri/aac2582/TopBoard-31-B_Cu.svg
   2.         - 30dbd9c
-------------------------------------------------------------------------------------------

Thanks in advance for your help.

Best regards.

Stephan

leoheck, 2021-12-28:
Hi @stevaedrum , 
What is your Operating System and version of it? 
What is the version of Kicad you have installed?
Is it possible to share the repo of your Kicad project so I can test it myself?

leoheck, 2021-12-28:
What is the version of kiri you have installed? You can check it inside kiri's folder with `git rev-parse HEAD`

stevaedrum, 2021-12-28:
Thank for your quick answer.

I use Fedora 34 as OS. The version of Kicad is v 5.1.12
I have tested on other small Kicad project, it's the same problem.
Yes it's possible to share the project.

git rev-parse HEAD return the key below:

391a6ba66355c7549f7c147d693f2d39a2798221

I have installed kiri by the cmd below:
# Installing KiRI and Kicad Plugin
bash -c "$(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_kiri.sh)"

kiri -v return this informations:

    plotgitsch v0.9.0-6-gf26942b
    kidiff
    kiri 391a6ba 2021/12/25 19h52

leoheck, 2021-12-28:
Nice. I may have added some issues when trying to make it work with Kicad 6. I am trying to check this issue. If you don't mind sharing the project you are experiencing issues with, I would be glad. The version I have locally (maybe with some new features) is not failing when running on my sample project.

stevaedrum, 2021-12-28:
Please could you find the project below ;)

[SchemaPCB.zip](https://github.com/leoheck/kiri/files/7784826/SchemaPCB.zip)

leoheck, 2021-12-28:
Hey, @stevaedrum I pushed some changes. I hope I could have addressed your issues! Let me know if you have any other problems, please. If it is ok, please close this issue for me.

leoheck, 2021-12-28:
ah, there is one extra dependency (scour). check the dependencies script, or run it again

stevaedrum, 2021-12-28:
Hey @leoheck,

Thanks for your quick answer.
Please "scour" is obsolete or depreciated on Fedora distribution.
What do scour ?

Thanks in advance.

Stephan.

leoheck, 2021-12-28:
Oh, ok. I will try to replace it then. It looks it is not that easy to find
on Ubuntu too.

It is used to optimize generated Svg files. Please, comment this command
and check if it works for you. You will find it in 4 places.

leoheck, 2021-12-28:
I see. Your issue is not kiri. It may be plotgitsch installation. The output of plotgitsch is a single image. Not a compound image. 

Maybe it is just an issue with the path.

This script has to override the same one from plotgitsch repo.
It must be pre-appended in path
```
➜ which git-imgdiff
/home/lheck/Dropbox/Documents/kiri/bin/git-imgdiff
```

How are you setting the PATH?
It must be something like this. In this exact order:
```
eval $(opam config env)
source $HOME/Documents/kiri/env.sh
source $HOME/Documents/kiri/submodules/KiCad-Diff/env.sh
```

stevaedrum, 2021-12-29:
hi @leoheck,

I have checked, the path all are correct !!!

I continue to investigate.

leoheck, 2021-12-29:
What is the output of the 
```
which git-imgdiff
```

stevaedrum, 2021-12-29:
Please could you find the request.

[a.stephan1@frtn01s SchemaPCB]$ which git-imgdiff
~/.opam/4.09.1/bin/git-imgdiff

leoheck, 2021-12-29:
> Please could you find the request.
>
> ~/.opam/4.09.1/bin/git-imgdiff

So, this is wrong. There is one script with the same name. This script path
has to preappened in the path to be able to ignore the original one.

stevaedrum, 2021-12-30:
Whaow, yes it's working.
I have remove the incorrect version, thanks fot your quick answer.

But, I have detected always a bug concerning the variable, see below the screen shot.

[image]
[image]

Great Tools

For your information and idea, could you know this pro tools
[https://cadlab.io/]

leoheck, 2021-12-30:
this is not a bug. these fields come from the schematic and the layout.
each one has a sheet property where O am getting these information. if you
update these fields in your Kicad project you are going to see this working

leoheck, 2021-12-30:
See the Title, date, rev are empty 

[image]

[image]

stevaedrum, 2021-12-30:
Ohhhh sorry ;) ;)

Great, good news, thanks a lot for your support and quick answer.
If you want some help to test new function don't heisted ;)

Have a nice day.

closed 2021-12-30 by stevaedrum

leoheck, 2021-12-30:
Alright! Good to know that the issue was fixed for you! Cheers.