leoheck/kiri #32: Kiri under Windows 10 Home/Ubuntu (issue/install log)

kharar, 2022-01-05:
First problem I ran into was the Ubuntu installer, at time of my attempts it seems to fetch me an 80kb corrupted zip-file that neither be expanded within PowerShell nor opened by WinRAR.
So since the URL points to a Microsoft site anyways I decided to continue with the Ubuntu 20.04 TLS installer from the Windows Store and after that, the installation of dependencies and submodules using the single line commands seemed to work without too much hassle albeit it took several hours to complete on my HP EliteBook 8570w with SSD. While waiting for this I downloaded an installed Xming on the Windows side...

It should be noted that when attempting to start KiRi from the current directory it is important to include _./_ before _kiri_ == _./kiri_ otherwise KiRi can not be started (it is a Linux security issue). I'd suggest this be included in the README.md

After successfully starting KiRi, then Xming gracefully took over and I found my Windows drives were mounted in the file system at /mnt... so far so good.
I went to my favorite project folder and loaded a KiCAD project file... and **Bump!** lots of errors..

A browser tab was opened at 127.0.0.1:8080 with the following text:
```
Error response
Error code: 404

Message: File not found.

Error code explanation: HTTPStatus.NOT_FOUND - Nothing matches the given URI.
```

And here's the output of the Ubuntu terminal:
```~/.local/share/kiri/bin$ ./kiri
mkdir: cannot create directory ‘/home/kharar/.local/share/kiri/bin/kiri’: File exists
./kiri: line 1404: kicad_version.py: command not found

      kicad_pro_path = /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board/2S_serial_charging_board.pro
           repo_path = /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board
   kicad_pro_subpath = .
           kicad_pro = 2S_serial_charging_board.pro
           kicad_sch = 2S_serial_charging_board.sch
           kicad_pcb = 2S_serial_charging_board.kicad_pcb
       KICAD_VERSION =
   KICAD_PCB_VERSION = 20171130
  kicad_pro_filename = 2S_serial_charging_board
 kicad_pro_extension = pro
 kicad_sch_extension = sch
 kicad_pcb_extension = kicad_pcb
          output_dir = /home/kharar/.local/share/kiri/bin/kiri

Output folder: /home/kharar/.local/share/kiri/bin/kiri
mkdir: cannot create directory ‘/home/kharar/.local/share/kiri/bin/kiri’: Not a directory
./kiri: line 1001: kiri-git-log: command not found

Generating artifacts
1 commits, totaling 0 comparisons

Assembling the HTML
cp: cannot stat '/home/kharar/.local/share/kiri/bin/kiri/web//': Not a directory
cp: cannot stat '/home/kharar/.local/share/kiri/bin/kiri/web//': Not a directory
cp: cannot stat '/home/kharar/.local/share/kiri/bin/kiri/web//': Not a directory
cp: cannot stat '/home/kharar/.local/share/kiri/bin/kiri/web//': Not a directory
cp: cannot stat '/home/kharar/.local/share/kiri/bin/kiri/web//': Not a directory
cp: cannot stat '/home/kharar/.local/share/kiri/bin/kiri/web//../index.html': Not a directory
cp: cannot stat '/home/kharar/.local/share/kiri/bin/kiri/': Not a directory
- Adding commits list to the page
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//index.html: Not a directory
- Adding schematic pages list to the page
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//index.html: Not a directory
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//index.html: Not a directory
- Setting sch/pcb titles
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//index.html: Not a directory
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//index.html: Not a directory
- Setting sch/pcb revision
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//index.html: Not a directory
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//index.html: Not a directory
- Setting sch/pcb date
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//index.html: Not a directory
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//index.html: Not a directory
- Setting board name
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//kiri.js: Not a directory
- Setting initial commits
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//index.html: Not a directory
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//index.html: Not a directory
- Setting initial schematics
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//index.html: Not a directory
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//index.html: Not a directory
- Setting initial layouts
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//index.html: Not a directory
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//index.html: Not a directory
- Adding layers list to the page
sed: can't read /home/kharar/.local/share/kiri/bin/kiri/web//index.html: Not a directory

Output folder: /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board/kiri
Starting web-server at http://127.0.0.1:8080
Hit Ctrl+C to end session```

Now I wonder what went wrong?

leoheck, 2022-01-05:
It looks everything is wrong. haha.
What is the OS you are using?

kharar, 2022-01-05:
Yeah, but I notice the path on many lines have a double //,,, maybe a parsing bug somewhere?
I'm using Windows 10 Home

leoheck, 2022-01-05:
It is not meant to use on Windows directly. You have to have WSL (Windows Subsystem for Linux) which is a Linux inside Windows. Do you have WSL configured?

kharar, 2022-01-05:
Yes, as I said in forst post I had to revert to the Ubuntu 20.04 TLS installer from the Windows Store

leoheck, 2022-01-05:
Alright. How did you install it? Where did you install it? Having to put this "./" in front of kiri, means you are not setting the environment correctly.

So, use this code here, update this thing `[PATH_TO_KRIRI]` with the actual path
And the type these lines directly in the terminal
```
eval $(opam env)
export TK_SILENCE_DEPRECATION=1
export PATH=[PATH_TO_KRIRI]/kiri/submodules/KiCad-Diff/:PATH
export PATH=[PATH_TO_KRIRI]/kiri/bin:PATH
```

Now, go inside any folder and then check if you can run `kiri` as it is.

kharar, 2022-01-05:
This is currently at the end of my .bashrc file
```
# Kiri environment setup
eval $(opam env)
export TK_SILENCE_DEPRECATION=1
export PATH=/home/kharar/.local/share/kiri/submodules/KiCad-Diff/:${PATH}
export PATH=/home/kharar/.local/share/kiri/bin:${PATH}

```

leoheck, 2022-01-05:
then check if the kiri is available in your environment

kharar, 2022-01-05:
I just misunderstood instructions that says:
```
Using KiRI
cd [kicad_git_repo]
kiri board.pro
```
Thought I had to go to the KiRi folder... Duh!!

Starting it from my home folder seemed to do the trick!
:-D

leoheck, 2022-01-05:
Nice!
Glad to hear that.

kharar, 2022-01-05:
Which just means KiRi is NOT supposed to be run from within its folder... weird behavior but very glad we got that settled. Thanks!

Now that the thing is running I just quickly checked it out but it seems the diff window stays empty (the big middle window that is supposed to show a graphical representation of the selected commits)

leoheck, 2022-01-05:
It runs from its own folder. You are making some mistakes.

Now, since it is running, we can debug other things. This is a pretty hard setup, this is not a simple tool. It is an experiment that kind of works. 

Now, it does not work with Kicad-6 (or the 5.99). It works with Kicad 5.*
Which one do you have installed?

kharar, 2022-01-05:
This machine has only 5.1.10 installed :-)

leoheck, 2022-01-05:
Nice. This is a good start. 
Can you share the command line you are using to run it?

kharar, 2022-01-05:
I better give you all of it

```
kharar@M5:/mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board$ kiri 2S_serial_charging_board.kicad_pcb
Traceback (most recent call last):
  File "/home/kharar/.local/share/kiri/bin/kicad_version.py", line 47, in <module>
    pcbnew_version = pn.GetBuildVersion()
AttributeError: module 'pcbnew' has no attribute 'GetBuildVersion'

      kicad_pro_path = /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board/2S_serial_charging_board.kicad_pcb
           repo_path = /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board
   kicad_pro_subpath = .
           kicad_pro = 2S_serial_charging_board.kicad_pcb
           kicad_sch = 2S_serial_charging_board.sch
           kicad_pcb = 2S_serial_charging_board.kicad_pcb
       KICAD_VERSION =
   KICAD_PCB_VERSION = 20171130
  kicad_pro_filename = 2S_serial_charging_board
 kicad_pro_extension = kicad_pcb
 kicad_sch_extension = sch
 kicad_pcb_extension = kicad_pcb
          output_dir = /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board/kiri

Output folder: /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board/kiri

Generating artifacts
6 commits, totaling 30 comparisons

   1. 4f1ace5 - b924aa9
   2.         - d065127
   3.         - c410a99
   4.         - a695f85
   5.         - local

Assembling the HTML
- Adding commits list to the page
- Adding schematic pages list to the page
- Setting sch/pcb titles
- Setting sch/pcb revision
- Setting sch/pcb date
- Setting board name
- Setting initial commits
- Setting initial schematics
- Setting initial layouts
- Adding layers list to the page

Output folder: /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board/kiri
Starting web-server at http://127.0.0.1:8080
Hit Ctrl+C to end session
```

leoheck, 2022-01-05:
Ok. Do you have the Kicad installed?

kharar, 2022-01-05:
Only on the windows side...

leoheck, 2022-01-05:
so.. it wont work...

leoheck, 2022-01-05:
What if you run the install script?

leoheck, 2022-01-05:
This thing here
[image]

kharar, 2022-01-05:
that's what I did.. not sure if Kicad came along that way, will check...

kharar, 2022-01-05:
Well, issuing _pcbnew_ did give me this response:  `Error: Unable to initialize GTK+, is DISPLAY set properly?`

leoheck, 2022-01-05:
and this `kicad_version.py`, what is the output?

kharar, 2022-01-05:
So is **DOES** look like KiCAD is installed on the WSL side too, I'm just not able to run it from the terminal it seems

kharar, 2022-01-05:
```
$ kicad_version.py
Traceback (most recent call last):
  File "/home/kharar/.local/share/kiri/bin/kicad_version.py", line 47, in <module>
    pcbnew_version = pn.GetBuildVersion()
AttributeError: module 'pcbnew' has no attribute 'GetBuildVersion'
```

leoheck, 2022-01-05:
I would say that it is better to install 5.1.12 to make sure the python is working

leoheck, 2022-01-05:
This here
https://launchpad.net/~kicad/+archive/ubuntu/kicad-5.1-releases

```
sudo add-apt-repository ppa:kicad/kicad-5.1-releases
sudo apt-get update
sudo apt install kicad
```

kharar, 2022-01-05:
Thanks a lot
I'll try and have a go with that

leoheck, 2022-01-05:
To problem dude!

kharar, 2022-01-05:
Okay, Kicad is now installed:
```
kharar@M5:~$ kicad_version.py
5.1.12-84ad8e8a86~92~ubuntu20.04.1
```
but the main diff window is still blank
```
kharar@M5:~$ kiri

      kicad_pro_path = /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board/2S_serial_charging_board.pro
           repo_path = /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board
   kicad_pro_subpath = .
           kicad_pro = 2S_serial_charging_board.pro
           kicad_sch = 2S_serial_charging_board.sch
           kicad_pcb = 2S_serial_charging_board.kicad_pcb
       KICAD_VERSION = 5
   KICAD_PCB_VERSION = 20171130
  kicad_pro_filename = 2S_serial_charging_board
 kicad_pro_extension = pro
 kicad_sch_extension = sch
 kicad_pcb_extension = kicad_pcb
          output_dir = /home/kharar/kiri

Output folder: /home/kharar/kiri

Generating artifacts
6 commits, totaling 30 comparisons

Assembling the HTML
- Adding commits list to the page
- Adding schematic pages list to the page
- Setting sch/pcb titles
- Setting sch/pcb revision
- Setting sch/pcb date
- Setting board name
- Setting initial commits
- Setting initial schematics
- Setting initial layouts
- Adding layers list to the page

Output folder: /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board/kiri
Starting web-server at http://127.0.0.1:8080
Hit Ctrl+C to end session
```

kharar, 2022-01-05:
I'm running out of time for this session but I'll check back soon.
Can't wait to see this toolset in action :-D

kharar, 2022-01-05:
In the meantime here's a screenshot 
[image]
And just to rule any bugs in this particular project, I also get the same result with other projects.

leoheck, 2022-01-05:
Ok, this is a good start

Now you can run it like to see something more in the logs, as the commands that it is running.
```
kiri name_of_the_board.pro -V 
```
Seeing commands you woull be able to copy and past the command, to see  what each command is doing

leoheck, 2022-01-06:
kiri uses 2 commands. `plotgitsch` that generates plots for each schematic and `kicad-diff` (kidiff) that does the same for layout. if you pass the `-V` flag to kiri, it is going to show you the whole command line for each commit. You can copy these lines to run manually to see what is happening

leoheck, 2022-02-13:
Hi, checking if it is all good here. I am closing this issue right now. Please, reopen it if you still have this issue.

kharar, 2022-02-14:
I gave up, sorry.
I am not a software developer so I guess I'll have to wait for this project to mature.
But thanks for the effort anyway.

kharar, 2022-02-17:
Okay, you got me trying one more time.
Here's the repository (it's public already): https://github.com/kharar/2S_serial_charging_board
But please note that I have previously tried with several other repositories only too get the same result.

I had already cleaned up and removed kiri and dependencies from the system so I installed dependencies and then tool and finally I added the lines to ~/.bashrc according to your description above and made sure that all packages were updated and upgraded.
I ran the tool using the -V option but keep getting the same result: diff window is still blank.

Here is the terminal output:

```
kharar@M5:~$ kiri /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board/2S_serial_charging_board.pro -V

      kicad_pro_path = /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board/2S_serial_charging_board.pro
           repo_path = /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board
   kicad_pro_subpath = .
           kicad_pro = 2S_serial_charging_board.pro
           kicad_sch = 2S_serial_charging_board.sch
           kicad_pcb = 2S_serial_charging_board.kicad_pcb
       KICAD_VERSION = 5
   KICAD_PCB_VERSION = 20171130
  kicad_pro_filename = 2S_serial_charging_board
 kicad_pro_extension = pro
 kicad_sch_extension = sch
 kicad_pcb_extension = kicad_pcb
          output_dir = /home/kharar/kiri

Output folder: /home/kharar/kiri

        kicad_pcb: ./2S_serial_charging_board.kicad_pcb
        kicad_sch: 2S_serial_charging_board.sch
        sch_files: 2S_serial_charging_board.sch
       output_dir: /home/kharar/kiri
          web_dir: /home/kharar/kiri/web/

Getting commits:
kiri-git-log master -- 2S_serial_charging_board.sch ./2S_serial_charging_board.kicad_pcb

Commits list:
     1: a695f85 | 2020-10-13 02:03:51 +0200 | Martin Axelsen | add Casing stl
     2: c410a99 | 2020-10-11 23:52:40 +0200 | Martin Axelsen | Add Github web address in silkscreen
     3: d065127 | 2020-10-11 23:40:00 +0200 | Martin Axelsen | Fixes from 1.st prototype run
     4: b924aa9 | 2020-09-29 23:35:57 +0200 | Martin Axelsen | Fixed the XT60 silkscreen issue
     5: 4f1ace5 | 2020-09-06 01:15:15 +0200 | Martin Axelsen | Revision 1 (initial commit)

Generating artifacts
6 commits, totaling 30 comparisons

Assembling the HTML
- Adding commits list to the page
- Adding schematic pages list to the page
- Setting sch/pcb titles
- Setting sch/pcb revision
- Setting sch/pcb date
- Setting board name
- Setting initial commits
- Setting initial schematics
- Setting initial layouts
- Adding layers list to the page

Output folder: /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board/kiri
Starting web-server at http://127.0.0.1:8080
Hit Ctrl+C to end session
^C
** Trapped CTRL-C
Cleaning up and leaving
kharar@M5:~$ ls kiri
068a0a3  2ae1447  4f1ace5  7ee821a  995cbad  b924aa9  ca1a919  index.html   local
1381561  3fef7fd  799ee36  894343f  a695f85  c410a99  d065127  kiri-server  web
kharar@M5:~$
```

I also noticed that after running the tool the "kiri" work folder is not cleaned up even though the terminal ends with the text "Cleaning up and leaving"

leoheck, 2022-02-17:
Hey, @kharar thanks for testing it one more time. 
I could find some issues that were preventing generating schematics for your project. Your project helped me to improve other things too. 

Could you update your files to check if it works? I am not developing it on windows and I don't have one here with me to give it a try. If you have any issues please, let me know.

Please, upgrade submodules too 
```
git submodule update
```

Update your environment with this
```
# Kiri environment setup
eval $(opam env)

# Make sure the KIRI_HOME is setting the right path of the installation
export KIRI_HOME=${HOME}/.local/share/kiri

##
##  THIS LINE HAS A NEW "bin" folder in the END
##
export PATH=${KIRI_HOME}/submodules/KiCad-Diff/bin:${PATH}

export PATH=${KIRI_HOME}/bin:${PATH}
```

Here are some screenshots of your project.

Schematics
[image]

Layout
[image]

kharar, 2022-02-18:
Issued this:
```
kharar@M5:~/symlinks/KiCad/2S_serial_charging_board$ git submodule update
```
But there were no output from that...

My ~/.bashrc contains this at the very bottom:
```
# Kiri environment setup
eval $(opam env)
export KIRI_HOME=${HOME}/.local/share/kiri
export PATH=${KIRI_HOME}/submodules/KiCad-Diff/bin:${PATH}
export PATH=${KIRI_HOME}/bin:${PATH}
```

WSL terminal output:
```
kharar@M5:~$ kiri /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board/2S_serial_charging_board.pro -V

      kicad_pro_path = /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board/2S_serial_charging_board.pro
           repo_path = /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board
   kicad_pro_subpath = .
           kicad_pro = 2S_serial_charging_board.pro
           kicad_sch = 2S_serial_charging_board.sch
           kicad_pcb = 2S_serial_charging_board.kicad_pcb
       KICAD_VERSION = 5
   KICAD_PCB_VERSION = 20171130
  kicad_pro_filename = 2S_serial_charging_board
 kicad_pro_extension = pro
 kicad_sch_extension = sch
 kicad_pcb_extension = kicad_pcb
          output_dir = /home/kharar/kiri

Output folder: /home/kharar/kiri

        kicad_pcb: ./2S_serial_charging_board.kicad_pcb
        kicad_sch: 2S_serial_charging_board.sch
        sch_files: 2S_serial_charging_board.sch
       output_dir: /home/kharar/kiri
          web_dir: /home/kharar/kiri/web/

Getting commits:
kiri-git-log master -- 2S_serial_charging_board.sch ./2S_serial_charging_board.kicad_pcb

Commits list:
     1: a695f85 | 2020-10-13 02:03:51 +0200 | Martin Axelsen | add Casing stl
     2: c410a99 | 2020-10-11 23:52:40 +0200 | Martin Axelsen | Add Github web address in silkscreen
     3: d065127 | 2020-10-11 23:40:00 +0200 | Martin Axelsen | Fixes from 1.st prototype run
     4: b924aa9 | 2020-09-29 23:35:57 +0200 | Martin Axelsen | Fixed the XT60 silkscreen issue
     5: 4f1ace5 | 2020-09-06 01:15:15 +0200 | Martin Axelsen | Revision 1 (initial commit)

Generating artifacts
6 commits, totaling 30 comparisons

Assembling the HTML
- Adding commits list to the page
- Adding schematic pages list to the page
- Setting sch/pcb titles
- Setting sch/pcb revision
- Setting sch/pcb date
- Setting board name
- Setting initial commits
- Setting initial schematics
- Setting initial layouts
- Adding layers list to the page

Output folder: /mnt/c/Users/Martin/Documents/KiCad/2S_serial_charging_board/kiri
Starting web-server at http://127.0.0.1:8080
Hit Ctrl+C to end session
```

But the kiri inspector still shows blank as usual

kharar, 2022-02-21:
Just tried again, this time on a different machine running WSL2, but sadly I got the same result, plus an error message:

```
kharar@M6:~$ kiri /mnt/c/Users/Martin\ A/Documents/KiCad/2S_serial_charging_board/2S_serial_charging_board.pro -V
Traceback (most recent call last):
  File "/home/kharar/.local/share/kiri/bin/kicad_version.py", line 47, in <module>
    pcbnew_version = pn.GetBuildVersion()
AttributeError: module 'pcbnew' has no attribute 'GetBuildVersion'

      kicad_pro_path = /mnt/c/Users/Martin A/Documents/KiCad/2S_serial_charging_board/2S_serial_charging_board.pro
           repo_path = /mnt/c/Users/Martin A/Documents/KiCad/2S_serial_charging_board
   kicad_pro_subpath = .
           kicad_pro = 2S_serial_charging_board.pro
           kicad_sch = 2S_serial_charging_board.sch
           kicad_pcb = 2S_serial_charging_board.kicad_pcb
       KICAD_VERSION =
   KICAD_PCB_VERSION = 20171130
  kicad_pro_filename = 2S_serial_charging_board
 kicad_pro_extension = pro
 kicad_sch_extension = sch
 kicad_pcb_extension = kicad_pcb
          output_dir = /home/kharar/.kiri

Output folder: /home/kharar/.kiri

        kicad_pcb: ./2S_serial_charging_board.kicad_pcb
        kicad_sch: 2S_serial_charging_board.sch
        sch_files: 2S_serial_charging_board.sch
       output_dir: /home/kharar/.kiri
          web_dir: /home/kharar/.kiri/web/

Getting commits:
kiri-git-log master -- 2S_serial_charging_board.sch ./2S_serial_charging_board.kicad_pcb

Commits list:
     1: a695f85 | 2020-10-13 02:03:51 +0200 | Martin Axelsen | add Casing stl
     2: c410a99 | 2020-10-11 23:52:40 +0200 | Martin Axelsen | Add Github web address in silkscreen
     3: d065127 | 2020-10-11 23:40:00 +0200 | Martin Axelsen | Fixes from 1.st prototype run
     4: b924aa9 | 2020-09-29 23:35:57 +0200 | Martin Axelsen | Fixed the XT60 silkscreen issue
     5: 4f1ace5 | 2020-09-06 01:15:15 +0200 | Martin Axelsen | Revision 1 (initial commit)

Generating artifacts
6 commits, totaling 30 comparisons

Assembling the HTML
- Adding commits list to the page
- Adding schematic pages list to the page
- Setting sch/pcb titles
- Setting sch/pcb revision
- Setting sch/pcb date
- Setting board name
- Setting initial commits
- Setting initial schematics
- Setting initial layouts
- Adding layers list to the page

Output folder: /mnt/c/Users/Martin A/Documents/KiCad/2S_serial_charging_board/.kiri
Starting web-server at http://127.0.0.1:8080
Hit Ctrl+C to end session
```

After updating kicad to version 6.0.2 the error disappeared:
```
kharar@M6:~$ kiri /mnt/c/Users/Martin\ A/Documents/KiCad/2S_serial_charging_board/2S_serial_charging_board.pro -V

      kicad_pro_path = /mnt/c/Users/Martin A/Documents/KiCad/2S_serial_charging_board/2S_serial_charging_board.pro
           repo_path = /mnt/c/Users/Martin A/Documents/KiCad/2S_serial_charging_board
   kicad_pro_subpath = .
           kicad_pro = 2S_serial_charging_board.pro
           kicad_sch = 2S_serial_charging_board.sch
           kicad_pcb = 2S_serial_charging_board.kicad_pcb
       KICAD_VERSION = 6
   KICAD_PCB_VERSION = 20171130
  kicad_pro_filename = 2S_serial_charging_board
 kicad_pro_extension = pro
 kicad_sch_extension = sch
 kicad_pcb_extension = kicad_pcb
          output_dir = /home/kharar/.kiri

Output folder: /home/kharar/.kiri

        kicad_pcb: ./2S_serial_charging_board.kicad_pcb
        kicad_sch: 2S_serial_charging_board.sch
        sch_files: 2S_serial_charging_board.sch
       output_dir: /home/kharar/.kiri
          web_dir: /home/kharar/.kiri/web/

Getting commits:
kiri-git-log master -- 2S_serial_charging_board.sch ./2S_serial_charging_board.kicad_pcb

Commits list:
     1: a695f85 | 2020-10-13 02:03:51 +0200 | Martin Axelsen | add Casing stl
     2: c410a99 | 2020-10-11 23:52:40 +0200 | Martin Axelsen | Add Github web address in silkscreen
     3: d065127 | 2020-10-11 23:40:00 +0200 | Martin Axelsen | Fixes from 1.st prototype run
     4: b924aa9 | 2020-09-29 23:35:57 +0200 | Martin Axelsen | Fixed the XT60 silkscreen issue
     5: 4f1ace5 | 2020-09-06 01:15:15 +0200 | Martin Axelsen | Revision 1 (initial commit)

Generating artifacts
6 commits, totaling 30 comparisons

Assembling the HTML
- Adding commits list to the page
- Adding schematic pages list to the page
- Setting sch/pcb titles
- Setting sch/pcb revision
- Setting sch/pcb date
- Setting board name
- Setting initial commits
- Setting initial schematics
- Setting initial layouts
- Adding layers list to the page

Output folder: /mnt/c/Users/Martin A/Documents/KiCad/2S_serial_charging_board/.kiri
Starting web-server at http://127.0.0.1:8080
Hit Ctrl+C to end session
```
...but sadly, still no luck getting any content in the center window
[image]

leoheck, 2022-02-22:
The layout is there? Toggle it with `s`
Run it with -D to show the full debugging messages, please.

kharar, 2022-02-22:
No, layout is still the same as schematic, just like it's always been for me.

```
kharar@M6:~$ kiri /mnt/c/Users/Martin\ A/Documents/KiCad/2S_serial_charging_board/2S_serial_charging_board.pro -V -D

      kicad_pro_path = /mnt/c/Users/Martin A/Documents/KiCad/2S_serial_charging_board/2S_serial_charging_board.pro
           repo_path = /mnt/c/Users/Martin A/Documents/KiCad/2S_serial_charging_board
   kicad_pro_subpath = .
           kicad_pro = 2S_serial_charging_board.pro
           kicad_sch = 2S_serial_charging_board.sch
           kicad_pcb = 2S_serial_charging_board.kicad_pcb
       KICAD_VERSION = 6
   KICAD_PCB_VERSION = 20171130
  kicad_pro_filename = 2S_serial_charging_board
 kicad_pro_extension = pro
 kicad_sch_extension = sch
 kicad_pcb_extension = kicad_pcb
          output_dir = /home/kharar/.kiri

Output folder: /home/kharar/.kiri

        kicad_pcb: ./2S_serial_charging_board.kicad_pcb
        kicad_sch: 2S_serial_charging_board.sch
        sch_files: 2S_serial_charging_board.sch
       output_dir: /home/kharar/.kiri
          web_dir: /home/kharar/.kiri/web/

Getting commits:
kiri-git-log master -- 2S_serial_charging_board.sch ./2S_serial_charging_board.kicad_pcb

Commits list:
     1: a695f85 | 2020-10-13 02:03:51 +0200 | Martin Axelsen | add Casing stl
     2: c410a99 | 2020-10-11 23:52:40 +0200 | Martin Axelsen | Add Github web address in silkscreen
     3: d065127 | 2020-10-11 23:40:00 +0200 | Martin Axelsen | Fixes from 1.st prototype run
     4: b924aa9 | 2020-09-29 23:35:57 +0200 | Martin Axelsen | Fixed the XT60 silkscreen issue
     5: 4f1ace5 | 2020-09-06 01:15:15 +0200 | Martin Axelsen | Revision 1 (initial commit)

Generating artifacts
6 commits, totaling 30 comparisons

Assembling the HTML
- Adding commits list to the page
- Adding schematic pages list to the page
- Setting sch/pcb titles
- Setting sch/pcb revision
- Setting sch/pcb date
- Setting board name
- Setting initial commits
- Setting initial schematics
- Setting initial layouts
- Adding layers list to the page

Output folder: /mnt/c/Users/Martin A/Documents/KiCad/2S_serial_charging_board/.kiri
Starting web-server at http://127.0.0.1:8080
Hit Ctrl+C to end session
```

leoheck, 2022-02-22:
I have to prepare my windows to give it a try.
Meanwhile you can try to run

```
plotkicadsh # to check if schematic diff works

and

kidiff # to check is pcb diff works
```

if both works, then I will have to o find the issue on kiri for windows.

kharar, 2022-02-22:
```
kharar@M6:~$ plotkicadsh
plotkicadsh: command not found
```
What did I possibly do wrong? I searched for the file plotkicadsh but it does not exist.

Then I tried kidiff and it threw a lot of errors about rsvg-convert not being found
`/home/kharar/.local/share/kiri/submodules/KiCad-Diff/bin/kicad6_svg_fix: line 25: rsvg-convert: command not found`
so I installedit using
`sudo apt install librsvg2-bin`
I guess this one should have been included in the dependencies script, because after that was installed kidiff seems to work.

However something else is still bugging, because still no window content in kiri whatsoever

kharar, 2022-02-22:
I think maybe that is a spelling error:
> plotkicadsh # to check if schematic diff works

it probably should have been plotkicadsch, nonetheless:
```
kharar@M6:~$ plotkicadsch
kharar@M6:~$
```
it doesn't seem to do anything

leoheck, 2022-02-23:
Alright, it looks like Kiri is not working fine with Kicad-5 anymore (I still have to investigate this)
So, first of all, you should install Kicad-6
You can install it with these commands:

```
sudo add-apt-repository ppa:kicad/kicad-6.0-releases
sudo get update
sudo get upgrade
sudo apt purge kicad
sudo get install -y kicad
```

Run the install dependencies again, just to make just you have everything installed
```
bash -c "$(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_dependencies.sh)"
```

Then if the installation of the dependencies went well, you are going to have `opam` available.
This is important to build and install `plotkicadsch` which is performed with the second script `install_kiri`
```
➜  ~ which opam
/usr/bin/opam
```

Alright, now run the installation script again
```
bash -c "$(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_kiri.sh)"
```

In the end, you should have these tools

Copy and paste these commands on the terminal, to check if they are available.
```
which plotkicadsch
which kidiff
which kiri
```

This is the expected output
```
➜  ~ which plotkicadsch
/home/lheck/.opam/4.09.1/bin/plotkicadsch
➜  ~ which kidiff
/home/lheck/.local/share/kiri/submodules/KiCad-Diff/bin/kidiff
➜  ~ which kiri
/home/lheck/.local/share/kiri/bin/kiri
```

And if everything went well, you will be able to see this
[image]

If you see this but the browser does not launch automatically. Just open one and go to this page here `http://127.0.0.1:8080` 
```
Output folder: /home/lheck/2S_serial_charging_board/.kiri
Starting web-server at http://127.0.0.1:8080
Hit Ctrl+C to end session 
```

This was installed on a clean windows, so I think I could address all your issues this time.

kharar, 2022-02-23:
Had to replace `sudo get` with `sudo apt`

Got stuck at `➜  ~ which opam` what to do with that?

kharar, 2022-02-23:
hmmm. seems if I just type `which opam` it returns `/usr/bin/opam` ... seems legit enough..

kharar, 2022-02-23:
Had to replace `sudo get` with `sudo apt`
Then typing `which opam` returns `/usr/bin/opam`
Can also confirm installation
```
kharar@M6:~$ which plotkicadsch
/home/kharar/.opam/4.09.1/bin/plotkicadsch
kharar@M6:~$ which kidiff
/home/kharar/.local/share/kiri/submodules/KiCad-Diff/bin/kidiff
kharar@M6:~$ which kiri
/home/kharar/.local/share/kiri/bin/kiri
```
My ~/.bashrc ends like this:
```
# Kiri environment setup
eval $(opam env)
# Make sure the KIRI_HOME is setting the right path of the installation
export KIRI_HOME=${HOME}/.local/share/kiri
export PATH=${KIRI_HOME}/submodules/KiCad-Diff/bin:${PATH}
export PATH=${KIRI_HOME}/bin:${PATH}
```
On running kiri the terminal goes like this:
```
kharar@M6:~$ kiri

      kicad_pro_path = /mnt/c/Users/Martin A/Documents/KiCad/2S_serial_charging_board/2S_serial_charging_board.pro
           repo_path = /mnt/c/Users/Martin A/Documents/KiCad/2S_serial_charging_board
   kicad_pro_subpath = .
           kicad_pro = 2S_serial_charging_board.pro
           kicad_sch = 2S_serial_charging_board.sch
           kicad_pcb = 2S_serial_charging_board.kicad_pcb
       KICAD_VERSION = 6
   KICAD_PCB_VERSION = 20171130
  kicad_pro_filename = 2S_serial_charging_board
 kicad_pro_extension = pro
 kicad_sch_extension = sch
 kicad_pcb_extension = kicad_pcb
          output_dir = /home/kharar/.kiri

Output folder: /home/kharar/.kiri

Generating artifacts
6 commits, totaling 30 comparisons

Assembling the HTML
- Adding commits list to the page
- Adding schematic pages list to the page
- Setting sch/pcb titles
- Setting sch/pcb revision
- Setting sch/pcb date
- Setting board name
- Setting initial commits
- Setting initial schematics
- Setting initial layouts
- Adding layers list to the page

Output folder: /mnt/c/Users/Martin A/Documents/KiCad/2S_serial_charging_board/.kiri
Starting web-server at http://127.0.0.1:8080
Hit Ctrl+C to end session
```
And the browser window comes up like it should, so far so good.. but still no cigar :-(

kharar, 2022-02-23:
Please notice I have migrated to the new computer which is running WSL2, this one was never running KiCad 5.. only version 6 so the version 5 problem can be excluded

leoheck, 2022-02-23:
Wierd. You may be doing something different.

Well, now, I fixed the small issue with Kicad-5, not working related to the missing `BuildVersion` function. Also, removed scour optimization since it was also having issues with Kicad-5 on Windows.

Now both Kicad-5 and 6 are working on Linux, Windows and macOS.

kharar, 2022-02-23:
Sounds good.
Hmmm, when you say Windows, I suppose you do still refer to WSL, which version is that?

> Wierd. You may be doing something different.
If I only knew...

leoheck, 2022-02-23:
Yes, this WSL, version 1. Since I am running on a virtualbox. And the WSL 2 was not working. But I feel they are the same.

leoheck, 2022-02-25:
Any luck? I push some changes, not related to your project. But you could try to check if it improves it for you.

kharar, 2022-02-25:
updated, upgraded, reinstalled dependencies, reinstalled kiri, still nothing

leoheck, 2022-02-25:
what is the output of the command `env` in your system?

kharar, 2022-02-25:
```
kharar@M6:~$ env
SHELL=/bin/bash
CAML_LD_LIBRARY_PATH=/home/kharar/.opam/4.09.1/lib/stublibs:/home/kharar/.opam/4.09.1/lib/ocaml/stublibs:/home/kharar/.opam/4.09.1/lib/ocaml
KIRI_HOME=/home/kharar/.local/share/kiri
OCAML_TOPLEVEL_PATH=/home/kharar/.opam/4.09.1/lib/toplevel
WSL_DISTRO_NAME=Ubuntu
PKG_CONFIG_PATH=/home/kharar/.opam/4.09.1/lib/pkgconfig:/home/kharar/.opam/4.09.1/lib/pkgconfig
NAME=M6
PWD=/home/kharar
LOGNAME=kharar
MANPATH=:/home/kharar/.opam/4.09.1/man:/home/kharar/.opam/4.09.1/man
OPAM_SWITCH_PREFIX=/home/kharar/.opam/4.09.1
MOTD_SHOWN=update-motd
HOME=/home/kharar
LANG=C.UTF-8
WSL_INTEROP=/run/WSL/7278_interop
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
WAYLAND_DISPLAY=wayland-0
LESSCLOSE=/usr/bin/lesspipe %s %s
TERM=xterm-256color
LESSOPEN=| /usr/bin/lesspipe %s
USER=kharar
DISPLAY=:0
SHLVL=1
BASH_ENV=/etc/bash.bashrc
XDG_RUNTIME_DIR=/mnt/wslg/runtime-dir
WSLENV=BASH_ENV/u
XDG_DATA_DIRS=/usr/local/share:/usr/share:/var/lib/snapd/desktop
PATH=/home/kharar/.opam/4.09.1/bin:/home/kharar/.local/share/kiri/bin:/home/kharar/.local/share/kiri/submodules/KiCad-Diff/bin:/home/kharar/.opam/4.09.1/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/lib/wsl/lib:/mnt/c/Program Files (x86)/Intel/iCLS Client/:/mnt/c/Program Files/Intel/iCLS Client/:/mnt/c/Windows/system32:/mnt/c/Windows:/mnt/c/Windows/System32/Wbem:/mnt/c/Windows/System32/WindowsPowerShell/v1.0/:/mnt/c/Program Files (x86)/Intel/Intel(R) Management Engine Components/DAL:/mnt/c/Program Files/Intel/Intel(R) Management Engine Components/DAL:/mnt/c/Program Files (x86)/Intel/Intel(R) Management Engine Components/IPT:/mnt/c/Program Files/Intel/Intel(R) Management Engine Components/IPT:/mnt/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/mnt/c/WINDOWS/System32/OpenSSH/:/mnt/c/Program Files/PuTTY/:/mnt/c/ProgramData/chocolatey/bin:/mnt/c/Program Files/NVIDIA Corporation/NVIDIA NvDLISR:/mnt/c/WINDOWS/system32:/mnt/c/WINDOWS:/mnt/c/WINDOWS/System32/Wbem:/mnt/c/WINDOWS/System32/WindowsPowerShell/v1.0/:/mnt/c/WINDOWS/System32/OpenSSH/:/mnt/c/Program Files/PowerShell/7/:/mnt/c/Users/Martin A/AppData/Local/Microsoft/WindowsApps:/mnt/c/Users/Martin A/AppData/Local/gitkraken/bin:/snap/bin
OPAMNOENVNOTICE=true
HOSTTYPE=x86_64
PULSE_SERVER=/mnt/wslg/PulseServer
OLDPWD=/mnt/c/Windows/System32
_=/usr/bin/env
```

leoheck, 2022-02-25:
Everything looks good. You should have a working plotkicadsh and also
working kidiff available. They both should be good with a board made with
Kicad 5 as I could see running on a VM with Windows.

kharar, 2022-02-25:
I just wish there was some way I could clone my system and send it to you :-D

leoheck, 2022-02-26:
Maybe there is a way to create a snapshot of the important things.
```
env > env
tar cvzf $USER-$(date "+%Y%m%d-%H%M%S").tgz ~/.local/share/kiri/ ~/.bashrc ~/.zshrc env
```

It is going to generate a file called `kharar-xxxxxxx-xxxxxx.tgz`
Put it somewhere online and share a link with me here. 
I may be able to simulate your environment, I guess.

designer2k2, 2022-02-26:
Yes i am able to run it, but so far im unable to generate a usable output...  
There is a kind of wild mix between KiCad 5/6 on my PC and within the Projects, im still sorting it out before i want to report a issue.

If somebody wants to give it a try: https://github.com/designer2k2/BerlinUhr  i can only get 1 pcb to show, but not correctly.

[2 more replies]

closed 2022-05-16 by leoheck