leoheck/kiri #33: Schematic diffs not output

metropt, 2022-01-20:
When I run kiri there is no schematic diffs. During the setup I notice that now there is no `./install.sh` on `cd kiri/submodules/plotkicadsch`

When running kiri I got:
```
Output folder: /home/pc/development/kiri/kiri
/home/pc/development/kiri/bin/kiri: line 1002: receiver.sch: command not found
/home/pc/development/kiri/bin/kiri: line 1003: remote-hw.sch: command not found
```

hkennyv, 2022-02-06:
Hey metropt@, I think you can use the following to install `plotkicadsch`. I'm getting the same issue as you though, the schematic diffs are not shown.

```
opam install plotkicadsch
```

This is the output from my chromedevtools, it looks like some things are missing:

```
Failed to load resource: the server responded with a status of 404 (File not found)
sch-test.svg:1 Failed to load resource: the server responded with a status of 404 (File not found)
test-00-F_Cu.svg:1 Failed to load resource: the server responded with a status of 404 (File not found)
test-00-F_Cu.svg:1 Failed to load resource: the server responded with a status of 404 (File not found)
kiri.js:778 Image Error (missing or problematic) = ../local/sch-test.svg
kiri.js:778 Image Error (missing or problematic) = ../0048ed1/sch-test.svg
kiri.js:778 Image Error (missing or problematic) = ../local/test-00-F_Cu.svg
kiri.js:778 Image Error (missing or problematic) = ../0048ed1/test-00-F_Cu.svg
```

leoheck, 2022-02-07:
Guys, sorry I am not with my computer to help you better. Previously I was using a fork of plotkicadsh. The I reverted it to the original one, since I prefer to use the original version of it but I forgot that I had this install script made by myself. You could try to download tbe install script from my fork and place it in the same folder it is there trying to install it again. Or, instead, clone my fork instead of the original one. I will have to fix this, but it will take some days until I have a computer with me.

leoheck, 2022-02-07:
Here is the url of my fork.
https://github.com/leoheck/plotkicadsch

leoheck, 2022-02-07:
This is the install.sh script. Adding it to tbe plotkicadsh folder may help. If this is the issue you are facing. 
https://github.com/leoheck/plotkicadsch/blob/master/install.sh

leoheck, 2022-02-07:
Now, if you are using Kicad 6. This won't work since plotkicadsh is not working for Kicad 6 yet. Maybe some thumbs up on the issue I have on plotkicadsh repo could help us to ask the author to speed up his development.

leoheck, 2022-02-07:
Ah, actually the issue was not created by me. Here it is.
https://github.com/jnavila/plotkicadsch/issues/40

hkennyv, 2022-02-07:
@leoheck wow, thanks for the response and for maintaining these package(s), great work, really. I'll try with your fork tomorrow and let you know how it goes!

hkennyv, 2022-02-08:
Still having some issues, I'm testing with this repository here, I'm testing with the sample repository from the [plotkicadsch guide](https://jnavila.github.io/plotkicadsch/plotgitsch_usersguide.html#_set_up)

```
khuynh@kpc:kicad_playground/schematic ‹master*›$ ll
total 52
-rw-r--r-- 1 khuynh khuynh  1955 Feb  6 15:35 LEDs.sch
-rw-r--r-- 1 khuynh khuynh   416 Feb  7 21:56 from_LEDs.svg
-rw-r--r-- 1 khuynh khuynh   416 Feb  7 21:56 from_kicad_playground.svg
-rw-r--r-- 1 khuynh khuynh  2571 Feb  6 15:35 kicad_playground-cache.lib
-rw-r--r-- 1 khuynh khuynh    51 Feb  7 21:56 kicad_playground.kicad_pcb
-rw-r--r-- 1 khuynh khuynh   688 Feb  6 15:35 kicad_playground.pro
-rw-r--r-- 1 khuynh khuynh  2850 Feb  6 15:35 kicad_playground.sch
drwxr-xr-x 9 khuynh khuynh  4096 Feb  7 21:56 kiri
-rw-r--r-- 1 khuynh khuynh  7653 Feb  7 21:56 to_LEDs.svg
-rw-r--r-- 1 khuynh khuynh 10303 Feb  7 21:56 to_kicad_playground.svg
```

```
khuynh@kpc:kicad_playground/schematic ‹master*›$ kiri --all --skip-cache --remove --debug --debug-sch --debug-pcb --verbose kicad_playground.pro
head: cannot open 'kicad_playground.kicad_pcb' for reading: No such file or directory

      kicad_pro_path = /home/khuynh/rejoule/kicad_playground/schematic/kicad_playground.pro
           repo_path = /home/khuynh/rejoule/kicad_playground
   kicad_pro_subpath = schematic
           kicad_pro = kicad_playground.pro
           kicad_sch = kicad_playground.sch
           kicad_pcb = kicad_playground.kicad_pcb
       KICAD_VERSION = 5
   KICAD_PCB_VERSION =
  kicad_pro_filename = kicad_playground
 kicad_pro_extension = pro
 kicad_sch_extension = sch
 kicad_pcb_extension = kicad_pcb
          output_dir = /home/khuynh/rejoule/kicad_playground/schematic/kiri

Output folder: /home/khuynh/rejoule/kicad_playground/schematic/kiri
Removing leftovers from the previous run

        kicad_pcb: schematic/kicad_playground.kicad_pcb
        kicad_sch: kicad_playground.sch
        sch_files: schematic/LEDs.sch
schematic/kicad_playground.sch
       output_dir: /home/khuynh/rejoule/kicad_playground/schematic/kiri
          web_dir: /home/khuynh/rejoule/kicad_playground/schematic/kiri/web/

kiri-git-log master -- schematic

Commits list:
     1: cdb8005 | 2018-10-17 22:24:22 +0200 | Jean-Noël Avila | Add a subsheet
     2: 064acea | 2018-10-17 22:18:43 +0200 | Jean-Noël Avila | Fix Led direction
     3: fc6818f | 2018-10-17 22:17:10 +0200 | Jean-Noël Avila | Small update to schematic
     4: 8b0696a | 2018-10-17 22:10:27 +0200 | Jean-Noël Avila | Draft of the design
     5: c0a3b8e | 2018-10-17 21:57:38 +0200 | Jean-Noël Avila | First commit

Generating artifacts
6 commits, totaling 30 comparisons

   1. c0a3b8e - 8b0696a
   plotgitsch -r schematic  -k -m -c FF0000:008FFF:FFFFFF:222222 c0a3b8e 8b0696a
use compare (ImageMagick) between bitmaps between Git rev c0a3b8e and Git rev 8b0696a
/home/khuynh/.opam/4.09.1/bin/git-imgdiff: line 3: montage: command not found
/home/khuynh/.opam/4.09.1/bin/git-imgdiff: line 3: display: command not found
rm: cannot remove '/tmp/tmp.Pihtz9j9pi': No such file or directory

Schematic: /home/khuynh/rejoule/kicad_playground/schematic/kiri/c0a3b8e/kicad_playground.sch
Pages:
     1: kicad_playground

Schematic: /home/khuynh/rejoule/kicad_playground/schematic/kiri/8b0696a/kicad_playground.sch
Pages:
[... 139 lines cut ...]

   kidiff kicad_playground.kicad_pcb --webserver-disable -b c0a3b8e -a kicad_playground.kicad_pcb -o "/home/khuynh/rejoule/kicad_playground/schematic/kiri"
Kicad file kicad_playground.kicad_pcb does not exit
   Status: 1
sed: can't read /home/khuynh/rejoule/kicad_playground/schematic/kiri/local/kicad_playground.kicad_pcb: No such file or directory

Board: /home/khuynh/rejoule/kicad_playground/schematic/kiri/local/kicad_playground.kicad_pcb
Used layers:

Assembling the HTML
- Adding commits list to the page
- Adding schematic pages list to the page
- Setting sch/pcb titles
- Setting sch/pcb revision
- Setting sch/pcb date
- Setting board name
- Setting initial commits
- Setting initial schematics
- Setting initial layouts
- Adding layers list to the page

Output folder: /home/khuynh/rejoule/kicad_playground/schematic/kiri
Starting web-server at http://127.0.0.1:8080
Hit Ctrl+C to end session
```

metropt, 2022-02-08:
With the script @leoheck provided I hadn't any problems.

leoheck, 2022-02-13:
Hi. Checking if everything is good here. If yes, I will close this issue. 

I have extra information that may be related. I am not generating `blank.svg` images when there is no image anymore. I was doing this because the browser displays something weird when the image is missing and that was the easiest thing for me to implement. Now, I am hiding the image with javascript instead of displaying a broken image. This is why you will see these Image errors from now on. I also don't like this, but I could not find a way to hide this message.

Second thing. *Kiri does not generate Schematic Diffs for Kicad 6 yet.* This is done by `plotkicadsh` which does not support it yet.

hkennyv, 2022-02-23:
Thanks @leoheck ! I haven't had a chance to double check yet, but if it worked for metropt, it's likely something with my WSL :)

leoheck, 2022-02-23:
I am finishing the setup of a windows machine here to check if the tool has any issues. I thought this was done. But you can open this again if the issue persists.

leoheck, 2022-02-25:
Hey, any luck here?

I would say that you guys could upgrade it again to check if it is better for you. I fixed some issues that prevented Kiri from displaying layout-related images when the board was made with Kicad 6. 

If it is working, then I will ask you to close this issue.

leoheck, 2022-05-16:
I am closing this. There is a new refactored version of Kiri. I would recommend you give it a try.

closed 2022-05-16 by leoheck