leoheck/kiri #34: using alternative port does not work, hardcoded server check on 8080

designer2k2, 2022-02-25:
Hello,
i need to use a different port as the default 8080, with the -p option i changed it to 8081.  
But that results in a blocked surface due to the function check_server_status() beeing hardcoded to 8080:
https://github.com/leoheck/kiri/blob/5ea326384a34d1fb4c124541a75dde9657626fa2/assets/kiri.js#L799

Browser console:
[image]

leoheck, 2022-02-25:
Ah, good catch. I will fix this. Which Operating System are you using?

designer2k2, 2022-02-25:
Windows 10 with the WSL setup as instructed here

leoheck, 2022-02-25:
Hey buddy, I pushed some fixes. Could you try to run it again? The fix is not perfect but it may work for your case.

leoheck, 2022-02-25:
If it works, please, close this issue then.

designer2k2, 2022-02-26:
It works now! 
Thanks for the quick fix.

closed 2022-02-26 by designer2k2

leoheck, 2022-02-26:
Nice! Awesome.