leoheck/kiri #37: Project with mixed Kicad 5 and 6 version

leoheck, 2022-02-26:
This issue was reported here, https://github.com/leoheck/kiri/issues/32#issuecomment-1052163945

Hey @designer2k2, thanks for sharing this project https://github.com/designer2k2/BerlinUhr as a use case 

I can see this project is not well organized which is pretty complex for Kiri, the way it is designed today.

First of all, you have files of both versions of Kicad 5 and 6 together. This is not good. You should decide which version you want to keep. To have Kiri working fully, it is important to keep the project on Kicad 5 since `plotgitsch` (the schematic exporter) does not work with the new `.kicad_sch` files. Now, if you don't care much about being able to review schematics visually, then Kicad 6 is good to go however keeping both versions in the same repo, in the same commit, is pretty bad for Kiri.

Also, to keep both versions of the files you have to work on both versions of the project having both versions of the Kicad to keep compatibility.

===

Now, running your project, I am seeing that there is always a commit that is empty (no image) between 2 commits. I don't know what is happening and why, but it feels like you are removing the board file every odd or even commit. I still have to understand what is going on here.

If you navigate up and down the commit list with one commit between 2 commits selected, then you can see boards changes.
[image]
 
Schematics suffer from the same problem.

===

Another problem with Kicad 6, is that it generates not pretty good svgs, at least until the current version 6.0.2
See, this big ball here. It is not there on your design but It is a bad-formed svg file. 
I am already applying some fixes on the exported svgs. But this one I was not able to fix yet.
[image]

leoheck, 2022-02-26:
See, the latest commits mix both versions

```
6a0cb708b89303fa836f7e2c907eb223e195275d
Berlin_Uhr.kicad_pcb
Berlin_Uhr.kicad_pro
Berlin_Uhr.kicad_sch
Berlin_Uhr.pro
Berlin_Uhr.sch

a346bc2eff37a41ccc810220932bcde0369e2b95
Berlin_Uhr.kicad_pcb
Berlin_Uhr.kicad_pro
Berlin_Uhr.kicad_sch
Berlin_Uhr.pro
Berlin_Uhr.sch

fd4f0a452a508d4cee06977a20ac8c99c24ae060
Berlin_Uhr.kicad_pcb
Berlin_Uhr.kicad_pro
Berlin_Uhr.kicad_sch
Berlin_Uhr.pro
Berlin_Uhr.sch

936190cf2b5dadab6f0e65fcbf568795d3d87c9d
Berlin_Uhr.kicad_pcb
Berlin_Uhr.kicad_pro
Berlin_Uhr.kicad_sch
Berlin_Uhr.pro
Berlin_Uhr.sch

0334b5a8f34cac937827de80c8c1c95c03252770
Berlin_Uhr.kicad_pcb
Berlin_Uhr.kicad_pro
Berlin_Uhr.kicad_sch
Berlin_Uhr.pro
Berlin_Uhr.sch

4a4ecb2e70e1b779c2a12f97b3326cef312fda88
Berlin_Uhr.kicad_pcb
Berlin_Uhr.kicad_pro
Berlin_Uhr.kicad_sch
Berlin_Uhr.pro
Berlin_Uhr.sch

b25263157f75f2942e12b31bb7bc9ac6dd772654
Berlin_Uhr.kicad_pcb
Berlin_Uhr.kicad_pro
Berlin_Uhr.kicad_sch
Berlin_Uhr.pro
Berlin_Uhr.sch

c828a02968ca5b66421da52a999f66b671c28647
Berlin_Uhr.kicad_pcb
Berlin_Uhr.kicad_pro
Berlin_Uhr.kicad_sch
Berlin_Uhr.pro
Berlin_Uhr.sch
```

You can generate this output with this code here
```
branch=main
for commit in $(git rev-list $branch)
do
	echo -e "\n$commit"
	git ls-tree --name-only -r $commit | grep ".*.pro\|*.kicad_pro\|.*.kicad_sch\|.*.sch\|.*.kicad_pcb"
done
```

leoheck, 2022-02-26:
Hm what do you mean by footprint error? You can easily fix this. Just import it to a custom lib (inside the project so it will be versioned too. Fix the issue in that particular footprint. Change the instance of that footprint to come from your custom lib and that's it.

leoheck, 2022-02-26:
> thanks for taking a look into this!
> I needed to switch mid-project to Version 6 as in 5 was a Footprint error.
> And it seems that i should have cleaned up the project on the switch...
> Why there are some empty commits in between i dont know.
>
> Actually im impressed by how well it handles the complex Silkscreen here!
>
> The first commits work, until i made the Version jump:
> [image: grafik]
> <https://user-images.githubusercontent.com/1591573/155851209-da6d3694-6427-4767-a857-9740b4b3f123.png>

Ah also thank you for sharing your project. This is an interesting use case
to help me find other issues with the tool.

leoheck, 2022-02-26:
At least if you run with -D flag, it shows less issues
```
kiri Berlin_Uhr.pro -D -r
```

leoheck, 2022-02-26:
I think I could improve a bit those big balls.

[image]

leoheck, 2022-04-02:
Hey, @designer2k2 I am creating a new version of Kiri, and it is working better (branch `refactoring`). I want to upgrade some images in the readme. Do you mind if I use your project https://github.com/designer2k2/BerlinUhr as the project to take screenshots for Kiri? Your board is awesome.

designer2k2, 2022-04-02:
Hello, yes go ahead and use it!

closed 2022-05-16 by leoheck