leoheck/kiri #39: Workaround for Plotting Schematics on Kicad 6 with Plotgitsch

leoheck, 2022-03-09:
`Plotgitsch` just plots Kicad 5 schematics known by the extension `.sch`. 
Version 6 of Kicad uses a new format and extension `.kicad_sch`. 

Kicad 6, has code that can open `.kicad_sch` files and save `.sch` files. This was used during the early days of Kicad 6 when Kicad 5 was the latest released version.

So, the idea is to modify the Kicad 6 code, more specifically EEschema, to launch `.kicad_sch`  files as it already does from the command line, save/create an alternative version of the files using `.sch`  format, and exit.

With that, `plotgitsch` can be used to create a diff in the project. 

I can already save files manually. Now I still have to make this process automatic, without interaction. 

I would appreciate any help to understand Kicad's code since it is pretty complex.

leoheck, 2022-03-13:
The initial version is working. Put some instructions here.

This is the code
https://gitlab.com/leoheck/kicad/-/tree/kicad_sch2sch-6.0.2

Some notes are here
https://gitlab.com/leoheck/kicad/-/blob/kicad_sch2sch-6.0.2/NOTES.md

Some of the discussion can be found here
https://forum.kicad.info/t/solved-convert-the-new-kicad-sch-to-the-old-sch/34334

bit0fun, 2022-04-01:
Following the notes provided, after compiling the branch of kicad I run into these issues when trying to convert the schematic files:

*** BUG ***
In pixman_region32_init_rect: Invalid rectangle passed
Set a breakpoint on '_pixman_log_error' to debug

Can't provide the exact file used as it's for work, but when I get home I can test it with other files and see if I get the same issue. I did try with a number of files to see what happens, but I wonder if this has to do with hierarchal schematics.

leoheck, 2022-04-02:
Hey @bit0fun there is a new plot_kicad_sch script that handles kicad_sch files on linux using xdotool. You know this since we were talking in another thread. But I want to share here it is kind of working well with Kicad 6 now.

This is the branch
https://github.com/leoheck/kiri/tree/refactoring

bit0fun, 2022-04-02:
I'll take a look soon, haven't been able to test it yet. Appreciate it!

leoheck, 2022-04-02:
Yesterday you helped me to believe again it is good to plot using xdotool
It is also working for Wayland which is great.

This is a comparison with commits comparing Kicad 5 and Kicad 6.
This happens when the project is ported from Kicad 5 to Kicad 6
It shows the images are different, but it is still pretty good to compare changes.
 
[image]

I am still going to commit some changes after this post.

mundodisco8, 2022-05-10:
Hi, 

I know this is work in progress but I tried the refactoring branch, and I get these messages (one for each commit)

```bash
   Plotting 334c92f, /mnt/c/Users/joel.santos/Workspace/t200_controller/.kiri/334c92f/
   plot_kicad_sch Controller.kicad_sch
   | Error: Supported display servers: x11, wayland
```

I'm running

```bash
 kiri -o OLDCOMMIT -n NEWCOMMIT Controller.kicad_pro -r -D -V
```

I'm trying to diff a KiCad 6 project, using Ubuntu on WSL (in W11). The webserver launches, and a lot of info is populated (names of different sheets in the schematic, different commits to compare, names of layers in the PCB,...). The PCBs are visually diffed, but the schematic view is blank.

[image]

[image]

I had to manually install a couple of tools to reach this point in WSL (for example, I run Ubuntu 20.04 LTS in WSL, and I had to install python-is-python3 as the code calls `python` and not `python3` and Ubuntu doesn't like that. I also had to install Imagemagick as there's a script (`git-imgdiff`) that calls `montage` and `display` and they weren't present in my system.

leoheck, 2022-05-10:
This branch works for me on pure Ubuntu Linux. It is kind of finished work that is about to be merged, except I did not test it extensively on Windows and MacOs. Maybe something is missing as you already saw. As far as I understood, you are reporting missing dependencies, when running on Windows WSL. 

The list of missing dependencies is
- `python-is-pytohn3` (or a better way to call the required python version)
- `imagemagick` required for missing `montage` and `display`

Is this right or am I missing something?

mundodisco8, 2022-05-10:
That, but also that `plot_kicad_sch` is returning an error about not having a display server. I am not an expert, but I think that WSL doesn't have one, or at least doesn't have X11/Wayland. My WSL installation has the WSLg "add-on", so I can run graphic linux apps on WSL, but maybe they use a different display server, and that's why `plot_kicad_sch` is complaining?

BTW, not a 100%, but maybe KiCad 6 is also needed. On my computer (WSL/Ubuntu LTS) only v5 is installed by default, you have to manually add the v6 repo.

leoheck, 2022-05-10:
Kicad 6 should not be needed if the project was made with Kicad 5, but since you are commenting it may have a bug somewhere. 

On windows you must have a display server installed (I have something related in the Readme, I think) and you have to setup the DISPLAY environment variable accordingly.

bit0fun, 2022-05-11:
Been a while, but have some time to test some things out. 

I've still run into issues with it working, and have a potential fix for nested hierarchal schematics but I find that the program will hang when trying to create plots, as well as hang when getting the final layer count. 

Not sure what's up, it also could be my work computer being slow. 

Will post more details as I find them

bit0fun, 2022-05-11:
On a side note, also added support for installing dependencies with arch based systems, will make a PR later today

leoheck, 2022-05-11:
@bit0fun that's ok, but create a new issue if your issue does not relate to this one, please.

mundodisco8, 2022-05-12:
Hi guys,

My plan was to use it with KiCad 6 projects, because that's what I've been working with. The project started on KiCad 5 but I don't think I will need to go that much back in time.

I did some tests. As you mentioned that you tested it on pure Linux, I tried on my Raspberry Pi, only to realise that I don't have it connected to a monitor. As I connect over RDP, the window server and all that stuff works differently (sorry, I don't know much about the details), so that failed too.

So I doubled down on the WSL on Windows 11. As WSL2 has the whole graphics thing baked in, I thought I could make it work. What I ended up doing was installing Gnome completely (see [this gist](https://gist.github.com/Ta180m/e1471413f62e3ed94e72001d42e77e22)) which might be overkill, but kind of worked. I had to tweak a couple of things, though. I think I had to set the `DIPSLAY` and `XDG_SESSION_TYPE` vars by hand, if I recall correctly, and also I had an issue installing `ydotool` as it's not on Ubuntu 20.04 LTS (the one I had running on my WSL), so I had to add the 21.04 repos and install it from there.

With all that, I managed to get it working partially. I can see some activity (KiCad's Eeschema opens and some buttons are clicked (I guess ydotool in action?). I get the diffs of a bunch of commits and then I get an error and the script hangs.

In the first bunch of commits, it complains about the schematig having nested pages, but it appears to do some stuff. Is having nested pages a blocker?

After a bunch of commits, it tries to open the schematic and simply complains "Error loading schematic blabla" and exits KiCad and the script stays there, without progressing.

So, if the problem is I'm using KiCad 6, and that it chokes on multipage schematics, seeing that the owner of `Plotgitsch` has not had the time to implement KiCad 6 support, I think I will have to wait a bit until this tool matures a bit (not complaining, I think you did a wonderful job, the web interface is very slick!)

Thanks!

leoheck, 2022-05-12:
For Wayland I could implement an alternative with ydotool, but it was not working fine for me in the past.

bit0fun, 2022-05-12:
I am basically having the same issues as @mundodisco8 with the program hanging, but with xdotool

Still not entirely sure what is happening, but with the debug flag set while running kiri, I can see plotgitsch runs but hangs unless I kill the application sometimes. Not entirely sure why that is the case.

leoheck, 2022-05-12:
I could help you guys if you give me more info..

mundodisco8, 2022-05-13:
I can't fully share details as it includes info that I don't own (I'm reviewing a project)

When I think it's working (at least, the script progresses):

The terminal opens a commit, shows the pages in the schematic correctly and the layers in use. It displays

```
Plotting COMMIT_HASH, /path/to/project/.kiri/COMMIT_HASH
plot_kicad_sch schematic.kicad_sch 
```

And then this interaction

https://user-images.githubusercontent.com/8402707/168249509-88ddb61d-11d1-494b-b971-dd6510767d38.mp4

And when KiCad closes, kidiff is called in the console window. After 10-15s, a new commit is processed.

It does this for a bunch of commits, and then I get this:

Same output in the console (different commit, obviously)

[image]

And KiCad Exits and nothing else happens in the terminal

Please, don't spend much more time on this on my behalf, unless you are intrigued by the error. I think the key is achieving full support from either eeschema to output schematics as graphics (which apparently comes in KiCad v7) with python, or get `plotgitsch` to support KiCad 6.

leoheck, 2022-05-14:
Hey, @mundodisco8 I pushed an update of the `plot_kicad_pcb` script being more resilient to the popups that may appear when launching Kicad with the xdotool. It may be working better to generate the schematics for Kicad 6 projects. If it does not work, just relaunch kiri with the flag -d and it may work in the 2nd run.

mundodisco8, 2022-05-15:
I started from scratch, with a VM in HyperV with Ubuntu 22.04. The only problem I faced was that wxpython would refuse to build, and I had to run `apt install build-essential libgtk-3-dev`. wxWidgets took around 1h to build!!!!

Anyway, I ran kiri again, and this time it completed all the steps. The performance was much better than with WSL2 (disk access is still a tad slow in WSL2).

After your suggestion of running `plot_kicad_sch` by itself, I noticed that no svg version of any schmatic is exported. I checked the script and saw that you were asking xdotool in line 214 to press `Backspace`. Why is that? I tried on KiCad 6 and pressing backspace doesn't select the active radio button. After changing the line to `space` (without quotes, or it wouldn't work at all), it selects SVG (otherwise stays in my default option, HPGL). Two comments about this. KiCad (at least 6) remembers the last format used, so if I open the project, plot to SVG by hand, and then run your script, it will print "two options" above SVG, which is Postscript. I can't think of a way to detect the currently selected option, other than going all the way up, or down, selecting that and then pressing up (or down) twice: messy 😵‍💫 . The other thing is that after that, you press tab 4 times to select "plot in black and white", but again at least in 6, if you tab 4 times you move to "plot background color", which ignores your PgDn command. Then you press enter twice, the plots are generated, and then you tab 11 times, that gets you into the "info" checkbox. I think you want to move to the "save as.." button and the script waits for "plot_time" and then the window is killed (so the results are not saved to report.txt), that would be 12 tabs, not 11.

It also struggled when the drawing sheet was not found. It appears that some versions of the schematic have a problem in the path to the wks file that contains the drawing sheet. Your script fails there, understandably, as I imagine that you can't foresee all the potential problems a schematic might have.

Another issue I see randomly happen (although I can't reliably trigger it, is that sometimes instead of opening the plot dialog, it opens another dialog with a big text box (I looked around but couldn't find it) and starts to tab away in that big text box, instead of navigating through the plot menu. If I get the title of the window I will report (PS: it's text properties, so it fails to get to the file menu and starts to edit the properties of an item in the schematic??)

This is all in the old branch, I haven't checked out your last changes. With all that, I get svg outputs, but the web server only displays the main schematic page, and not any subschematics. Also, if I change the commits to review, the view gets blank, and I need to refresh the page (F5) on the browser, and then I get the diff between the selected comments. Sadly, refreshing also changes the setting to schematic view, so I can't see differences in the PCB.

As I said I really appreciate your efforts but I'm afraid this is not "production ready" (nor that you claim it is anywhere, so not blaming you here) enough for my needs. Definitely will track the progress because I think the idea is incredibly good and the output is super polished. So keep up the good work!

I attach my copy of the script in case it helps, as .txt so github accepts it.

[plot_kicad_sch.txt](https://github.com/leoheck/kiri/files/8695546/plot_kicad_sch.txt)

leoheck, 2022-05-15:
Can you summarize by highlighting important parts? I don't need to know the whole process. 

Also, what is the version of your Kicad?

Running it on Ubuntu, with X11 should work fine. Unless you have a complicated git repository.

Have you tested Kiri with another repo? I would start with another repo just to make sure it works.
You can find a repo online, but you can use one of mine as an example https://github.com/leoheck/assoc-board

leoheck, 2022-05-15:
Sometimes `xdotool` or my script `plot_kicad_sch` gets lost for an unknown reason. However, it fixes itself after running it a couple of times. 

If you identify a case where it is not working. Go inside that folder and rerun the script a couple of times until it generates an image.

For instance, it was not working for me here...
[image]

Then you can go inside that folder `/home/lheck/Documents/assoc-board/.kiri/5be11dd/` 
And run `plot_kicad_sch board.kicad_sch` until it works. 

When it is working... then you can go back to the root of your project and run `kiri` again. It may be ok this time. 

You can identify it is working when you see an output like this showing the generated .svg file and no red status
[image]

I also put this variable PLOTPCB_TIMEOUT to interrupt if it is taking to much time to fail.
The current default time is 5s, but if it is not enough on your machine you can change it like this, before calling `kiri` 
```
export PLOTPCB_TIMEOUT=10
```

leoheck, 2022-05-15:
It is hard to force this particular issue to happend on `eeschema` and `xdotool`. But I have improved the `plot_kicad_sch` a bit. It may work better for you, this time.

You are experiencing these issues more often or always because you are running inside a VM, so this may be the culprit.

Also, I added some variables to control better the events.

```bash
# This one was renamed and it is the timeout for the plot_kicad_sch, the default value is 15s (before it was 5s)
PLOTSCH_TIMEOUT=15

# This one is the time in seconds waiting for xdotool to detect when eeschema has the circuit loaded.
WAIT_FOR_EESCHEMA_S=0.5
```

leoheck, 2022-05-15:
@mundodisco8 
I feel that this is a good time for you to try the repo again. The code is in the main branch now. So move to that branch and upgrade your repo before anything else.

mundodisco8, 2022-05-24:
I tried again, and I must be doing something wrong, because on the latest commit, after reinstalling all dependencies and kiri itself, I get exactly the same behaviour described above. That is, for example, 
- SVG output is never selected, the cursor is moved there, but "space" is never pressed, so the schematic is plotted in HPGL
- From then on, all the "tabs" are wrong because the dialog options are totally different.

mundodisco8, 2022-05-25:
If you see my [previous comment](https://github.com/leoheck/kiri/issues/39#issuecomment-1126982873) I managed to make it reliably in my script. I changed the key pressed in configure_and_plot (line 222 of the plot_kicad_sch script that I attached as a txt as github doesn't allow attaching files without extension) from `"backspace"` to `space` (notice that `space` doesn't have quotes, if I leave the quotes, then it wouldn't trigger the press). I would also recommend adding 

```
xdotool key --repeat 4 --delay "${TIME}" "Down"
```
before line 217. That way, no matter what the currently selected option (based on the user's previous runs of the plot command) were, as it would go down 4 times to HPGL and then up twice to SVG.

With the modified script I sent I was able to reliable export schematics in black and white in SVG format and export the log for the plotting process.

You said that your script works, and that's something I don't understand **UNLESS THE DIALOG HAS CHANGED FROM KICAD 5 TO 6**.  This is the plot dialog in KiCad 6, when SVG is selected

[image]

1) In your script, in configure_and_plot(), line 206, you press tab, which takes you from "output directory" to "output format", then press up twice which, if in HPGL takes you to SVG, and then press backspace, which does nothing. 
1) Then the script has a comment that says "set colour mode to B/W", and then you press tab 4 times, which in SVG mode takes you to "plot background colour", not to "output mode". This is a checkbox, so pressing page_down does nothing (it would in output mode, as that is a dropdown).
1) Finally, you have a comment saying "Plot all pages" and you tab 11 times. From the position of the cursor, "plot background colour", 11 tabs leaves you in the "save..." button, that saves the output messages as a txt file. You want to end in "Plot all pages" which is 12 tabs away, not 11.

Check my previous videos, where you can see the actual behaviour in my computer, running the latest KiCad ([here](https://github.com/leoheck/kiri/issues/39#issuecomment-1125856044))

Now, all this might be running on your computer, I'm not saying it doesn't as obviously you wouldn't push it to github otherwise, but as you can see I'm providing you evidence about an issue and steps to reproduce it. 

And please don't take this the wrong way, I'm not pissing on your work, just trying to lend a hand.

leoheck, 2022-05-25:
The last used "Output Format" will be selected by default. 
Try to plot something in PDF, close the window, then open it again and the PDF will be selected.
The new HPGL seems to be the default if you have never plotted anything with Kicad 6.0.5

Now, if you use arrows (up/down) there is no way to move to the top/bottom of the list since it keeps cycling. 
Page-up/Page-down does not work here too, they don't change focus at all. 

So, there is no reliable way of doing this. At least as I could verify.

Something that is working for you may not work for me since there is no reliable way of doing this. 
The best thing here is to select and plot something with SVG first, and then after, use Kiri.

leoheck, 2022-05-25:
nevermind, I found a way to get this reliable, I am going to fix it.

Please, check for me, this file
`grep format /.config/kicad/6.0/eeschema.json`

if the format is 5, then it will open with SVG selected
change it manually to 4, and open the plot window to check if it changes to pdf in your kicad, please.

these are the current possible values
```
// format sequence
  ps - 2
 pdf - 4
 svg - 5
 dxf - 3
hpgl - 0
```

leoheck, 2022-05-25:
I've pushed this fix. It uses fewer xdotool commands which means it is more reliable and it also goes faster since it does not need to execute extra and not necessary commands. What I am doing is pre-configuring the plot window, then I am launching it to just plot the files.

leoheck, 2022-05-25:
Since the graphical plot is good I expanded the possibility of plotting Kicad 5 with it too.
If you have 5.0 files and if the graphical plot is good, then you can set the environment with this variable 
```
export FORCE_PLOTTING_GRAPHICALLY=1
```