leoheck/kiri #40: Broken branch for KiDiff submodule

bit0fun, 2022-04-01:
In commit 0dfbbd18cba4c5e3f32486eab059eee08c8f9158, The KiDiff submodule was updated, but that branch no longer exists. Can’t make a pull request this second to find a suitable branch at the moment hence making this issue, but it should be a simple fix

leoheck, 2022-04-01:
Thanks for reporting. 
Hey @bit0fun could you check if the submitted commit fixs this?

leoheck, 2022-04-01:
Nice, thanks. 

> Now to try and figure out more kicad 6 issues

Yeh, this is a bit hard, since Plotgitsch does not work for Kicad 6 and I could not find any other way to plot kicad_sch files yet. However, I am refactoring Kiri to be a bit easier to fix things. It may also work better in the end. This is the `refactoring` branch if you are interested. 

I hope Kicad team can release the Python API for schematics, to at least be able to export the schematic to SVG as it is possible for layouts.

bit0fun, 2022-04-01:
That definitely would be nice. And yeah I'll have to check out the refactoring branch

Though the same submodule issue is there, just as a heads up ;)

Was attempting to get the script you have for the workaround going for plotgitsch, but looks like I'm getting some errors there. Will post in the relevant directory

leoheck, 2022-04-01:
Ah, I think the alternate script is not working. I was testing it and keep there to make my life easier. It depends on xdootool and since I don't know how to use it well, it is not working. But if you know a bit of bash, you can help me to make it work. It should not need anything else to work. Just that script and the .kicad_pro/.kicad_sch as far as I remember.

leoheck, 2022-04-01:
> Though the same submodule issue is there, just as a heads up ;)

Where, on refactoring branch?

bit0fun, 2022-04-01:
> > Though the same submodule issue is there, just as a heads up ;)
> 
> Where, on refactoring branch?

Yup, just saw the refactoring branch has the broken commit as mentioned above

bit0fun, 2022-04-01:
> Ah, I think the alternate script is not working. I was testing it and keep there to make my life easier. It depends on xdootool and since I don't know how to use it well, it is not working. But if [...]

I'm definitely down to try and help out; and I do have a decent amount of experience with bash. 

Though feel like efforts to try and parse the schematics in the newer format would make more sense than to try and go after making the new stuff work like the old. Essentially it'll just create more baggage in the long run

leoheck, 2022-04-01:
Nice, I appreciate any possible help, to have an exported svg image from kicad_sch file. Any way is better than no way haha.

leoheck, 2022-04-01:
Hey, @bit0fun I pushed my latest changes on refactoring... it is plotting better now.. Please, update your version, and use it. This is almost ready to merge into the master. So, I would appreciate if you could use it to check if it has any issues. 

I will reopen this issue to make it easier to find.

bit0fun, 2022-04-01:
> Hey, @bit0fun I pushed my latest changes on refactoring... it is plotting better now.. Please, update your version, and use it. This is almost ready to merge into the master. So, I would appreciate [...]

No worries! 

And the commit you used for the refactoring branch is also showing issues where it isn't referenced anywhere. Did you have edits in the submodule but didn't push it to the repo itself?

bit0fun, 2022-04-01:
To be clear, the main branch commit for kicad-diff is fine, but the refactoring branch does not

leoheck, 2022-04-01:
Hmm, check then one more time, please

leoheck, 2022-04-01:
Now I have an initial version o sch2svg which was renamed to `plot_kicad_pcb`  that uses xdotool on X11.
It works, except the generated image is colored, not tuned for Kiri yet. But it is good. But it takes times.

leoheck, 2022-04-01:
Ah, would you like to test?

closed 2022-05-14 by leoheck