leoheck/kiri #42: Possibility of dragging / moving the graphic files?

mundodisco8, 2022-05-12:
I don't know if the diffs are displayed after being generated or if they are calculated on the fly, but it would be a great feature if the different versions can be moved around. This is because sometimes, you don't really any changes on a file, you just dragged everything to a new place, and the whole project is shows as different.

For PCBs, maybe the solution is not as much as plotting the diff, as plotting the appropriate gerbers in XOR mode, so the difference can be displayed on the fly? Same thing for schematics, if they are exported to a graphic format, maybe?

Maybe a lot to ask but something you could consider for the future

leoheck, 2022-05-12:
Diffs are generated on the fly, by overlapping 2 existing images. 

> This is because sometimes, you don't really any changes on a file, you just dragged everything to a new place, and the whole project is shows as different.

What does this mean?

> For PCBs, maybe the solution is not as much as plotting the diff, as plotting the appropriate gerbers in XOR mode

How do you do it? What is the difference? Do you have any pictures showing the difference to share here?

closed 2022-05-16 by leoheck

mundodisco8, 2022-05-24:
> This is because sometimes, you don't really any changes on a file, you just dragged everything to a new place, and the whole project is shows as different.
It looks like I couldn't speak the England that day. What I meant is that sometimes, you are working on your PCB (or schematic, for that matter, but I'll focus on the case of the PCB) and decide to move the board or relocate the origin of coordinates. You effectively haven't made any changes to the layout, but you loose the capability of diff'ing with past versions from that point onwards. Other use-case where this would happen: you set your gerbers to output the title block on certain layers (where one of the corners of the title block becomes your (0,0) position), but later decide to remove it from your layers as it shouldn't not belong there (for example, the copper layers). On that case, both gerbers have almost the same info (except for the title block) but they are offset, making visual diffs imposible.

About your second comment, I am not really sure about how your program works, but I suspect you use kidiff to generate an image with the comparison of two gerbers, right? Why don't you generate an image of each gerber, set their colours (like the ones chosen by kidiff, red and cyan, and give them some transparency. This way, overlapping areas are shown white (or grey, depending on the colours used) and non-overlapping areas appear in red or cyan. The improvement I'm requesting would be to be allowed to drag those images independently, so you can adjust your gerbers by hand if the automatic option of "aligning position (0,0) on both" is not good enough.

This would apply to schematics too. It would shift the way you operate from
1) I move to a commit
2) Generate intermediary outputs for the diffing tool
3) Feed them to my diffing tool (one for schematics, one for gerbers)
4) Store the diffed output
5) Display the diffed outputs in a nice web interface

to 
1) move to a commit
2) Generate outputs (svg, for example, or png)
3) Display the outputs in a nice web interface