leoheck/kiri #44: Images don't draw?

vfelsner, 2022-06-25:
Hi,
after getting the install to finish, I am seeing that the webserver starts up and I can see the webpage. It seems operational but I do not see the generated diff images. I **did see a flicker** of what may have been the images once but the image is missing.

[image]

I am on Mac OS X 10.14.6 with brew installed.

Here is my -D output of the kiri call:

> kiri -D Dimmerer.kicad_pro
> 
> INFO: kidiff is missing, skiping its use
> 
>  KIRI_HOME_PATH = /Applications/KiCad/Plugins/Kiri/kiri
>  KICAD_VERSION = 6.0.0-0
>  CURRENT_PATH = /Users/nose/git/Dimmerer/KiCAD
>  KICAD_PRO_ABSOLUTE_PATH = /Users/nose/git/Dimmerer/KiCAD/Dimmerer.kicad_pro
>  REPO_ABSOLUTE_PATH = /Users/nose/git/Dimmerer
>  KICAD_PROJ_NESTED_PATH = KiCAD
>  KICAD_PROJ_ABSOLUTE_PATH = /Users/nose/git/Dimmerer/KiCAD
>  OUTPUT_DIR_PATH = /Users/nose/git/Dimmerer/KiCAD/.kiri
>  WEB_DIR_PATH = /Users/nose/git/Dimmerer/KiCAD/.kiri/web/
> 
>  project_path: /Users/nose/git/Dimmerer/KiCAD
>  project: Dimmerer.kicad_pro
>  schematic: Dimmerer.kicad_sch
>  layout: Dimmerer.kicad_pcb
> 
> Get commits command
>    git log --date=format:'%Y-%m-%d %H:%M:%S' --pretty='format:%h | %ad | %an | %s' master -- \
>    Dimmerer.kicad_sch  Dimmerer.kicad_pcb
> 
> Info: KiCAD/Dimmerer.kicad_sch does not have changes
> Info: KiCAD/Dimmerer.kicad_pcb does not have changes
> At least one file has changes: false
> 
> Commits list
>      1	2625e09    2022-06-25 23:10:47    Valle    Working tree mods...
>      2	c893547    2022-02-16 21:36:28    Valle    Vishay PowerPAK Footprint korrigiert
>      3	d77dcd4    2022-01-18 06:46:32    Valle    Initial Restore nach GIT reinstall
> 
> Generating artifacts
> 3 commits, totaling 6 comparisons
> 
>      project_path: /Users/nose/git/Dimmerer/KiCAD
>           project: Dimmerer.kicad_pro
>         schematic: Dimmerer.kicad_sch
>            layout: Dimmerer.kicad_pcb
> 
> Assembling the HTML
> - Adding commits list in index.html
> - Adding schematic pages list in index.html
>   > INFO: Pages lists differ on both commits. What should we do?
> - Adding layers list in index.html
> - Project's title: Dimmerer
> - Project's name: Dimmerer
> - Sch Title: Dimmerer
> - PCB Title: [missing]
> - Sch Revision: V1.1
> - PCB Revision: [missing]
> - Sch Date: 2022-01-03
> - PCB Date: [missing]
> - Initial selected commits: 2625e09, c893547
> - Initial schematic images:
>   Sch 1: /Users/nose/git/Dimmerer/KiCAD/.kiri/2625e09/sch-Dimmerer.svg
>   Sch 2: /Users/nose/git/Dimmerer/KiCAD/.kiri/c893547/sch-Dimmerer.svg
> - Inicial layer: id=00, name=F_Cu 
> - Initial layer images:
>   PCB 1: /Users/nose/git/Dimmerer/KiCAD/.kiri/2625e09/Dimmerer-00.svg
>   PCB 2: /Users/nose/git/Dimmerer/KiCAD/.kiri/c893547/Dimmerer-00.svg
> 
> Output folder: /Users/nose/git/Dimmerer/KiCAD/.kiri
> 
> Starting webserver at http://127.0.0.1:8080/web/index.html
> (Hit Ctrl+C to exit)

vfelsner, 2022-06-25:
Update:
Okay it seems some prerequisites had failed to install. I could finish setting up KiCad-Diff using pip3 and finally managed to get kiri to draw the pictures **by plotting the schematics manually** in the .kiri/commit-id output folder. This was after eeschema was opened up by kiri and the script was telling me that the plot was currently undergoing. But nothing was happening here but kiri did recognize me plotting the schematics manually.

--> Stil a bit cumbersome this way but at least it finally works. Awesome tool, thanks!

vfelsner, 2022-06-26:
Update #2: 
Opening KiCad and eeschema beforehand seems to solve the manual plotting of schematics issue. That part seems to work now. But kiri still cannot open kidiff for plotting the pcb svg files.
I can run kidiff manually just fine and inspect the results via its webserver but kiri somehow doesn't seem to call kidiff (correctly)?

Here is my latest console ouput (schematic svgs already created):

<details>
<summary>Click</summary>

```
kiri -D

          KIRI_HOME_PATH = /Applications/KiCad/Plugins/kiri
           KICAD_VERSION = 6.0.0-0
            CURRENT_PATH = /Users/nose/git/Dimmerer/KiCAD
 KICAD_PRO_ABSOLUTE_PATH = /Users/nose/git/Dimmerer/KiCAD/Dimmerer.kicad_pro
      REPO_ABSOLUTE_PATH = /Users/nose/git/Dimmerer
  KICAD_PROJ_NESTED_PATH = KiCAD
KICAD_PROJ_ABSOLUTE_PATH = /Users/nose/git/Dimmerer/KiCAD
         OUTPUT_DIR_PATH = /Users/nose/git/Dimmerer/KiCAD/.kiri
            WEB_DIR_PATH = /Users/nose/git/Dimmerer/KiCAD/.kiri/web/

     project_path: /Users/nose/git/Dimmerer/KiCAD
          project: Dimmerer.kicad_pro
        schematic: Dimmerer.kicad_sch
           layout: Dimmerer.kicad_pcb

Get commits command
   git log --date=format:'%Y-%m-%d %H:%M:%S' --pretty='format:%h | \
   %ad | %an | %s' master -- Dimmerer.kicad_sch  Dimmerer.kicad_pcb

Info: KiCAD/Dimmerer.kicad_sch does not have changes
Info: KiCAD/Dimmerer.kicad_pcb does not have changes
At least one file has changes: false

Commits list
     1	46d6100    2022-06-25 23:10:47    Valle    Pulldown R1 für Q10 hinzug
     2	c893547    2022-02-16 21:36:28    Valle    Vishay PowerPAK Footprint 
     3	d77dcd4    2022-01-18 06:46:32    Valle    Initial Restore nach GIT r

Generating artifacts
3 commits, totaling 6 comparisons

1/6 46d6100 | 2022-06-25 23:10:47 | Valle | Pulldown R1 für Q10 hinzugefügt 
2/6 c893547 | 2022-02-16 21:36:28 | Valle | Vishay PowerPAK Footprint korrig

Retrieving commits:
- Source path: /Users/nose/git/Dimmerer/KiCAD
- Output path: /Users/nose/git/Dimmerer/KiCAD/.kiri/46d6100
- Exporting: KiCAD/Dimmerer.kicad_pcb
- Exporting: KiCAD/Dimmerer.kicad_pro
- Exporting: KiCAD/Dimmerer.kicad_sch

Retrieving commits:
- Source path: /Users/nose/git/Dimmerer/KiCAD
- Output path: /Users/nose/git/Dimmerer/KiCAD/.kiri/c893547
- Exporting: KiCAD/Dimmerer.kicad_pcb
- Exporting: KiCAD/Dimmerer.kicad_pro
- Exporting: KiCAD/Dimmerer.kicad_sch

     project_path: /Users/nose/git/Dimmerer/KiCAD/.kiri/46d6100
          project: Dimmerer.kicad_pro
        schematic: Dimmerer.kicad_sch
           layout: Dimmerer.kicad_pcb

     project_path: /Users/nose/git/Dimmerer/KiCAD/.kiri/c893547
          project: Dimmerer.kicad_pro
        schematic: Dimmerer.kicad_sch
           layout: Dimmerer.kicad_pcb

[... 569 lines cut ...]
  > INFO: Layers lists differ on both commits. What should we do?
- Project's title: Dimmerer
- Project's name: Dimmerer
- Sch Title: Dimmerer
- PCB Title: [missing]
- Sch Revision: V1.1
- PCB Revision: [missing]
- Sch Date: 2022-01-03
- PCB Date: [missing]
- Initial selected commits: 46d6100, c893547
- Initial schematic images:
  Sch 1: /Users/nose/git/Dimmerer/KiCAD/.kiri/46d6100/sch-Dimmerer.svg
  Sch 2: /Users/nose/git/Dimmerer/KiCAD/.kiri/c893547/sch-Dimmerer.svg
- Inicial layer: id=00, name=F_Cu 
- Initial layer images:
  PCB 1: /Users/nose/git/Dimmerer/KiCAD/.kiri/46d6100/Dimmerer-00.svg
  PCB 2: /Users/nose/git/Dimmerer/KiCAD/.kiri/c893547/Dimmerer-00.svg

Output folder: /Users/nose/git/Dimmerer/KiCAD/.kiri

Starting webserver at http://127.0.0.1:8080/web/index.html
(Hit Ctrl+C to exit)
```

</details>

vfelsner, 2022-06-26:
Okay, issue solved. I have KiCad v5 and v6 installed and the scripts had the install locations hardcoded so it wouldn't necessarily find kicad v6 and therefore default back to running python 2.

Changing this script to ignore the KiCad v5 installation solved it:
[image]

Will leave this thread here for future reference, in case someone else is also struggling on Mac and KiCad v5 + v6 dual install. I should probably remove v5 ...

leoheck, 2022-06-26:
Hey @vfelsner thanks for this report. I would like to have kiri working with any version of Kicad available, but it looks like you have found this small bug where the project was created with Kicad 6 but the Kicad 5 was being selected. I will check how to improve this issue by selecting the right version of Kicad, or the latest if both are installed. I am going to keep this open while I think something.

leoheck, 2022-06-26:
Also, Kicad website shows they have version 6.0.6 available already. From your first post, it shows you have Kicad 6.0.0. I also recommend you to install at least 6.0.5 they have some improvements on generated SVG files.

leoheck, 2022-06-26:
I am installing Kicad 5 (5.1.9) after having Kicad 6 (6.0.5) installed but both have folder named `KiCad` now. Could you share the original Kicad folder names you have?

I believe it should be `KiCad` (for Kicad 6) and and `kicad` (for Kicad 5). But the system I am using may not differentiate upper and lower case letters. But I am not sure yet.

leoheck, 2022-06-27:
If you want to try, I've updated Kicad-Diff `kidiff` script to prioritize Kicad 6.

leoheck, 2022-06-28:
@vfelsner I will close this, feel free to open it again if you have any related issue.