leoheck/kiri #48: WSL Web server address not reachable

ONLYA, 2022-06-29:
I run kiri on a simple KiCad project on WSL Debian. It successfully runs but 127.0.0.1:8080 is not the right address for Windows OS.
As it is bridged to the local network, it is ideal to get 0.0.0.0 as the server address and using `cut -d' ' -f1 <<< $(hostname -I)` as the actual address.
With the modification above (in the `kiri` main script), I am able to get the page loading. However, the image and server checks are still `127.0.0.1`, which makes it grays out, unable to load the image and JavaScript substitutes the class as "server_down". 

All in all, I think it's a good idea to make the server address to be "0.0.0.0" instead of "127.0.0.1" so that everyone including WSL can access the generated pages.

I will have a look and submit a PR later. It sounds some more modifications to do.
Or do you have the other good idea to solve this problem for WSL?

leoheck, 2022-06-29:
Alright, you can try to comment out the part that checks if the server is on. I put that thing because if the server is not on, the images are not ok anymore. 

Also, if I am not mistaken, this code is the one that has this 127.0.0.1 kind of hardcoded. To have it working nicely it would be good to get this address from the server. 

Then we could also have a flag to pass to the command line to change the address. Maybe having this flag is just enough to make it work more flexibly, including for your case.

leoheck, 2022-06-29:
I am not an expert with networks but my feeling is that it should work with 127.0.0.1

leoheck, 2022-06-29:
> I run kiri on a simple KiCad project on WSL Debian. It successfully runs but 127.0.0.1:8080 is not the right address for Windows OS.

I got you could make it run. 

> Or do you have the other good idea to solve this problem for WSL?

What is the problem you are having? I could not understand it completely, I guess.

ONLYA, 2022-06-29:
Sorry I should express this more clearly.
You can consider WSL as a virtual machine with pre-mounted Windows drives. So the local loopback of the WSL is not the same as Windows, and `127.0.0.1` in WSL is not reachable in Windows OS.
In order to access the WSL application, you need to find the IP address of WSL in local network. In this case, `0.0.0.0` means that you are hosting the webserver in the local network instead of local loopback. As said here: https://docs.microsoft.com/en-us/windows/wsl/networking.

I could make it show the page by modifying the code as I said in the comment. The original code cannot run properly with WSL.

ONLYA, 2022-06-29:
Never mind, it works on my another windows machine. It just did not work on the first machine. I believe it is because of the Windows version. As it is not universal, I will find a walkaround for that computer. Thanks for help.

closed 2022-06-29 by ONLYA

leoheck, 2022-06-30:
Ah, this is why I did not understand. I don't use Windows, but I test these things sometimes. And this always worked for me on Windows, but it could be the case that it stopped working after any casual change.

Now, since I am using a webserver, it should also work from another computer in the same network with the real IP. And I think that the hardcoded part of the javascript is not good for that reason. Having this functionally working would be good in cases where you want to share the website with a coworker or where you have it running on another computer. I did not have these needs yet, but they are all interesting and possible.