leoheck/kiri #51: rename has different syntax on my system

dsperlich, 2022-08-04:
Hi,

the layer renaming does not work on my system (gentoo using normal util-linux version of rename). 
Man page state:
       rename [options] expression replacement file...

So i need to change the rename call to get it working:
diff --git a/bin/kiri b/bin/kiri
index 2603563..ef4f76d 100755
--- a/bin/kiri
+++ b/bin/kiri
@@ -1519,9 +1519,25 @@ generate_layout_artifacts()
                fi
 
                # Rename files folowing format: layer-xx.svg
-               rename "s/${kicad_project_name_1}-/layer-/" "${OUTPUT_DIR_PATH}/${commit_hash_1}/"*.svg &> /dev/null
-               rename "s/${kicad_project_name_2}-/layer-/" "${OUTPUT_DIR_PATH}/${commit_hash_2}/"*.svg &> /dev/null
-
+               rename ${kicad_project_name_1}- layer-  "${OUTPUT_DIR_PATH}/${commit_hash_1}/"*.svg &> /dev/null
+               rename ${kicad_project_name_2}- layer-  "${OUTPUT_DIR_PATH}/${commit_hash_2}/"*.svg &> /dev/null
+

Cheers
Dennis

leoheck, 2022-08-04:
Hi Dennis, could you format your issue report so I can understand it better?

leoheck, 2022-08-04:
Can you share the version of your rename command?

The one on my systems are:
```bash
➜  rename -V
/usr/bin/rename using File::Rename version 1.30, File::Rename::Options version 1.10
```

leoheck, 2022-08-04:
My man page states this

[image]

Your commands abod does not have the sed-like regexp that I was looking for.

dsperlich, 2022-08-04:
Mine is: 
`rename -V
rename from util-linux 2.38`

So interesting and somewhat concerning that Ubuntu uses a different rename than e.g. redhat based linux like rocky. So yours is from perl, mine from util-linux. Difficult...

leoheck, 2022-08-04:
Found this, https://stackoverflow.com/questions/31408764/using-regex-with-rename-version-from-util-linux

leoheck, 2022-08-04:
Ah, maybe gentoo have the util-linux rename by default. There is no rename on Ubuntu by default.
Maybe this rename is the the one compatible with the version on Ubuntu.

https://packages.gentoo.org/packages/dev-perl/rename

dsperlich, 2022-08-04:
Yes this would work for gentoo, even though annoying to require a certain version of such a package, but just checked Rocky Linux (RedHat derivative) and this also has rename from util-linux by default and there is no obvious perl rename.

dsperlich, 2022-08-04:
It is called emerge. But as gentoo user I would actually prefer not have a install script install dependencies, but rater tell me a list of packages needed. Usually i'm fine translating myself from the ubuntu names though. You consider echo out the list of ubuntu packages needed if the system is not officially supported.

leoheck, 2022-08-04:
I see and makes sense but then I is kind of harder to make sure the packages are the ones Kiri needs

Since the required rename would be this one:
[image]

Source: https://stackoverflow.com/questions/49267913/renaming-files-using-regular-expressions-linux

leoheck, 2022-08-04:
Actually I should use a better why to install and check dependencies...

leoheck, 2022-08-07:
A note regarding the rename command used was added inside the install_dependencies script. I am going to close this issue now, but feel free to reopen it if you still think it is important.

closed 2022-08-07 by leoheck