leoheck/kiri #52: Produced output not easy hostable in e.g. nginx

dsperlich, 2022-08-04:
Hi,

i want to host the produced output (multiple projects) on a normal webserver (nginx in my case). But the offline detection essentially blocks the simple, just copy the folder over and point to the index.html method. I also don't see a benifit in running the python server in the first place as it seems just to serve the plain files as they are in the folder. So why not just point the browser to the file:///<path to project>/.kiri/web/index.html in the first place? Do i miss some functionality the python http server provides?
I just disabled the offline check in the kiri.js and it seems to still work fine either with the nginx solution or pointing my browser directly to the files.

Cheers
Dennis

leoheck, 2022-08-04:
I am blocking it because otherwise, it won't work fine. But you can try it by commenting out the `check_server_status();` inside the `kirir.js` file.

leoheck, 2022-08-04:
Maybe the `check_server_status` is using the wrong URL or Port since it is currently hard coded.

You could try replacing the `img.src` inside the `function check_server_status()`
With this better version of it:
```js
img.src = window.location.origin + "/web/favicon.ico" + url_timestamp();
```

dsperlich, 2022-08-04:
Yes then it works. But i don't understand what is not working, well except the obvious one that new pictures are not loaded. Here i removed the check server in the js so it works on nginx: https://hacol16.physik.uni-freiburg.de/kiri/pbv3R3/web/index.html

I just added the minimal change to my fork to test loading the webside with the file path instead of the server: https://github.com/dsperlich/kiri obviously needs cleanup before merging.

dsperlich, 2022-08-04:
with your now `img.src` line i get now cross origin errors, when using local file instead of server, which apparently is expected behaviour: https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS/Errors/CORSRequestNotHttp

leoheck, 2022-08-04:
hmm, I was thinking that my changes would get the right link and not the "local" file. 
Please, add this `console.log(img.src)` after the improved `img.src` so we can check the whole URL that is being used.

dsperlich, 2022-08-04:
just `img.src = "favicon.ico?" + Date.now();`  seems to work fine.

leoheck, 2022-08-04:
Nice! And does the same functionality still works? If the webserver is not online anymore, then the page is still getting blocked?

dsperlich, 2022-08-04:
Yes, but i would prefer if instead of starting the server this is called:
`python -mwebbrowser file://${OUTPUT_DIR_PATH}/web/index.html`
See: https://github.com/dsperlich/kiri/commit/1ede861f22e03f3a3e5765211d4d0e7c9cdaa24d

leoheck, 2022-08-04:
If you pass the flag `-l` (lowercase L) it wont start the server

leoheck, 2022-08-04:
Maybe you can do something like this:
```bash
kiri -l && python -mwebbrowser file://${OUTPUT_DIR_PATH}/web/index.html
````

dsperlich, 2022-08-04:
The relative path does not work from command line. But how about adding an argument supressing the server start but rather start the file with my command containing the variable. Still don't understand the use case of the server, may be except for WLS.

leoheck, 2022-08-04:
Yeah, that would be an interesting idea. So, I could give an env var, so you can specify the command, externally. 
Ah, you can also make a file with the name kiri-server and pre-append it in the PATH, so Kiri would use yours instead of its own server. This may work out of the box already.

>  Still don't understand the use case of the server, may be except for WLS.

I may not understand your question completely. But it looks like the SVG images do not work well without having the server working. This is why I am using the server. I tired without it in the early stages of development and the images were not loading nicely. Maybe it works for some browsers but it may not work for others. I am considering it has to work on Chrome, Firefox, and Safari.

dsperlich, 2022-08-04:
The nice thing about the python command is it starts the "preferred" browser already. Like the python server does by the way as well.
Hmm changing more in PATH, not sure this is a great way to go.
Ok i can see it might be / have been a problem when zooming SVG. I sometimes loose the highest magnification in firefox. Chrome seems to work fine though. Didn't test safari.

leoheck, 2022-08-04:
If this is the issue, I am working to improve the code around the svg-pan-zoom. But I am struggling with if a bit. If I can improve it, it may work better without a server. Let's see how this will work when I get a more finished version of the code.

leoheck, 2022-08-07:
Let me know if you still think it would be good to have a custom way to specify the webserver command at the end. Then I will implement one so you can test it.

closed 2022-08-15 by leoheck