leoheck/kiri #53: Blind vias not shown correctly

dsperlich, 2022-08-04:
Hi,

i have a project where i use blind vias to different layers and want to track how they change. Problem is that vias are shown in all copper layers independet whether they are present there or not. This might be a problem of either kidiff or the pcbnew interface, but i worked around this in the kiri script for now.

What i did for my project at hand, i removed all vias from the layers and then modified all the circles with are also still in the svg to have white center and correct sized black rim. This works to some degree, the color coding of the delta is the wrong way round. So not sure how a proper solution would look like, even if kidiff / pcbnew would provide a proper via displaying svg.
For reference what i did to get blind vias working "correctly" in my project (note i hardcoded the via sizes):
               for f in "${OUTPUT_DIR_PATH}/${commit_hash_1}/"*.svg
                do
                    cat $f | sed -e ':a;N;$!ba;s/\n/\t/g' | sed -e "s/<\/g>/<\/g>\n/g" | grep -v 'style="fill:\#FFFFFF;' | sed -e "s/\(g style=\"fill:\#\)000000\(.*stroke-width:\)0.001000\(.*circle.*r=\"\)175.000000\(.*\)/\1FFFFFF\298.8\3125.6\4/" > $f.1
                    head -n -1 $f.1 | grep -v circle > $f
                    head -n -1 $f.1 | grep  circle >> $f
                    tail -1 $f.1 >> $f
                    rm $f.1
                done     
                for f in "${OUTPUT_DIR_PATH}/${commit_hash_2}/"*.svg
                do
                    cat $f | sed -e ':a;N;$!ba;s/\n/\t/g' | sed -e "s/<\/g>/<\/g>\n/g" | grep -v 'style="fill:\#FFFFFF;' | sed -e "s/\(g style=\"fill:\#\)000000\(.*stroke-width:\)0.001000\(.*circle.*r=\"\)175.000000\(.*\)/\1FFFFFF\298.8\3125.6\4/" > $f.1
                    head -n -1 $f.1 | grep -v circle > $f
                    head -n -1 $f.1 | grep  circle >> $f
                    tail -1 $f.1 >> $f
                    rm $f.1
                done     

Cheers
Dennis

leoheck, 2022-08-04:
Could you format the code here so it is easier to understand?

leoheck, 2022-08-04:
I never get a project with blind vias, it may be a limitation of Kicad. Do you have a sample project with those blind vias so I can check how the output would be? Do you have any picture showing how the blind-vias would be in Kicad and how they appear on Kiri?

dsperlich, 2022-08-04:
Sure this repo might be public, if not i can get it to you in a different way: https://gitlab.cern.ch/theim/powerboard_v3/-/tree/master/EC/pbv3R01

It contains multiple projects, but this subfolder also has working schematics :-). This is how i made this project render for me: https://hacol16.physik.uni-freiburg.de/kiri/pbv3R01/web/index.html
But by default all blind vias are visible on all layers, so the changes where i moved just some vias from from layer pair to an other is not represented at all.

leoheck, 2022-08-04:
Nice, I was also able to create my first design with blind/buried vias, thanks for giving me the opportunity to do that.

My Kiri is a bit broken now since I struggle to handle the SVG better in the HTML. 

But I could reproduce your issue, and I was also able to test your code, which seems to be working too. 
I just have to put some time into understanding your commands to check if they are going to work every time or if I want to improve them a bit. And then if it works fine then I will add it to Kiri for sure.

dsperlich, 2022-08-04:
so one annoying thing is that the head / head / tail is there to reorder the vias to the bottom, so they are not covered by traces. The sed most certainly needs automatic understanding how big the vias actually are, if you want i can also look into making this rebust.

dsperlich, 2022-08-04:
Technically it needs more information from kicad_pcb to have the correct size of the drill. But one can cheat and just make them "look good" by just have halve diameter of the copper pad as drill or so. The other problem of the inverted colors i'm not sure how to solve.

leoheck, 2022-08-04:
Good that now I am exporting the board for each commit, and then we can get a list of vias easily with:
```
grep "(via " board.kicad_pcb
```

leoheck, 2022-08-04:
I am thinking here that we could have a script, that you pass one svg and the related board.
And it updates the svg removing/showing/fixing related vias.

Lets say...

```bash
fix_vias .kiri/40ff518/kiri/pcb/layer-00.svg .kiri/bc06be3/board.kicad_pro
```

Then you could create this without having to touch Kiri which is kind of messy. 
And once it is done we can add it in Kiri iterating available svgs and fixing them.
We could parse something inside the pcb to check if it supports blind/buried vias so we can skip this step when it is not necessary.

Loking inside the sample project I made.
I have some interesting things in each via like if it is `blind` or not, the `layers used`, and  also this flag `remove_unused_layers`

```
  (via blind (at 159.6 112.2) (size 0.8) (drill 0.4) (layers "In1.Cu" "In2.Cu") (remove_unused_layers) (net 1) (tstamp 1201cc23-0293-45b8-abdc-276d1e8be62d))
  (via blind (at 148.155 112.245) (size 0.8) (drill 0.4) (layers "In1.Cu" "In2.Cu") (remove_unused_layers) (net 1) (tstamp 8ea86a54-62d6-46f5-8fec-e30ad3b65ffe))
  (via blind (at 149.9 81.9) (size 0.8) (drill 0.4) (layers "F.Cu" "B.Cu") (net 2) (tstamp 6161a8bb-98fb-49c6-b4c0-8fa089606584))
  (via (at 149.9 86.3) (size 0.8) (drill 0.4) (layers "F.Cu" "B.Cu") (net 2) (tstamp 6b6f377b-c2d9-4aab-a05c-21bd4465a14f))
```

dsperlich, 2022-08-04:
Annoyingly, layers is only start and stop, so not a full list of all layers, so one would need to keep track which layers are in between. That's why i'm wondering whether it makes more sense to do it in the pcbnew svg productionn step. The option remove_unused_layers is to remove the copper ring in layers where the via is not used. I guess also this is only usefull in the pcbnew export function, as otherwise figuring out whether a via is used is quite difficult.

leoheck, 2022-08-05:
Ah, Kicad is full of those little headaches. It is not easy to make this Kiri too. haha

But we can use layers IDs for that. They are also sequential and the last layer ID of a copper layer is 31 `B.Cu`. 
[image]

leoheck, 2022-08-05:
Are you sure blind/buried vias are not working? Now I could fix the branch I was working on. But it looks like the vias are working right. Which version of Kicad do you have? I am using the latest one, 6.0.7. If you are not using this one, then I would like to ask you to upgrade it and give it a try.

leoheck, 2022-08-05:
@edahlseng, since you are here, do you mind testing my updates in this branch here?

It would help me a speeding up the merge of this new thing in the main branch.

- I am avoiding updating the layers list when it is not needed (the new layer list is equal to the old layer list when changing commits)
- Added extra keyboard shortcuts (pan with Alt+Arrows, zoom with +/-/0)

This branch here,
https://github.com/leoheck/kiri/tree/dynamic-sheets-and-layers

After that I can tackle the issue of the vias here.

leoheck, 2022-08-05:
I made this script to help investigate this issue.

It echoes vias found inside kicad_pcb
And it also echos circles inside the svgs of each layers

https://www.dropbox.com/s/sq020q52yi9x5zr/fix_svg_vias?dl=0

The usage is this 

```
# USAGE:
# fix_svg_vias <kicad_pcb> <svg_folder>

# EXAMPLE:
# fix_svg_vias kicad-blind-burried-vias.kicad_pcb .kiri/40ff518/kiri/pcb/
```

Now, the next step is to understand how do we map kicad_pcb vias to svg circles since they have nothing in common. Maybe the placements, but the numbers are not quite the same since they are getting rounded.

leoheck, 2022-08-06:
I am not sure if it is right, but if you download the above script, and put it inside the kiri/bin folder
You test this adding these lines at line 1519 in kiri
It it kind of slow... since it is iterating a lot... 
But sit and wait, we just have to test if it is working... then we can improve it

``` 
fix_svg_vias "${OUTPUT_DIR_PATH}/${commit_hash_1}/${kicad_pcb}" "${OUTPUT_DIR_PATH}/${commit_hash_1}/kiri/pcb/"
fix_svg_vias "${OUTPUT_DIR_PATH}/${commit_hash_2}/${kicad_pcb}" "${OUTPUT_DIR_PATH}/${commit_hash_2}/kiri/pcb/"
```

leoheck, 2022-08-06:
this is still ignoring layers F.Maks and B.Mask that have to be taken in consideration too.

leoheck, 2022-08-07:
I updated the above branch with my attempt to fix these issues with blind/buried vias... now, it is a quick fix, that takes some time to process... you can use -t flag to limit the number of commits being proceeded to test this branch.

leoheck, 2022-08-07:
By the way, your hosted project is awesome.. now I am thinking that I have to do something with the svgs to make them faster for the internet and also to cache them, or to have a button to download the project in the background to it will be more responsive online. Currently, it is taking a lot of time to change images. But this is awesome.

I was too focused to create a fix for the vias issue that I did not access the links you have shared before, and also since I could generate quickly a project with buried vias.

Can I share this link on Kicad forum? So people can see Kiri working? I can tag you there too.
https://hacol16.physik.uni-freiburg.de/kiri/pbv3R01/web/index.html

leoheck, 2022-08-07:
Now, testing my quick script to fix those blind bias with your project, it looks like it is going to fail since it has toooooo many vias. I have to improve the speed otherwise it won't work. My script says it has 651 vias.

leoheck, 2022-08-07:
This is the project I was using to test my script in case you want to see how it works.
https://github.com/leoheck/kicad-blind-burried-vias

leoheck, 2022-08-07:
Hey, good news. I think I have fixed this in a better way.
When ploting the PCB (using KicadDiff), the drillmarks were being ploted on every layer.
Then disabling it, fixed this issue without extra post-processing
```python
popt.SetDrillMarksType(pcbnew.PCB_PLOT_PARAMS.NO_DRILL_SHAPE) # this "closes" the holes, so the output (when monochrome) is the same as PDF plot from Pcbnew GUI!
```

The fix is already in this branch here: https://github.com/leoheck/kiri/tree/dynamic-sheets-and-layers

Not sure if this proves something, but here is a picture of your design
[image]

leoheck, 2022-08-07:
Let me know if you still have issues related with those vias after this fix. You can reopen this if needed.

closed 2022-08-07 by leoheck