leoheck/kiri #56: THT holes with corresponding pads in upper left corner

koejul, 2022-08-24:
I am using your tool and am mostly satisfied with the result. Only one issue occured to me. 
As visible in attached picture, all through holes with their corresponding pads are collected in the upper left corner instead of being in their designated places.
I have tried the same pcb file in kidiff alone, their the holes with pads stay in the correct place.
If any more information is required, I will gladly attach it.
[image]

koejul, 2022-08-24:
The KiCAD version used to create and edit the inspected project is KiCAD 5.1.6

leoheck, 2022-08-25:
Interesting, thanks for sharing this issue.

Which version of the Kiri are you using? 

By the way, could you share the project so I can check if I can reproduce your results here?

Alternatively, the latest version of Kicad 5 is 5.1.12 of 2021-11-05, could you check if the newest version will give you the same results?

koejul, 2022-08-26:
I am using the latest available Kiri Version, downloaded just a few days ago.
I can not share the project in the picture but an alternative project with similar issues. I also attach a second project, where the issue does not occur but no rememberable differences where made while creating the two projects, so same libs, same kicad version, same layer count etc.
In total, I now have eight projects where this issue occurs using kicad 5.1.6 and one project, in which it does not occur.
Unfortunately, I will not try version 5.1.12 for now as there is also a project which does not show this issue on kicad 5.1.6.
If I can provide any further information for solving this issue, please just tell me.

In following project, the issue does NOT occur:
[20220826_KiCAD_Testprojekt.zip](https://github.com/leoheck/kiri/files/9431787/20220826_KiCAD_Testprojekt.zip)

In following project, the issue DOES occur:
[a.Box.2.print_2pSwitch_w_SOM.zip](https://github.com/leoheck/kiri/files/9431807/a.Box.2.print_2pSwitch_w_SOM.zip)

leoheck, 2022-08-26:
> I can not share the project in the picture but an alternative project with similar issues.

This is enough for me to start investigating this. Thanks for helping me, hoever I cannot run Kiri directly on them, since these are not git repos. Could you share the repo instead of the export?

Meanwhile, let me know if you tried cleaning the generated files once on those projects that fail to place vias. This is done with the flag `-r`. Inside the project directory, you can run it again using the command `kriri -r`.

Now, a word about versions. You should definitely use the latest Kicad of that version. Since you are using the old Kicad 5, then you should upgrade to Kicad 5.1.12 at least since it brings bug fixes for 5.1.6 issues. This is different than upgrading to the latest Kicad 6. It is like using an old and comfortable shoe that has the sole rubber loose. If you glue it, making it work and look better, it is still the same old and comfortable shoe but now it gives better quality walks and allows you to walk farther.

leoheck, 2022-08-26:
Right now, at this machine, I have the latest Kicad 6 installed. So I am going to test quickly check if I have the same issues with my current version of Kicad. 

Kiri uses the `plot_kicad_pcb` command to plot the boards, so I dont have to use Kiri.

=====================

This is one of the layers of the project you said DOES have the issue, right?
`plot_kicad_pcb a.Box.2.print_2pSwitch_w_SOM.kicad_pcb` 

This is F.Cu layers, it looks good to me. Could you confirm?
[image]

=====================

To compare, I am doing the same with the prokect that does NOT have the issue.

The same layer, F.Cu, and this also seems not to have the issue, as you said.
[image]

=====================

I need some time to prepare a way to test with the old Kicad 5. But it looks like this is a Kicad issue. I remeber that I made a script to workaround that in some way. I just have create a way to reproduce your issue to understand how to proceed.

But I feel that if you upgrade the Kicad to 5.1.12 your problems will be solved. (You could also help me try the same on a virtual machine, I use Virtualbox for these tests)

leoheck, 2022-08-26:
(Off-topic), I like the traces of your designs, they are nice and well organized. Thanks for sharing.