leoheck/kiri #59: Changing schematic sheet fails

topherbuckley, 2022-09-27:
When attempting to open another sheet on a hierarchical design the firefox dev tools claims:

```
Uncaught ReferenceError: change_page_onclick is not defined
    onclick http://127.0.0.1:8080/web/index.html:1
```

It appears all the svg files needed for the diffs exist in the sub directories of `.kiri` (i.e. they plotted just fine), but the html or js is missing something.

topherbuckley, 2022-09-27:
Not a fix, but this can be worked around by clicking the relevant sheet and refreshing the page.

leoheck, 2022-09-27:
Does it show the line number?

leoheck, 2022-09-27:
I could not reproduce this issue. Can you share the console logo of your test?

leoheck, 2022-09-27:
Or maybe your project... maybe it is an issue with the software instead.

leoheck, 2022-09-27:
But I understand that multi-sheet operation may have issues.

topherbuckley, 2022-09-28:
I'm using [this sandbox example](https://github.com/topherbuckley/kicad_sandbox/tree/master/top) so feel free to test on this. 

Here is the full console from a new kiri generated tab in firefox after running `kiri -D` from the `top` directory. 

```
================================================================================ kiri.js:472:13
commit1: 5f93a6c kiri.js:493:13
commit2: 6d7660e kiri.js:494:13
----------------------------------------- kiri.js:549:13
filePath: ../5f93a6c/kiri/sch_sheets?t=5f93a6c kiri.js:535:13
Synchronous XMLHttpRequest on the main thread is deprecated because of its detrimental effects to the end user’s experience. For more help http://xhr.spec.whatwg.org/ kiri.js:539:12
filePath: ../6d7660e/kiri/sch_sheets?t=6d7660e kiri.js:535:13
Server is Online kiri.js:1197:17
[SCH]  Sheets = 2 kiri.js:657:13
sheets 
Array [ "top", "subsheet" ]
kiri.js:658:13
current_sheets_list =  
Array [ "top", "subsheet" ]
kiri.js:684:13
new_sheets_list =  
Array [ "top", "subsheet" ]
kiri.js:685:13
Keep the same list of sheets kiri.js:687:17
[SCH] page_filename = top kiri.js:569:13
[SCH]  image_path_1 = ../5f93a6c/kiri/sch/top.svg kiri.js:570:13
[SCH]  image_path_2 = ../6d7660e/kiri/sch/top.svg kiri.js:571:13
>=============================================< kiri.js:1321:13
removeEmbed... kiri.js:1322:13
>> lastEmbed:  null kiri.js:1323:13
>> panZoom_instance:  null kiri.js:1324:13
createNewEmbed... kiri.js:1214:13
>>> SVG:  
<div id="div-svg" style="display: inline; width: …t; max-height: inherit;">
kiri.js:1257:13
panZoom_instance: 
Object { enablePan: enablePan(), disablePan: disablePan(), isPanEnabled: isPanEnabled(), pan: pan(t), panBy: panBy(t), getPan: getPan(), setBeforePan: setBeforePan(t), setOnPan: setOnPan(t), enableZoom: enableZoom(), disableZoom: disableZoom(), … }
kiri.js:1294:13
----------------------------------------- kiri.js:549:13
filePath: ../5f93a6c/kiri/sch_sheets?t=5f93a6c kiri.js:535:13
filePath: ../6d7660e/kiri/sch_sheets?t=6d7660e kiri.js:535:13
[SCH]  Sheets = 2 kiri.js:657:13
sheets 
Array [ "top", "subsheet" ]
kiri.js:658:13
current_sheets_list =  
Array [ "top", "subsheet" ]
kiri.js:684:13
new_sheets_list =  
Array [ "top", "subsheet" ]
kiri.js:685:13
Keep the same list of sheets kiri.js:687:17
[SCH] page_filename = top kiri.js:569:13
[SCH]  image_path_1 = ../5f93a6c/kiri/sch/top.svg kiri.js:570:13
[SCH]  image_path_2 = ../6d7660e/kiri/sch/top.svg kiri.js:571:13
function onload kiri.js:1073:13
Uncaught ReferenceError: change_page_onclick is not defined
    onclick safe-area-provider.css:1
index.html:1:1
Uncaught ReferenceError: change_page is not defined
    onchange safe-area-provider.css:1
index.html:1:1

```
I don't normally work with javascript or any web tools like this, so if there is a better way to share this info, please let me know.

topherbuckley, 2022-09-28:
Interestingly, I am able to get the sheets to change if I firsts scroll through the commits with the keyboard, THEN try changing the sheet. Maybe something is initialized regarding this `change_page_onclick` when the different commits are opened?

topherbuckley, 2022-09-28:
It appears this may have something to do with an update to one or all of the  `sheets`, `current_sheets_list`, or `new_sheets_list` Array. I say this because you must scroll to the first commit in my sandbox repo (where only one sheet exists) then moving to new commits before you are able to change sheets. See gif of this behavior below:
[image]

leoheck, 2022-09-28:
Yeah, this is an issue to solve.

leoheck, 2022-09-28:
I will have to check how can I solve this. But I could reproduce this issue too.

leoheck, 2022-09-30:
I am experimenting with your project. Thanks for sharing it. 
Just for the sake of comparison, I am starting with Chrome, since this is the one I use. 

Initially, navigating the commits list using arrow up/down works fine.
Also, when you navigate the commits list, and then navigate between the sheets with arrows left/right, it works fine. 

However, if you reload the page going to the initial (sane) state. And navigate the subsheets, and then navigate to a commit that does not have this sheet like the initial 2 commits, and then navigate the commits list upwards... the view loses itself since there is no sheet selected anymore. To reproduce what I am saying, do this, please:

Like:
1. Refresh the page
2. Arrow_Right
3. Arrow_Down (2 times)
4. Arrow_Up <==== Then the view does not update, commits do not change and the console show some red lines.

Also, I saw it is not quite good to click on each Sheet in the Sheets list. Something funny is happening there.

In short, I could reproduce 2 issues. 
I have to re-read your messages to if you are reporting something else.

leoheck, 2022-09-30:
@topherbuckley I pushed a code to improve the issue where the list is not being updated correctly. Could you test?

This is my PR https://github.com/leoheck/kiri/pull/60
This is the branch that could test, please. https://github.com/leoheck/kiri/tree/hierarchical_sheets_fix

leoheck, 2022-09-30:
Nevermind, it was merged, you can test it directly from the master branch.

leoheck, 2022-09-30:
Please, open the issue again if you have any other related problem, or create a new one if it is a different problem.

closed 2022-09-30 by leoheck

topherbuckley, 2022-10-05:
I pulled from master and reran, but get the same console errors and behavior as before. I verified the changes to the kiri.js script from your PR are being used by firefox by viewing the kiri.js file within the Developer Tools --> Debugger. 

Console log from same procedure as before:

```
Layout was forced before the page was fully loaded. If stylesheets are not yet loaded this may cause a flash of unstyled content. index.html
================================================================================ kiri.js:475:13
commit1: 5f93a6c kiri.js:496:13
commit2: 6d7660e kiri.js:497:13
----------------------------------------- kiri.js:552:13
filePath: ../5f93a6c/kiri/sch_sheets?t=5f93a6c kiri.js:538:13
Synchronous XMLHttpRequest on the main thread is deprecated because of its detrimental effects to the end user’s experience. For more help http://xhr.spec.whatwg.org/ kiri.js:542:12
Server is Online kiri.js:1240:17
filePath: ../6d7660e/kiri/sch_sheets?t=6d7660e kiri.js:538:13
[SCH]  Sheets = 2 kiri.js:684:13
sheets 
Array [ "top", "subsheet" ]
kiri.js:685:13
current_sheets_list =  
Array [ "top", "subsheet" ]
kiri.js:711:13
new_sheets_list =  
Array [ "top", "subsheet" ]
kiri.js:712:13
Keep the same list of sheets kiri.js:714:17
[SCH] page_filename = top kiri.js:596:13
[SCH]  image_path_1 = ../5f93a6c/kiri/sch/top.svg kiri.js:597:13
[SCH]  image_path_2 = ../6d7660e/kiri/sch/top.svg kiri.js:598:13
>=============================================< kiri.js:1364:13
removeEmbed... kiri.js:1365:13
>> lastEmbed:  null kiri.js:1366:13
>> panZoom_instance:  null kiri.js:1367:13
createNewEmbed... kiri.js:1257:13
>>> SVG:  
<div id="div-svg" style="display: inline; width: …t; max-height: inherit;">
kiri.js:1300:13
panZoom_instance: 
Object { enablePan: enablePan(), disablePan: disablePan(), isPanEnabled: isPanEnabled(), pan: pan(t), panBy: panBy(t), getPan: getPan(), setBeforePan: setBeforePan(t), setOnPan: setOnPan(t), enableZoom: enableZoom(), disableZoom: disableZoom(), … }
kiri.js:1337:13
----------------------------------------- kiri.js:552:13
filePath: ../5f93a6c/kiri/sch_sheets?t=5f93a6c kiri.js:538:13
filePath: ../6d7660e/kiri/sch_sheets?t=6d7660e kiri.js:538:13
[SCH]  Sheets = 2 kiri.js:684:13
sheets 
Array [ "top", "subsheet" ]
kiri.js:685:13
current_sheets_list =  
Array [ "top", "subsheet" ]
kiri.js:711:13
new_sheets_list =  
Array [ "top", "subsheet" ]
kiri.js:712:13
Keep the same list of sheets kiri.js:714:17
[SCH] page_filename = top kiri.js:596:13
[SCH]  image_path_1 = ../5f93a6c/kiri/sch/top.svg kiri.js:597:13
[SCH]  image_path_2 = ../6d7660e/kiri/sch/top.svg kiri.js:598:13
function onload kiri.js:1116:13
Uncaught ReferenceError: change_page_onclick is not defined
    onclick safe-area-provider.css:1
index.html:1:1
Uncaught ReferenceError: change_page is not defined
    onchange safe-area-provider.css:1
index.html:1:1
```

Here is a diff of the above and the initial log if it is helpful:

```
1,9c1,10
< ================================================================================ kiri.js:472:13
[... 65 lines cut ...]
40c41
< kiri.js:658:13
---
> kiri.js:685:13
43c44
< kiri.js:684:13
---
> kiri.js:711:13
46,51c47,52
< kiri.js:685:13
< Keep the same list of sheets kiri.js:687:17
< [SCH] page_filename = top kiri.js:569:13
< [SCH]  image_path_1 = ../5f93a6c/kiri/sch/top.svg kiri.js:570:13
< [SCH]  image_path_2 = ../6d7660e/kiri/sch/top.svg kiri.js:571:13
< function onload kiri.js:1073:13
---
> kiri.js:712:13
> Keep the same list of sheets kiri.js:714:17
> [SCH] page_filename = top kiri.js:596:13
> [SCH]  image_path_1 = ../5f93a6c/kiri/sch/top.svg kiri.js:597:13
> [SCH]  image_path_2 = ../6d7660e/kiri/sch/top.svg kiri.js:598:13
> function onload kiri.js:1116:13
58d58
< 
```

topherbuckley, 2022-10-05:
I'm far from a JS expert, but from what I can tell, the kiri.js should have two methods that do not exist:

1. change_page
2. change_page_onclick

It seems you inject these methods to the each sheet name icons [here](https://github.com/leoheck/kiri/blob/main/assets/kiri.js#L694) but fail to define them elsewhere.

leoheck, 2022-10-05:
There is no change_page and change_page_onclick on this line. 
There is also none of them in the whole file. 
Are you sure you have the right file or the one updated with what is in the master?