leoheck/kiri #61: Install script not working on Fedora

TomSaw, 2022-10-28:
The [Fedora](https://getfedora.org/en/workstation/) distro is a famous choice by developers!

Developer -> KiCAD, Developer -> Fedora ;)

leoheck, 2022-10-28:
What does this mean?

TomSaw, 2022-10-28:
executing
```
bash -c "$(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_dependencies.sh)"
```
outputs
```
Error: Unknown system
Please, ask KiRI dev to adapt the dependencies installer
```
`install_dependencies.sh` does not contain any calls of `dnf` (default packet manager in fedora) so i guess it's not a bug but generally not supported yet.

TomSaw, 2022-10-28:
I've not been successful tweaking the script for fedora so far.

Most stuff, most notable the compiling part, works but .local/share/kiri is not created.

Unfortunately i'm not familiar with these build tools.

leoheck, 2022-10-28:
Yes, the installer does not work on Fedora. No one has asked this yet.

leoheck, 2022-10-28:
> install_dependencies.sh does not contain any calls of dnf (default packet manager in fedora) so i guess it's not a bug but not yet implemented.

Yes. I will give it a try. And the I will let you know so you can check if it works, coll?

leoheck, 2022-10-28:
Which version of Fedora are you using?

leoheck, 2022-10-29:
There is something working already... if you want to check...
The schematics worked for me. Still have to check missing dependencies for the layout part.

leoheck, 2022-10-29:
Ah, the wrong version of rename is in use. It has to be the perl rename (prename). I have to think on how to fix this.

leoheck, 2022-10-29:
Alright, for me it is working already, tested on Fedora 37. 
If you have issues running other version, please, let me know.

TomSaw, 2022-10-29:
I'm on fedora 36, this may be the reason.
wxpython does not want to upgrade due to locked packages i guess:
`cannot import name 'Mapping' from 'collections' (/usr/lib64/python3.10/collections/__init__.py)`

Since fedora 37 release is around the corner so i switch and hope python is refreshed and things work then ;)

leoheck, 2022-10-30:
What are you trying to do?
Just run the install dependencies and then install the tool and run it. 
It is working for me. 
I installed in a clean fresh Fedora that was downloaded yesterday.

leoheck, 2022-10-30:
Kicad-Diff (the piece that generates layouts) uses WxWidgets to run graphically, but Kiri does not need that, it runs it without the GUI. So, maybe you can ignore this.

leoheck, 2022-10-31:
Could you test if it is already working?

TomSaw, 2022-11-17:
Hey @leoheck. Upgraded to fedora 37 today and now kiri installs fluidly.
I've also tested 2 weeks ago with fedora 37 in gnome-boxes but didn't work... but who cares anymore.

Such a nice tool - Thank you!!!

leoheck, 2022-11-17:
> but who cares anymore.

haha

Awesome, glad it worked right out of the box this time.