leoheck/kiri #63: `plot_kicad_sch` not working because of non englisch language setting

worldwidemv, 2022-11-14:
Hello,

If the KiCad language is not English, the keyboard shortcut `alt+f` will not open the file menu but maybe another menu.
Therefore, ploting the schematic fails without any error message or info. 

Maybe you can have a section about the KiCad language requirements in the `Readme.md`?

leoheck, 2022-11-14:
Hi @worldwidemv 

Can you show me the output of this command?
```bash
locale -a
```

Can you check if you can launch the English version (with the English shortcuts) by doing this:
Note that the language you are going to use (English) has to be in the list above.

```bash
LANG=en_US.utf8 kicad

# or 
LANG=C.utf8 kicad

# or 
LANG=C kicad
```

worldwidemv, 2022-11-14:
Unfortunately, the language overrides are ignored by KiCad / eeschematic, probably because the have there own language system via `Preferences`-> `Language`.
Or maybe Python ignores that?

I would be ok, mentioning it in the install instructions. I am not sure, but the person installing this needs to add the `en_EN` locale anyway by hand ....   
**Changing the language in KiCad works.**

Here is the output of `locale -a` after I had to fight with it to work properly again...
```bash
>$ locale -a
C
C.utf8
de_AT.utf8
de_BE.utf8
de_CH.utf8
de_DE
de_DE.iso88591
de_DE.utf8
de_IT.utf8
de_LI.utf8
de_LU.utf8
deutsch
en_AG
en_AG.utf8
en_AU.utf8
en_BW.utf8
en_CA.utf8
en_DK.utf8
en_GB.utf8
en_HK.utf8
en_IE.utf8
en_IL
en_IL.utf8
en_IN
en_IN.utf8
en_NG
en_NG.utf8
en_NZ.utf8
en_PH.utf8
en_SG.utf8
en_US.utf8
en_ZA.utf8
en_ZM
en_ZM.utf8
en_ZW.utf8
german
POSIX
```

worldwidemv, 2022-11-14:
Another idea was to override the hotkey file [user.hotkeys](https://docs.kicad.org/6.0/en/eeschema/eeschema.html#hotkeys).
But `Alt+f` seems to be a system shortcut since it is not in there...

leoheck, 2022-11-14:
Can you check again if Kicad opens with the other language when telling Kicad to do so from the command line?
For me, it works.
I am forcing it to use my mother language here
[image]

For you, check please if kicad opens in your language when you do this
```
LANG=de_DE.utf8 kicad
```

leoheck, 2022-11-14:
Can you also show me the output of the locale command?
This is mine.

```bash
➜  ~ locale
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=
```

leoheck, 2022-11-14:
Here I installed the german since I did not have it on my system and then launched it setting the same variable
[image]

leoheck, 2022-11-14:
What is your operating system by the way?

leoheck, 2022-11-14:
I pushed a fix, it may work for you by default.

worldwidemv, 2022-11-15:
I will try it again tomorrow. You probably have the KiCad language set to `Default`. I will try that tomorrow. If it is set to German or English, then the `LANG` variable has no effect.

OS is ubuntu 22.04

leoheck, 2022-11-15:
Indeed, in mine, the language is the default. But why do you use the system in one language and kicad in another language? It does not make much sense to me.

worldwidemv, 2022-11-15:
I changed KiCad to English to get it to work.
And it is also sometimes easier because tutorials and so on are in English which makes following much easier....

leoheck, 2022-11-15:
Sure, then use the OS in English too.

worldwidemv, 2022-11-15:
Yes, sorry I thought this is not important, since the base system is Ubuntu22.04  and this is `bash` anyway.

Anyway, `LC_ALL=C` does the trick. With and without `export`.

leoheck, 2022-11-15:
In this case, it is important since it is not the default window manager. Nice it has worked, thanks.

closed 2022-11-15 by leoheck