leoheck/kiri #64: Docker

leoheck, 2022-12-10:
Hey @TadeasPilar, I saw you were making good progress to run Kiri with Docker. 

I was also doing the same some days ago. 

Did you make progress on that? 

I manage to make something today, and I put my work here https://github.com/leoheck/kiri-docker
It is a different repo since it does not need anything from the main Kiri repo since everything will be inside the Docker.

I invite you to check if it works for you.

I just tested it on my Ubuntu Linux, so for me, it is working already. I want to test on other systems and improve it a bit when I have more time. 

Also, it generates .kiri files in place (inside the repo of the project on the host)

To run it, you just need to do
```bash
make build # to build the docker image
./kiri PROJECT_PATH
```

TadeasPilar, 2022-12-13:
Hi,
I managed to get it working in my fork, but I must say it was lot harder than anticipated. And clean docker build takes me about 20 minutes. How is your build doing on that front? It seems very excessive to me.

--- original message ---

--- end of original message ---

TadeasPilar, 2022-12-13:
> I don't think it was hard. My image and script also allows you to open GUI's too. The run script works with the already created container. You don't have to rebuild the container everytime to put your files in it. I also shared the existing image on docker hub.
>
> This is my first approach tho. It can be improved with time.
>
> I don't think it takes 20 minutes to build. But I have to evaluate this again, later.
>
> You said it is very excessive. Very excessive in which sense?

My goal with this whole thing is to automatically generate project output files on Jenkins build server. So that with every commit, output files are generated. Right now, I am automatically generating renders, bom, ibom, gerbers, and I also want to add kidiff. This would allow for viewing KiCad files and changes online, without having to download the whole project and set up libraries. For this, GUI's aren't all that useful.

My biggest issue right now, is the need for active server running in the background. I know very little javascript, but I would like to figure out, how to rewrite your code into running correctly without active server.

--- original message ---

leoheck, 2022-12-13:
> For this, GUIs aren't all that useful.

Sure, I totally agree with you. In this case, it looks like it needs 2 different images. 
One that is as light as possible, to download/build fast.

Also, there is this case that I am creating initially, that allows me to debug and test things and evaluate this Docker thing, easily.

I think I will also provide this second image for the reason you are describing, and doing the same as you, using the kicad_auto image, to make other tools available.

> My biggest issue right now, is the need for active server running in the background.

It does not need that.. you can disable the server with the parameter `-l` to not launch the server in the end. There is also an option for Kiri to archive the generated files. This option, if I am not mistaken, generates a package with the webserver script inside... 

So, if you unpack the files on another computer, you can start the server quickly by running the script that is already inside the folder

leoheck, 2022-12-13:
ah, for kicad6, kiri needs gui tools, since it plots schematics using gui since there no tools that can do the same in the command line like plotgitsh

leoheck, 2022-12-16:
@TadeasPilar I figured out that my current Dockerfile was very badly optimized and the image was consuming a lot of space. Now it is a bit better. Also, I have removed some unnecessary dependencies like wxpython* that are not really necessary for command line operation. Feel free to check my [docker file](https://github.com/leoheck/kiri-docker/blob/main/Dockerfile) for the currently required dependencies. 

I still did not try to generate an image using `ghcr.io/inti-cmnb/kicad6_auto:dev` as the base image. But it may happen eventually, so all command line tools can be available easily.

leoheck, 2022-12-16:
The compressed image, if someone wants to donwload, is 1GB now, https://hub.docker.com/r/leoheck/kiri/tags

Without compression it takes 3GB

leoheck, 2022-12-16:
It is taking 8 minutes to build the image on my machine now.
It is using 2.84 GB.
And the compressed form, online, is 982.15 MB already.

I think I can still trim it a bit more...

leoheck, 2022-12-18:
@TadeasPilar  I added a new Docker file with kicad_auto tools. This one https://github.com/leoheck/kiri-docker/blob/main/Dockerfile_kicad-auto.

Example of usage, with other commands:
```bash 
./kiri /home/lheck/Documents/assoc-board -i leoheck/kiri:test -d -c "pcbdraw board.kicad_pcb board.svg"
```

edahlseng, 2023-01-06:
This looks interesting, maybe the Docker image could be used to support a GitHub Action that could be easily shared?

leoheck, 2023-01-06:
Yeah, that would be interesting too.

Conor-Burns, 2023-01-11:
I'm also interested in the GitHub action. That would be a great feature! 
Will try to experiment a bit.

leoheck, 2023-01-11:
Nice, I did not have time yet to check this. But in the past, I was using some existing actions to achieve that. So, I stopped doing that at that time because I could not find an easy way to host kiri's website while browsing an Issue or PR, for instance. Yes, I could send data back to the Issue/PR, and I also created a flag to easily archive the run ( -x|--archive)  and an easy way to review the results (https://github.com/leoheck/kiri#archiving-generated-files) without having kiri installed locally, but I really wanted to have a way to use the web service on the fly. Today I may find something. I will check what I can do but help on that can be used too. Thanks you.

closed 2023-02-06 by leoheck