leoheck/kiri #65: PCB Diffs are not shown in larger GIT repos

worldwidemv, 2022-12-15:
### Bug Symptoms / Observed Behaviour

`kiri` runs normal, but in the webpage, when I go to the PCB diffs, nothing is shown.

### Detailed Description

The problem occurs, if the repository reaches a certain size, because then, the default 7 digi short hash is no longer [unique](https://blog.cuviper.com/2013/11/10/how-short-can-git-abbreviate/).
GIT [automatically detects](https://lkml.org/lkml/2013/9/30/365) this and increases the short hash length to, e.g. 8 digis.

`kiri` works fine with larger hashes, but `kidiff/kidiff.py` is using only 7 digis (line 124), so renaming and moving the generated SVGs fails without any error message (`kiri` line 1525 ...).   

### Possible Solution / Workaround

It works fine, if I have removed the `[:7]` in `kidiff/kidiff.py` line 124 and 127 and in `kidiff/scms/git.py` line 17 and 22.
I am not sure why they where introduced.

@Gasman2014 Hello, do you remember why the commit has was cut to 7 digits?

### System Info

- KDE Neo (Ubuntu 22.04 base)
- git version = 2.34.1
- repo info
```
Processing blobs: 7829                        
Processing trees: 9061                        
Processing commits: 1378                        
Matching commits to trees: 1378                        
Processing annotated tags: 2                        
Processing references: 20                        
| Name                         | Value     | Level of concern               |
| ---------------------------- | --------- | ------------------------------ |
| Overall repository size      |           |                                |
| * Commits                    |           |                                |
|   * Count                    |  1.38 k   |                                |
|   * Total size               |   373 KiB |                                |
| * Trees                      |           |                                |
|   * Count                    |  9.06 k   |                                |
|   * Total size               |  2.57 MiB |                                |
|   * Total tree entries       |  65.7 k   |                                |
| * Blobs                      |           |                                |
|   * Count                    |  7.83 k   |                                |
|   * Total size               |  1.52 GiB |                                |
| * Annotated tags             |           |                                |
|   * Count                    |     2     |                                |
| * References                 |           |                                |
|   * Count                    |    20     |                                |
|     * Branches               |     2     |                                |
|     * Tags                   |     2     |                                |
|     * Remote-tracking refs   |    15     |                                |
|     * Git stash              |     1     |                                |
|                              |           |                                |
| Biggest objects              |           |                                |
| * Commits                    |           |                                |
|   * Maximum size         [1] |  1.50 KiB |                                |
|   * Maximum parents      [2] |     2     |                                |
| * Trees                      |           |                                |
|   * Maximum entries      [3] |   209     |                                |
| * Blobs                      |           |                                |
|   * Maximum size         [4] |   145 MiB | ***************                |
|                              |           |                                |
| History structure            |           |                                |
| * Maximum history depth      |  1.02 k   |                                |
| * Maximum tag depth      [5] |     1     |                                |
|                              |           |                                |
| Biggest checkouts            |           |                                |
| * Number of directories  [6] |   711     |                                |
| * Maximum path depth     [7] |    11     | *                              |
| * Maximum path length    [7] |   154 B   | *                              |
| * Number of files        [6] |  3.64 k   |                                |
| * Total size of files    [8] |   652 MiB |                                |
| * Number of symlinks         |     0     |                                |
| * Number of submodules       |     0     |                                |
```

leoheck, 2022-12-15:
@worldwidemv thanks for reporting this, I never saw this issue happening before. Is it possible to share the repo with me, so I can give it a try by myself?

As a workaround, instead of changing your sources, you can use the flag `-t N` to show only the latest N commits. Can you check it this work for you?

Now, talking about the hash size,  I found this right now.

[image]

It looks like there is no standard short version of hash as I thought before. 

I am going to experiment with your solution that uses longer or the full shash and if it is good, then I think I am going to adopt it.

leoheck, 2022-12-15:
Ah, since git was using short hash, the reason to keep that was to be able to pass a simple hash to kidiff from the command line, if the user wanted to skip the use of the GUI, but it is fine to use the long hash since it is easy to copy and paste it, and I believe no one will write the hash character after character by hand.

worldwidemv, 2022-12-15:
Unfortunately I can not share the repository. Personally, I don`t believe the repo is so large, that it normally would need 8 digits statistically. But if the first 7 match somehow, they match...  

Apparently the short hash size is configurable via `core.abbrev` but the default is `auto` with 7 as the minimum. 

I already called `kiri -r -t 3 -D X.kicad_pro`.

Using the full 40 digits hash also might be to much, because the short hash is way easier to compare and it is also used in the web interface as kind of a version number and all that makes sense, as long the short hash is unambiguous.
I think GIT makes sure that is the case if the length is not overridden. So in my opinion, using the current GIT output is better than using the full hash.

And if there is no reason to force `kidiff` to use only 7 digits then this would be fix.

leoheck, 2022-12-15:
Yeah, the current `Kicad-Diff` submodule `kiri` is using is my fork of it so I can have more control over it. 

The code to fix that is already been pushed, you can go ahead and update the Kicad-Diff submodule inside the kiri repo.

Kiri is still using a short version of the hash, please, check again if it works or if it already needs to be fixed.

leoheck, 2022-12-15:
fixing Kiri is not straightforward, since it messes up the design a bit, and this may take a bit of thinkering. but let me know if it needs to be fixed, or not, cool?

Example of what I am talking about. Not hard to improve but some details need some focus. 
[image]

leoheck, 2022-12-15:
Well, I started by increasing the hash size of kiri to 8, it this is not enough, we can try 10. This may not interfere with the layout too much that I have to do too many changes. =)

worldwidemv, 2022-12-15:
I don't think is necessary to force that. GIT does it automatically if necessary. The different parts of `kiri`just have to support it.

So the changes to `kidiff` are sufficient in my opinion. I would even revert the changes to line 106 in `git.py`. But it does not seam to hurt....
Forcing 8 digits might might not work for repos, that need 9 digits to be unambiguous.

I tested 1455d9646f7a1f73679b2dd69f3dccff82714766, and that works fine.

Gasman2014, 2022-12-15:
The chances of a hash collision for a 7 character hash is low (1% for around 2500 commits) and I've never made anything like number of commits in a hardware design but clearly its possible!

The only reason I trimmed to 7 digits was purely for layout/display reasons. I have no objection to this being increased - although I think displaying a huge hash is visually a bit overwhelming.

leoheck, 2022-12-15:
true, but in kidiff, I saw it was not a big deal. 

@worldwidemv it is good to keep the line 106 as it is now, since it will match the other places to keep the coherency.

@Gasman2014 I am about to make a pr with this new code to the main

about kiri, yeah, it looks like it is ok to keep using the default short hash as you said, I am going to revert this.

leoheck, 2022-12-15:
ah @worldwidemv now I got it, the way kidiff was getting the hash was a bit hardcoded. I am going to use the short version there too since it is automatic. the issue was the array-slicing method we were using. But now, it is slicing by the whitespace. Then we can use the short version nicely.

worldwidemv, 2022-12-15:
Hm. When I looked at the `commit1`  variable there was only the hash inside.
But, I don't know much Python.

Next time I will add a "how to recreate" section. 

Thanks for the fix. 
Have a nice Christmas.

leoheck, 2022-12-15:
You too @worldwidemv , have a nice Christmas.

closed 2022-12-16 by leoheck