leoheck/kiri #66: plot_kicad_sch distribution based locale error

robeg94, 2022-12-16:
I am running kiri on WSL2 `Ubuntu 20.04.5 LTS (GNU/Linux 5.15.79.1-microsoft-standard-WSL2 x86_64)`

When I run `plot_kicad_sch` I get the following error:

```
~/.local/share/kiri/bin/plot_kicad_sch: line 144: warning: setlocale: LC_ALL: cannot change locale (C.utf8): No such file or directory
   |
   | Display server = x11
   | Schematic path = PWM6FPC.kicad_sch
   |    Output path = .
   |  Kicad project = PWM6FPC.kicad_pro
   |
   | Launching Eeschema...
   | - Window 6291474 title = " (failure in conversion from UTF8_STRING to ANSI_X3.4-1968)]"
   |   Dismissing window
   Status: 124
```

This seems to be a similar issue to https://github.com/leoheck/kiri/issues/63, but is a slightly different root cause. I followed some of the same troubleshooting steps and seemed to find a temporary solution.

Output of `locale -a`:
```
C
C.UTF-8
POSIX
en_US
en_US.iso88591
en_US.utf8
```
The locale `C.utf8` does not seem to exist on my system.

I fixed it by modifying the executable in `~/.local/share/kiri/bin/plot_kicad_sch` changing line 144 from `export LC_ALL=C.utf8` to `export LC_ALL=C.UTF-8`.

You may want to add a handle for the different variations as it does not seem to be standardized between distributions (at least according to [a post I saw on StackExchange](https://unix.stackexchange.com/questions/564303/sort-lc-all-c-vs-lc-all-c-utf8/564305#564305)).

leoheck, 2022-12-16:
Well well well, windows is always doing something nasty haha 
Thanks for reporting this.

leoheck, 2022-12-16:
I saw you an I have POSIX in this list, could you try if it works setting POSIX instead?

leoheck, 2022-12-16:
This is what I have here, by the way.
```
➜  kiri-docker git:(main) ✗ locale -a
C
C.utf8
de_DE.utf8
en_AG
en_AG.utf8
en_AU.utf8
en_BW.utf8
en_CA.utf8
en_DK.utf8
en_GB.utf8
en_HK.utf8
en_IE.utf8
en_IL
en_IL.utf8
en_IN
en_IN.utf8
en_NG
en_NG.utf8
en_NZ.utf8
en_PH.utf8
en_SG.utf8
en_US.utf8
en_ZA.utf8
en_ZM
en_ZM.utf8
en_ZW.utf8
POSIX
pt_BR.utf8
pt_PT.utf8
```

leoheck, 2022-12-16:
I am going to use my magic and do this

```bash
C_UTF8=$(locale -a | grep -i "C.utf" | head)
export LC_ALL=${C_UTF8}
```

Please, if you test this let me know if it works.

leoheck, 2022-12-16:
I pushed this change already, I believe it will work on WSL too. If not, please, let me know.

closed 2022-12-16 by leoheck

robeg94, 2022-12-16:
Thanks for getting back to me so quickly! Windows always has to be "unique" haha

I tried `export LC_ALL=POSIX` and I still get the same error.

I tried
```
C_UTF8=$(locale -a | grep -i "C.utf" | head)
export LC_ALL=${C_UTF8}
```
and it works! I've been clearing the output files between attempts with `kiri -r -t 2 -dp PWM6FPC.kicad_pro`, so you seem to have found a solution.

robeg94, 2022-12-16:
I just pulled the new code and it works great too!

leoheck, 2022-12-16:
Awesome! And thank you for reporting and helping me.