leoheck/kiri #67: realpath errors

edahlseng, 2023-01-06:
I recently downloaded the latest version of kiri and tried running it for the first time in a while. I'm getting multiple of the following errors when running on macOS:

```
realpath: illegal option -- -
usage: realpath [-q] [path ...]
```

kiri version:
```
kiri df1308b 
kicad 6.0.9-0
plotgitsch v0.9.0-8-ge5f94e4
kidiff f0143ce 
```

leoheck, 2023-01-06:
Thanks for the feedback, haven't tested it on macos yet. I will definitely have to adapt something, maybe not realpath only. I will check this later, today.

leoheck, 2023-01-07:
Wait I thought this was about Kiri Docker. Can you send me the command line you are using? Running Kiri on MacOS is good as far as I know, but I have to check if the latest changes did not break anything.

leoheck, 2023-01-07:
Ah, I see. It looks like you don't have `coreutils` installed. 
Have you run the `install_dependencies.sh` script?
This may solve all your dependency issues.

leoheck, 2023-01-07:
Please, reopen this if you still have issues related.

closed 2023-01-07 by leoheck

mhavelka77, 2023-07-31:
Hi!
I also encountered this problem:

```
martin@macbookair % kiri -d ./ ./MUX/mux.pro

Artiffacts Summary
21 commits (210 comparisons)

Generating artifacts (may take some time)
  [[hidden commit details]]
realpath: illegal option -- -
usage: realpath [-q] [path ...]
realpath: illegal option -- -
usage: realpath [-q] [path ...]
    [[hidden commit details]]
realpath: illegal option -- -
usage: realpath [-q] [path ...]
realpath: illegal option -- -
usage: realpath [-q] [path ...]
   [[hidden commit details]]
realpath: illegal option -- -
usage: realpath [-q] [path ...]
realpath: illegal option -- -
usage: realpath [-q] [path ...]
    [[hidden commit details]]
realpath: illegal option -- -
usage: realpath [-q] [path ...]
realpath: illegal option -- -
usage: realpath [-q] [path ...]
   [[hidden commit details]]
realpath: illegal option -- -
usage: realpath [-q] [path ...]
realpath: illegal option -- -
usage: realpath [-q] [path ...]
```

Version:
```bash
    kiri 81add14 
    kicad 7.0.5-0
    plotgitsch v0.9.0-8-ge5f94e4
    kidiff 8f5df90 
```

I am running MacOS Ventura 13.0.1 on M1 Macbook Air. 

I installed the dependencies as described in the installation guide. The realpath used on my system doesn't seem to be the one installed with `brew install coreutils`:

```bash
martin@macbookair harmonizer % which realpath
/bin/realpath
```

I also did some light debugging of the main `kiri` bash script and I think that this error message does not originate from the `realpath` command invocation in this script. It probably comes from somewhere deeper.

mhavelka77, 2023-07-31:
I found that the intended binary `realpath` from the coreutils package is actually (at least on my system) called `grealpath` and is located in `/opt/homebrew/bin/grealpath`.

The issue described can be temporarily resolved by searching for the usage of this binary in the script and replacing it with the coreutils variant.

leoheck, 2023-07-31:
Yeah, macos is a bit different. But I tried to do something to normalize the tools.

Let me check here how can we fix this.

Meanwhile, you could get the version of it for me please?

```bash
➜ /opt/homebrew/bin/grealpath --version
```

leoheck, 2023-07-31:
Ah, could you please give me the value of this variable, 
```
echo ${OSTYPE}
```

leoheck, 2023-07-31:
I tried to fix it here, please update your kiri setup and let me know if it works.

leoheck, 2023-07-31:
thanks, I will fix this as soon as possible an then I will let you to know. I am far from my computer right now.