leoheck/kiri #68: Empty output

edahlseng, 2023-01-06:
When running kiri on a project I'm getting an empty output. I notice that no artifacts are being generated (I see the following output):

```
Artiffacts Summary
0 commits, totaling 0 possible comparisons
```

This directory does have several commits however. The KiCad project file is in a subdirectory; does that have anything to do with this issue?

leoheck, 2023-01-06:
Kiri should work with projects in subdirectories. Ah, right now as far as I remember to achieve that you have to do something special when running kiri docker. 

Can you share the command line you are using?

I tried to address that in one example in the README. Did you try to follow the example I have shown there?

leoheck, 2023-01-07:
You were having issues with dependencies, once you fix those issues, then you may be able to run Kiri. If you still have issues with the generated files, let me know.

closed 2023-01-07 by leoheck