leoheck/kiri #71: kicad 7?

MarcelRobitaille, 2023-03-22:
The README lists support for kicad 5 and kicad 6. Is there support for kicad 7? If not, is it planned? As far as I know, the file format changed again in 6 -> 7.

leoheck, 2023-03-22:
Hi @MarcelRobitaille 

Which file are you saying is not the same anymore?

As I know Kicad file format is still the same. I haven't tried it yet, but the way Kiri works it should work with Kicad 7 out of the box unless there are some issues in the code. 

Did you try Kiri with a Kicad 7 project? Do you have a repo with Kicad 7 so I can check that too?

I am working to use kicad-cli tools to plot schematics, since the current way it works with a GUI workaround for Kicad 6/7.

But the current version should work. However, I did not try to check it directly, yet.

MarcelRobitaille, 2023-03-22:
Hi leoheck. Thanks for your answer. No, I didn't try it yet. It seems not trivial to set up, so I didn't want to go through all of that if it wasn't going to work. I will try it now.

leoheck, 2023-03-23:
Yeah, it should work out of the box for Kicad 7.

I tested it myself on a project that I converted it from the Kicad 6.
For Schematic since Kicad 6 does not have this kicad-cli, it won't use it for Kicad 6 and 7 files. It will use the GUI with the xdotool which is a workaround. 
I am preparing a branch to test this new behavior if kicad-cli is present. It is almost there, but there is some work to be done yet.

Now, to install it, it is just a matter of copying/pasting 2 commands from the Readme.

leoheck, 2023-03-23:
@MarcelRobitaille, there is this branch `kicad-cli` that you can test. This should plot schematics of Kicad 6 and 7 better-using kicad-cli (from Kicad 7) if it is installed in your system. Hope it works for you.

MarcelRobitaille, 2023-03-23:
Thanks @leoheck 

> Now, to install it, it is just a matter of copying/pasting 2 commands from the Readme.

Those scripts don't support my system.

leoheck, 2023-03-23:
> Those scripts don't support my system.

Which system?

MarcelRobitaille, 2023-03-23:
Arch Linux. I was able to install it now, but I'm still running into #72

leoheck, 2023-03-25:
What is the package manager of Arch? I could add support..

leoheck, 2023-03-25:
Maybe it would be better since I am trying the `kicad-cli` branch and it looks good here. 
I use Ubuntu 22.10.

Could you also check it using my design here? https://github.com/leoheck/assoc-board
I am using it to test.

leoheck, 2023-03-26:
This looks working for me, but feel free to reopen the ticket if you still have problems.

closed 2023-03-26 by leoheck

leoheck, 2023-03-26:
Ah, the kicad-cli branch was merged on main branch if you wish to test it one more time, you could use it from there.

MarcelRobitaille, 2023-03-27:
> What is the package manager of Arch? I could add support..

It is pacman. However, it would be more idiomatic to package the software for the aur user repositories. I might be able to work on this.

leoheck, 2023-03-27:
Not sure if this is good. Kiri is a script that works on top of a bunch of other tools. 
I can start supporting it in the dependencies_installer, but I was not able to install arch easily. 
Do you know if it has an easy-to-install distro?

Changing topics, could you make kiri work?

MarcelRobitaille, 2023-03-27:
> Kiri is a script that works on top of a bunch of other tools.

That shouldn't be a problem. Kiri can depend on those packages and they will automatically be installed. I think some of them are also not in the AUR, in which case I would package those up too.

>  Do you know if it has an easy-to-install distro?

There are some arch-based distros that are easier to install: Manjaro, Garuda, and a few others I can't remember.

> Changing topics, could you make kiri work?

No, I'm still having the problem discussed in #72

leoheck, 2023-03-27:
I tried Manjaro, but using gnome-boxes I could not install it yet. I have to figure out why Manjaro is not finding any disks. But I will update my install_dependencies to have kiri installed with packman. If you did something to make that happen and want to share, this could speed up things for me.

FPiorski, 2023-08-23:
`install_plugin.sh` needs to be updated for everything to work out of the box

leoheck, 2023-08-23:
Hi @FPiorski I did not understand what you are requesting. Please, open a new issue so we can discuss it better, there.