leoheck/kiri #72: 404 on every schematic / PCB SVG

MarcelRobitaille, 2023-03-23:
Sorry to bother you again. I'm getting 404 on every schematic SVG. The webpage loads fine otherwise and I'm able to see all my commits. However, I'm not able to see any schematics or PCB.

MarcelRobitaille, 2023-03-23:
I am still getting this actually. Now, I have a folder `.kiri/<hash>/sch` and `.kiri/<hash>/pcb`, but the frontend seems to be looking for `.kiri/<hash>/kiri/pcb`. I have an empty folder `.kiri/<hash>/kiri/pcb` but `.kiri/<hash>/kiri/sch` is missing.

leoheck, 2023-03-23:
Run kiri one time with the flag -r to remove old files. And let me know if it is ok.

leoheck, 2023-03-23:
I ran it with one of my projects and it with working fine. I have to investigate then. 

Can you use the flags `-r -V -t 3` and share the output here?

leoheck, 2023-03-26:
@MarcelRobitaille this looks good from my side. Feel free to reopen the issue with extra details if the issue was not fixed for you.

leoheck, 2023-03-27:
Please, try to update your Kiri using the latest main branch. Ah, maybe your installation did not when well, and maybe there is something missing since you are using Arch.  Do you have your packman history to share here with me so I can update the install_dependencies for your case.. and then we could work together until kiri is working on your system.

leoheck, 2023-03-28:
There is also an issue with the `rename` command. It should be the `perl-rename`... and it should be. It is fixed in the master too.

leoheck, 2023-03-28:
Now it seems to be working. This is the latest commit, `25404d7cf9e1c4ac9353bcbd799e5ec613555aa2`

leoheck, 2023-03-28:
Still have to check how to install Kicad 7 and (check why xdotool is not plotting Kicad 6 files)

leoheck, 2023-03-28:
@MarcelRobitaille  Thanks for your help so far. 

Is Kiri already working for you? Do you still have issues running it?

closed 2023-03-28 by leoheck

MarcelRobitaille, 2023-03-29:
I'm still trying to get it working. I had problems with the graphical method of plotting the schematic. `plot_kicad_sch` was always exiting with 124. Now I'm trying to use the `kicad-cli`.

MarcelRobitaille, 2023-03-29:
OK, everything seems to be working now with `kicad-cli`. I had to make some changes to the code though. See #76

leoheck, 2023-03-29:
Cool awesome.