leoheck/kiri #80: Kiri sometimes does not overwrite _local_

MarcelRobitaille, 2023-03-31:
When running kiri for the second time, I noticed that I don't have uncommitted changes in the browser. Kiri is also not generating any artifacts even though there are local changes.

I think it's because of [this line](https://github.com/leoheck/kiri/blob/main/bin/kiri#L1826). It makes sense to skip generating artifacts if the commit hash exists, because the hash will change if the commit is rebased, but this does not handle new local changes.

leoheck, 2023-03-31:
If Kiri already passed through the commit, creaking it's folder it doesn't
do anything else. It assumes everything is created.

To (re)generate everything clean, you should use the flag `-r` or manually
deleting that folder or deleting .kiri folder (but this is carried by -r)

MarcelRobitaille, 2023-03-31:
I don't want to regenerate everything though as that would take a long time. I only want to regenerate `_local_`. I solved this by manually deleting `.kiri/_local_` and running kiri without `-r`.

I think [this line](https://github.com/leoheck/kiri/blob/main/bin/kiri#L1826) should be something like:
```bash
if [[ -d "${OUTPUT_DIR_PATH}/${hash_2}" -a "${hash_2}" != "_local_" ]]; then
```

leoheck, 2023-03-31:
Yeah glad you said that too. I came here to comment exactly that after
reflecting for a while.

MarcelRobitaille, 2023-03-31:
Would you accept such a pull request?

leoheck, 2023-03-31:
Sure, please, go ahead.

MarcelRobitaille, 2023-03-31:
Ok, it is more involved than I thought. With the change I proposed above, kiri is regenerating a lot of existing artifacts. I do not really understand why.

I see that in `generate_diffs` there is a nested for loop. Could you please help me understand why? I thought we would only need to iterate through every commit once and generate its artifacts. I know the frontend loads the svg for 2 commits separately and does the diff on the frontend. What is the need for these two loops. I think this is what's causing a lot of work to be redone with the change I proposed.

leoheck, 2023-03-31:
I will have to investigate later. But it is indeed an interesting proposal since it allows the constant work and check.

Now, you could test and use this which may mimic the intended behavior:

```bash
rm -rf .kiri/_local_/ && Kiri -V
```

This would work as you are expecting, I think. If this doesn't work, then kiri has a fundamental behavior that doesn't allow that in the first place. Then we should investigate this deeply.

Also, another approach, would be this:

```bash
kiri -r -t 2
|```

which will delete the folder but it will build the last 2 commits only

leoheck, 2023-04-01:
@MarcelRobitaille I think I could put this to work. Can you check if the `local-rev` branch works for you?
I already made a PR https://github.com/leoheck/kiri/pull/81

leoheck, 2023-04-01:
Hey, I thought it was good, then I pushed it to the master, you can test it from there, now.

MarcelRobitaille, 2023-04-01:
Yeah that's one way to do it too. I think it would work; that's how I solved it manually. I will check it out. Thanks!

closed 2023-04-01 by MarcelRobitaille