leoheck/kiri #84: Fails to report error from plot_kicad_pcb.py

MarcelRobitaille, 2023-05-11:
When I run kiri on my project, everything seems like it succeeds. Plotting the PCB with `kidiff` doesn't print any error and exits 0. However, when I go into the browser, I can't see the PCB. In the dev tools, the request for the PCB SVG gives 404.

I dug into this problem, and tracked the problem down to `plot_kicad_pcb.py`. If I run this directly, I get:
```
> Changing path: Piezo_HV_Pulser.kicad_pcb
Wrong version of the API
Try sourcing 'env-nightly.sh' instead.
```

I think the problem is that `plot_kicad_pcb.py` is printing on stdout and exiting 1. However, `kidiff.py` only detects errors if `plot_kicad_pcb.py` prints on stderr, and doesn't check the status code.

I think `plot_kicad_pcb.py` should print on stderr and I think that `kidiff.py` should check the exit status.

MarcelRobitaille, 2023-05-11:
It would probably be better to show the error created by `pn.LoadBoard` instead of the bare except.

leoheck, 2023-05-12:
Do you have any suggestions since you have a repo with this issue that you can test? Or, can you share the repo? It can be through email if you prefer.

MarcelRobitaille, 2023-05-13:
I created [a pull request](https://github.com/leoheck/KiCad-Diff/pull/3) in kicad-diff. 

To be clear, I eventually solved the root cause of my problem. I opened this issue because I think the troubleshooting could have been easier with a few small changes.

closed 2023-05-17 by leoheck