leoheck/kiri #88: Diff between branches or long spanning history.

topherbuckley, 2023-07-24:
I've got a long history of commits (>100) spanning between two versions of a project (e.g. v1 and v2). I don't care about the individual diffs, but just want to show someone the diff between v1 and v2. They each sit on their own branch tip, and I don't want to necessarily generate a kiri diff for each commit. Is there a way kiri can specific two particular commit hashes to just take a diff between them? 

I suppose I could just make a temporary branch and squash everything for this purpose, but wondered if there was a way to do this with kiri alone? 

Cheers! Looks like this is progressing well since I last checked in on it.

topherbuckley, 2023-07-24:
I was able to get the schematic diffs for what I wanted by squashing everything into a temporary branch, but the pcb layout is not generated for the earlier commit for some reason. I'd presume this is because it was using kicad 6 while the newer version is on kicad 7. I guess I need to import the kicad 6 version into kicad 7 before I can do the pcb diff?

leoheck, 2023-07-24:
I think that comparing different branches is not possible yet. I never seen anyone with this request. But it can be added, maybe. Is it fine to work with Kicad with multiple branches.

Now, for comparing 2 versions you have to o use the hash of the comimt. you can get them with `kiri ls` I guess.

Then the comparison will be made with commits inside in between those hashes, including them, if I ma not mistaken. If the mag of comparing less then 2 commits appear, try to add the next hash (having for instance a list of 3), since it may be a big.

leoheck, 2023-07-25:
Hey, @topherbuckley I added one extra parameter to allow comparing between branches as git does. Could you check if it works for you?

The syntax is this one.

```
kiri -g HEAD..HEAD~1
```

topherbuckley, 2023-07-28:
> Hey, @topherbuckley I added one extra parameter to allow comparing between branches as git does. Could you check if it works for you?
> 
> The syntax is this one.
> 
> ```
> kiri -g HEAD..HEAD~1
> 

Brilliant! Yep, this works as expected. I tested both on a linear history and two diverged branches. Works as expected. Thanks!

Is there a quicker way to update with git btw? 

Or do you also run:

`bash -c "$(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_dependencies.sh)"`

followed by `bash -c "INSTALL_KIRI_REMOTELLY=1; \
    $(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_kiri.sh)"` for every commit you do? 
    
I'm much more git friendly, so if you have a dev approach to updating, I'd like to know so I can pull in your commits easier. 
```

>This can be a bug. Could you create a new issue so we can discuss this better there?

Done. See [here](https://github.com/leoheck/kiri/issues/89)

closed 2023-07-28 by topherbuckley

topherbuckley, 2023-07-28:
> Is it fine to work with Kicad with multiple branches.

Sorry, forgot to respond to this. Yeah, it works fine if set up correctly. Although spanning multiple versions of KiCAD can make it a pain with the config files madness you saw me questioning in the kicad forums a month or so back. I'm reevaluating my way to track libraries though, so maybe that will work better when I start moving more toward the more common workflow. 

There are a couple of things to keep track of though other than the config files. Kicad does a few things only on close/open of files (i.e. no menu or refresh behavior) so when you swap commits/branches you need to close the project before or after this checkout. 

For this case, I had an old version (e.g. v1) that I thought needed to be saved as a KiCAD 7.0 project before I could do the kiri diffs but now that I tested your `-g` flag, I see you've outdone yourself and kiri can compare across versions as well! :open_mouth: :clap: :clap:) 

so I had a tag on v1, made a new branch just to update the config files etc. to 7.0. Then I have branch_v1, and branch_v2 on diverging branches, and was looking for a way to compare this (thus this question began). 

I also use branches for working with other people. Theres definitely some things that can cause headaches for merge conflicts, but its definitely doable if you setup the project and manage it properly. I was doing schematic design and another person doing the pcb layout, and so long as you kept your commits clean (i.e. don't include changes to the PCB if working on the schematic, and vice versa) it worked quite well.

leoheck, 2023-07-29:
Nice, thank you for this detailed answer. 

> There are a couple of things to keep track of though other than the config files. Kicad does a few things only on close/open of files (i.e. no menu or refresh behavior) so when you swap commits/branches you need to close the project before or after this checkout.

Indeed, I always think this is a bug or something haha

> I also use branches for working with other people. Theres definitely some things that can cause headaches for merge conflicts, but its definitely doable if you setup the project and manage it [...]

I mean, if the branch does not have any other parallel work, then it works like a linear repo.. and then it is easy, but I always thought that having parallel branches would be hard to handle when conflicts arise since we work on a GUI and then the diffs are on Text. Which are different places to put your head in. Also, there are always different kinds of people/coworkers working and they are not always easy with text like us. Actually, I love Kicad because it has easy-to-use hotkeys (they are about to destroy this) and also it has files that are readable text, not binary blobs.

> Is there a quicker way to update with git btw?

Hmm, I keep it always in the same place so it is always updated for me.
But I think you can go to the folder where it is installed, and then ```git pull```.
This would be enough 90% of the time.. unless a new dependency was added then I think that running that command is easier.