leoheck/kiri #89: PCB diffs not appearing.

topherbuckley, 2023-07-28:
Tangent from [here](https://github.com/leoheck/kiri/issues/88). I'm seeing schematic diffs, but not PCB ones. 

Some screenshots:
[image]

[image]

Here is the command that got me there:

```
~/git/robot-pcb/kicad/robot$ kiri -g HEAD..9fc7137 robot.kicad_pro
Commit 1: 00fc7f4 | 2023-07-27 19:51:33 | Christopher Buckley | Adding DNP bulk capacitance to SYS and Vboost rails. Some populared (HEAD)
Commit 2: 9fc7137 | 2023-06-06 16:37:47 | Christopher Buckley | Adding gitignore for config (9fc7137)

Artiffacts Summary
2 commits (1 comparisons)

Generating artifacts (may take some time)

Assembling the HTML
cp: cannot stat '/home/bo/.local/share/kiri/assets/gltfc': No such file or directory
- Adding list of commits
- Adding list of pages
- Adding list of layers

Latest rev 00fc7f4
- Project's title (folder name): robot-pcb
- Project's name (file name): robot
- Sch title: [missing]
- PCB title: [missing]
- Sch revision: [missing]
- PCB revision: [missing]
- Sch date: [missing]
- PCB date: [missing]

Kiri main page
- Initial project path:
  Pro 1: .kiri/00fc7f4/robot.kicad_pro
  Pro 2: .kiri/9fc7137/robot.kicad_pro

Starting webserver at http://127.0.0.1:8080/web/index.html
(Hit Ctrl+C to exit)
```
Tried a few other use cases as well:
e.g. `kiri --last 5 robot.kicad_pro` but with the same result.

leoheck, 2023-07-28:
Try to use it like this
 
```bash
kiri -r -V --last 5
```

- Enable the -r flag to remove leftovers from previous runs. If a previous run fails then the left overs won't allow kiri to run it again on the same commit.
- Enable the `-V` to increase the verbosity so you will see every command kiri uses on each step... then you can run the command manually if it has failed, to test it.
- This is extra but.. the project file is not needed unless you have multiple .kicad_pro/.pro in the same folder.

topherbuckley, 2023-07-28:
Not seeing any obvious error values/logs:

```
~/git/robot-pcb/kicad/robot$ kiri -r -V --last 5
Using Kicad project ./robot.kicad_pro

          KIRI_HOME_PATH = /home/bo/.local/share/kiri
           KICAD_VERSION = 7.0.6-7.0.6~ubuntu22.04.1
            CURRENT_PATH = /media/christopher/HDD/git/robot-pcb/kicad/robot
 KICAD_PRO_ABSOLUTE_PATH = /media/christopher/HDD/git/robot-pcb/kicad/robot/robot.kicad_pro
      REPO_ABSOLUTE_PATH = /media/christopher/HDD/git/robot-pcb
  KICAD_PROJ_NESTED_PATH = kicad/robot
KICAD_PROJ_ABSOLUTE_PATH = /media/christopher/HDD/git/robot-pcb/kicad/robot
         OUTPUT_DIR_PATH = /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri
            WEB_DIR_PATH = /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/web/

    # Rev robot files
         project_path: /media/christopher/HDD/git/robot-pcb/kicad/robot
              project: robot.kicad_pro
            schematic: robot.kicad_sch
      extra_sch_files: Charger.kicad_sch
                       LDO.kicad_sch
                       Protection.kicad_sch
                       USB-PD-ExtChg.kicad_sch
                       USB-PD-PhonePC.kicad_sch
                       USB-PD_Circuit.kicad_sch
                       USB-PD_Msc.kicad_sch
                       boostRegulator.kicad_sch
                       mcu/RP2040_minimal/RP2040_minimal.kicad_sch
                       motors.kicad_sch
                       powerMux.kicad_sch
                       usb-c-extChg.kicad_sch
                       usb-c-phonepc.kicad_sch
               layout: robot.kicad_pcb

Output folder: .kiri
Removing leftovers from the previous run
Get commits command
   git log --date=format:'%Y-%m-%d %H:%M:%S' --pretty='format:%h | %ad | %an | %s' pcb -- \
   robot.kicad_sch Charger.kicad_sch \
LDO.kicad_sch Protection.kicad_sch USB-PD-ExtChg.kicad_sch USB-PD-PhonePC.kicad_sch \
USB-PD_Circuit.kicad_sch USB-PD_Msc.kicad_sch boostRegulator.kicad_sch \
mcu/RP2040_minimal/RP2040_minimal.kicad_sch motors.kicad_sch powerMux.kicad_sch robot.kicad_sch \
usb-c-extChg.kicad_sch usb-c-phonepc.kicad_sch robot.kicad_pcb

Commits list
     1	00fc7f4    2023-07-27 19:51:33    Christopher Buckley    Adding DNP bulk capacitance to SY
     2	aa34d62    2023-07-27 19:31:24    Christopher Buckley    Eleminated 5V rail and corrected 
     3	c0ef2fa    2023-07-26 16:13:18    Christopher Buckley    Moving phone, motors, and battery
     4	347a8fa    2023-07-26 09:43:25    Christopher Buckley    Flipping battery to other side of
     5	27150ec    2023-07-26 00:14:24    Christopher Buckley    Adding pixel3a to kicad model

Artiffacts Summary
5 commits (10 comparisons)

Generating artifacts (may take some time)

-----------------------------------------------------------------------------------------------------

# 1/10 00fc7f4 | 2023-07-27 19:51:33 | Christopher Buckley | Adding DNP bulk capacitance to 
# 2/10 aa34d62 | 2023-07-27 19:31:24 | Christopher Buckley | Eleminated 5V rail and correcte

    # Rev 00fc7f4 files
         project_path: /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/00fc7f4
              project: robot.kicad_pro
            schematic: robot.kicad_sch
      extra_sch_files: Charger.kicad_sch
                       LDO.kicad_sch
                       Protection.kicad_sch
                       USB-PD-ExtChg.kicad_sch
[... 255 lines cut ...]
http://127.0.0.1:8080/00fc7f4/_KIRI_/pcb/layer-00.svg?t=00fc7f4

	
GET
	http://127.0.0.1:8080/00fc7f4/_KIRI_/pcb/layer-00.svg?t=00fc7f4
Status
404
File not found
VersionHTTP/1
Transferred654 B (469 B size)
Referrer Policystrict-origin-when-cross-origin
Request PriorityHighest

XHRGET
http://127.0.0.1:8080/aa34d62/_KIRI_/pcb/layer-00.svg?t=aa34d62

XHRGET
http://127.0.0.1:8080/00fc7f4/_KIRI_/pcb/layer-00.svg?t=00fc7f4

XHRGET
http://127.0.0.1:8080/aa34d62/_KIRI_/pcb/layer-00.svg?t=aa34d62

```

Let me know if there is something else from the dev console that would be helpful.

leoheck, 2023-07-29:
Hm, the dev view is not always helpful because it is ok to have missing svg images.

Is it possible to share this project with me so I can give it a look if I have the same issue, here?

leoheck, 2023-07-29:
You can manually check if there are svg files in those folders
```
ls -lsa /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/00fc7f4/_KIRI_/pcb/
ls -lsa /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/27150ec/_KIRI_/pcb/
```

You should see something like this
[image]

leoheck, 2023-07-29:
Also check the `kidiff` version, you have, please.

I guess you can use this command to get its version
```bash
cd $(dirname $(which kidiff)); git log -n1 | cat; cd -
```

Mine is, 
```bash
➜ cd $(dirname $(which kidiff)); git log -n1 | cat; cd -
───────┬─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
       │ STDIN
───────┼─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
   1   │ commit 8f5df9008ab0ffda8d3c2f4255dc058448156beb
   2   │ Author: Leandro Heck <leoheck@gmail.com>
   3   │ Date:   Mon May 29 16:55:56 2023 -0300
   4   │ 
   5   │     Fix popt.SetSvgPrecision for Kicad >= 7.0.5
───────┴─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
```

leoheck, 2023-07-29:
Offtopic, your log helped me to find a small issue with labels swapped. 
[image]

It is fixed already, you can do a ``git pull`` where kiri is installed 
```bash
cd $(dirname $(dirname $(which kiri))) | git pit pull
```

leoheck, 2023-07-29:
If you run this
```bash
kidiff -k -n --webserver-disable -a 00fc7f4 -b 27150ec -o TEST -f robot.kicad_pcb
```

Do you see svg files inside of these folders?
```bash
ls -lsa TEST/00fc7f4/pcb/
ls -lsa TEST/27150ec/pcb/
```

topherbuckley, 2023-07-30:
Thanks for all the ideas. Glad to see you are so active on these issues!

> Is it possible to share this project with me so I can give it a look if I have the same issue, here?

Unfortunately, I cannot share this one. I tried making a sandbox project to share in its stead, but found it worked for the sandbox project, so it definitely looks like a project-specific issue and not an issue with the kiri installation. 

> You can manually check if there are svg files in those folders

Interestingly I see this instead:

```
ls .kiri/00fc7f4/_KIRI_/pcb
robot-00.svg  robot-33.svg  robot-38.svg  robot-43.svg  robot-48.svg  robot-53.svg
robot-01.svg  robot-34.svg  robot-39.svg  robot-44.svg  robot-49.svg  robot-56.svg
robot-02.svg  robot-35.svg  robot-40.svg  robot-45.svg  robot-50.svg  robot-57.svg
robot-31.svg  robot-36.svg  robot-41.svg  robot-46.svg  robot-51.svg  robot-58.svg
robot-32.svg  robot-37.svg  robot-42.svg  robot-47.svg  robot-52.svg  robot.kicad_pcb
```

The dev console says it is looking for "layer-00.svg". Is there a hardcoded call somewhere to "layer-00.svg" instead a variable holding the svg names? Alternatively, should my svg files be named "layer-00.svg" instead of "robot-00.svg" and the code generating these has an issue?

> Also check the `kidiff` version, you have, please.

Same:
```
commit 8f5df9008ab0ffda8d3c2f4255dc058448156beb
Author: Leandro Heck <leoheck@gmail.com>
Date:   Mon May 29 16:55:56 2023 -0300

    Fix popt.SetSvgPrecision for Kicad >= 7.0.5
```

> If you run this
> 
> ```shell
> kidiff -k -n --webserver-disable -a 00fc7f4 -b 27150ec -o TEST -f robot.kicad_pcb
> ```
> 
> Do you see svg files inside of these folders?
> 
> ```shell
> ls -lsa TEST/00fc7f4/pcb/
> ls -lsa TEST/27150ec/pcb/
> ```

Same as before:

```
ls TEST/00fc7f4/pcb/
robot-00.svg  robot-33.svg  robot-38.svg  robot-43.svg  robot-48.svg  robot-53.svg
robot-01.svg  robot-34.svg  robot-39.svg  robot-44.svg  robot-49.svg  robot-56.svg
robot-02.svg  robot-35.svg  robot-40.svg  robot-45.svg  robot-50.svg  robot-57.svg
robot-31.svg  robot-36.svg  robot-41.svg  robot-46.svg  robot-51.svg  robot-58.svg
robot-32.svg  robot-37.svg  robot-42.svg  robot-47.svg  robot-52.svg  robot.kicad_pcb
```

leoheck, 2023-07-31:
> Unfortunately, I cannot share this one. I tried making a sandbox project to share in its stead, but found it worked for the sandbox project, so it definitely looks like a project-specific issue and not an issue with the kiri installation.

hm, indeed. 

I think you could make a new branch, then one commit removing the content of the main schematics and also also the other extra schematics pages, and also removing footprints in the layout but leaving something inside and share this single commit with me. so I can check if I can generate the board?... or maybe remove your board and reimport footprints and then commit it to check if kiri can generate the images of the layout on this commit.

leoheck, 2023-07-31:
Ah, I think I could find something.. 

I am renaming the `[board]-xx.svg` files to `layer-xx.svg` with the `rename` tool.
[image]

You may not have the same `rename` tool as Kiri expects:

The expected `rename` is a perl tool.
```
 rename --version      
/usr/bin/rename using File::Rename version 2.01
```

So, what is your operating system? Could you check the rename version too?

leoheck, 2023-07-31:
https://unix.stackexchange.com/questions/730894/what-are-the-different-versions-of-the-rename-command-how-do-i-use-the-perl-ver

leoheck, 2023-08-01:
@topherbuckley any luck resolving this?

leoheck, 2023-08-03:
Hi. I am closing this issue report since it seems that it is related to the setup of the environment. Please, feel free to reopen the issue if necessary.

closed 2023-08-03 by leoheck

leoheck, 2023-08-05:
@topherbuckley hey, how are you doing? I just added a command line flag to force plot layout using `kicad-cli` instead of `kidiff`. It is slow, but renaming is not necessary. But may be ok for your particular problem.

topherbuckley, 2023-08-09:
> enaming the `[board]-xx.svg` files to `layer-xx.svg` wit

Hi @leoheck sorry for the delay. Got absorbed into other work for a while. 

Yes, the rename program appears to be different:

```
rename --version
/usr/bin/rename using File::Rename version 1.30, File::Rename::Options version 1.10
```
Looks like this is the newest version on apt (Ubuntu package manager). 

I'm using Ubuntu 22.04. That rename versioning looks like quite the mess. Can you not just use `sed` instead of rename? Or is rename something that is supposed to have better cross-platform support? 

I tried installing via cpm (I've never done anything significant with pearl, so treading lightly here). I used the recommended method of `cpan -i File::Rename` and used all the defaults (local::lib, added something to my .bashrc file) then had to `. .bashrc` then got

```
rename -V
/home/bo/perl5/bin/rename using File::Rename version 2.01
```
But the issue persists. Same renaming to robot-##.svg and firefox looking for layer-##.svg. Is the design intent for these output files to always be layer-##.svg or the project_name-##.svg? I can see the code you quoted for the renaming, but where is the code that tries to display these output files in the server? I'm curious if they are hardcoded to look for ones named layer-##.svg or if the variable that stores the project_name is being used.

topherbuckley, 2023-08-09:
Also just noting that if I go in and forcefully rename all the svg files, everything shows up as expected. So it looks like the code that renames them should always rename them to layer-##.svg or the code calling them in the server to display should hold the variable for the project name used for project_name-##.svg.

topherbuckley, 2023-08-09:
Digging in a bit deeper, I added 

`echo "# kicad_project_name = ${kicad_project_name}"` 

before the rename calls, and it is empty for me, so the substitution fails.

topherbuckley, 2023-08-09:
The '-k' flag works for me. I noticed in your 'generate_layout_artifacts' you have an if-else logic to separate these two generators, but only the kidiff section seems to use this renaming scheme. When I used the sandbox project I made, it seemed to default to the kicad-cli without this `-k` flag.  Looking at your commit associated with the flag (404de4e8e225e51651c1881a9d276f3ee1e87395) it looks like the sandbox project was evaluating to true for the `if [[ -f $(which kicad-cli 2> /dev/null) ]] && [[ "${KIDIFF_MISSING}" == 1 ]];` while my problematic project was evaluating that to false. Not sure if that helps diagnose this or not, but just as far as I can get with the time I can allot to this. Cheers!

topherbuckley, 2023-08-09:
A few more things I found:

If I removed the &> /dev/null from the rename calls. I find out that there is no pcb directory:

```
Can't rename /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/885e8c6/pcb/robot-53.svg /media/christopher/HDD/git/layer-pcb/kicad/robot/.kiri/885e8c6/pcb/robot-53.svg: No such file or directory
Can't rename /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/885e8c6/pcb/robot-56.svg /media/christopher/HDD/git/layer-pcb/kicad/robot/.kiri/885e8c6/pcb/robot-56.svg: No such file or directory
Can't rename /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/885e8c6/pcb/robot-57.svg /media/christopher/HDD/git/layer-pcb/kicad/robot/.kiri/885e8c6/pcb/robot-57.svg: No such file or directory
Can't rename /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/885e8c6/pcb/robot-58.svg /media/christopher/HDD/git/layer-pcb/kicad/robot/.kiri/885e8c6/pcb/robot-58.svg: No such file or directory

```
I see you commented out the creating of this a [few lines up](https://github.com/leoheck/kiri/blob/2dc77b2ba686022b5f3f3c813246d4cf4c792254/bin/kiri#L1816) from there. Is there some other place this is occurring now? I can check where this is being created for the sandbox project and try to figure out why the pcb directory is not being created for my problematic case.

topherbuckley, 2023-08-10:
I'm at a bit of a loss. I added the following above the rename calls:

```bash
echo "OUTPUT_DIR_PATH = ${OUTPUT_DIR_PATH}"
			echo "kicad_project_name_1 = ${kicad_project_name_1}"
			echo "commit_hash_1 = ${commit_hash_1}"
			echo "------------------listing files in pcb dir.-----------------------------------"
			ls -al "${OUTPUT_DIR_PATH}/${commit_hash_1}/pcb/"*.svg
			echo "rename version:"
			rename -V
                        read -p "Press Enter to continue..."
			# Rename files folowing format: layer-xx.svg
			echo "trying rename just robot-00.svg"
                        rename -n "s/robot-/layer-/" "${OUTPUT_DIR_PATH}/${commit_hash_1}/pcb/robot-00.svg"
                        read -p "Press Enter to continue..."
			rename -n "s/${kicad_project_name_1}-/layer-/" "${OUTPUT_DIR_PATH}/${commit_hash_1}/pcb/"*.svg 
                        read -p "Press Enter to continue..."
			rename "s/${kicad_project_name_1}-/layer-/" "${OUTPUT_DIR_PATH}/${commit_hash_1}/pcb/"*.svg 
			rename "s/${kicad_project_name_2}-/layer-/" "${OUTPUT_DIR_PATH}/${commit_hash_2}/pcb/"*.svg 
                        read -p "Press Enter to continue..."
```
and can even execute the same rename calls upon pausing execution with the `read` commands and it works, but for some reason within the context of the script it is failing. The most notable oddity I see is that the rename function errors out and spits out the same name for the input and output (i.e. I'd expect one of the paths to point to layer-53.svg even if the operation failed) but both point to robot-53.svg). 

Here is the output from those debugging calls:

```
---------------------------Test 115---------------------------
/media/christopher/HDD/git/robot-pcb/kicad/robot
OUTPUT_DIR_PATH = /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri
kicad_project_name_1 = robot
commit_hash_1 = _local_
------------------listing files in pcb dir.-----------------------------------
-rw-rw-r-- 1 bo bo 1665291 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-00.svg
-rw-rw-r-- 1 bo bo  680486 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-01.svg
-rw-rw-r-- 1 bo bo  703573 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-02.svg
-rw-rw-r-- 1 bo bo  855076 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-31.svg
-rw-rw-r-- 1 bo bo   79006 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-32.svg
-rw-rw-r-- 1 bo bo   79006 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-33.svg
-rw-rw-r-- 1 bo bo  110900 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-34.svg
-rw-rw-r-- 1 bo bo  633179 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-35.svg
-rw-rw-r-- 1 bo bo  272150 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-36.svg
-rw-rw-r-- 1 bo bo  324583 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-37.svg
-rw-rw-r-- 1 bo bo  129491 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-38.svg
-rw-rw-r-- 1 bo bo  648135 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-39.svg
-rw-rw-r-- 1 bo bo  670122 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-40.svg
-rw-rw-r-- 1 bo bo   87572 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-41.svg
-rw-rw-r-- 1 bo bo   79486 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-42.svg
-rw-rw-r-- 1 bo bo  180534 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-43.svg
-rw-rw-r-- 1 bo bo   79530 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-44.svg
-rw-rw-r-- 1 bo bo   79006 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-45.svg
-rw-rw-r-- 1 bo bo   85868 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-46.svg
-rw-rw-r-- 1 bo bo  146109 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-47.svg
-rw-rw-r-- 1 bo bo  145330 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-48.svg
-rw-rw-r-- 1 bo bo  804875 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-49.svg
-rw-rw-r-- 1 bo bo   79006 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-50.svg
-rw-rw-r-- 1 bo bo   79006 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-51.svg
-rw-rw-r-- 1 bo bo   79006 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-52.svg
-rw-rw-r-- 1 bo bo   79006 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-53.svg
-rw-rw-r-- 1 bo bo   79006 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-56.svg
-rw-rw-r-- 1 bo bo   79006 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-57.svg
-rw-rw-r-- 1 bo bo   80284 Aug  9 20:38 /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/_local_/pcb/robot-58.svg
rename version:
/home/bo/perl5/bin/rename using File::Rename version 2.01

[... cut ...]
Can't rename /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/885e8c6/pcb/robot-57.svg /media/christopher/HDD/git/layer-pcb/kicad/robot/.kiri/885e8c6/pcb/robot-57.svg: No such file or directory
Can't rename /media/christopher/HDD/git/robot-pcb/kicad/robot/.kiri/885e8c6/pcb/robot-58.svg /media/christopher/HDD/git/layer-pcb/kicad/robot/.kiri/885e8c6/pcb/robot-58.svg: No such file or directory
Press Enter to continue...
```

I'll think about this more later, but let me know if you see any other things I can check.

leoheck, 2023-08-10:
Did you install the dependencies with this command?
```bash
bash -c "$(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_dependencies.sh)"
```

The rename command is indeed a bit complicated.
See here, this is the normalization I am doing for each distro.
But it should work fine on Ubuntu 22.
[image]

You can check if at least the rename is like mine regarding the version of it
[image]

Then you could replace the `rename()` function above, on line 480, with this one:
```bash
rename()
{
	/usr/bin/file-rename "$@"
}
```

leoheck, 2023-08-10:
Using apt you can find the correct rename tool

[image]

But I am on Ubuntu 23.04, so you have to check yourself, maybe it is different today.
```bash
sudo apt search "^rename"
```

topherbuckley, 2023-08-10:
> Using apt you can find the correct rename tool
> 
> [image]
> 
> But I am on Ubuntu 23.04, so you have to check yourself, maybe it is different today.
> 
> ```shell
> sudo apt search "^rename"
> ```

Yep that difference only appears to apply to 23.04:

See package listings [here](https://packages.ubuntu.com/search?suite=default&section=all&arch=any&keywords=rename&searchon=names)

That being said, I still 

> You can check if at least the rename is like mine regarding the version of it

That all looks similar, though I installed the 2.01 version locally rather than system-wide. I checked `rename -V` in the bash script as well before the call to rename, and it matched the 2.01, so I don't think it is being interpreted as something else. 
```
~/git/kiri/bin$ which rename
/home/bo/perl5/bin/rename
~/git/kiri/bin$ ls -lsa /home/bo/perl5/bin/rename 
8 -r-xr-xr-x 1 bo bo 4279 Aug  9 16:48 /home/bo/perl5/bin/rename
~/git/kiri/bin$ /home/bo/perl5/bin/rename -v
Usage:
    rename [ -h|-m|-V ] [ -v ] [ -0 ] [ -n ] [ -f ] [ -d ] [ -u [enc]]
    [ -e|-E perlexpr]*|perlexpr [ files ]

~/git/kiri/bin$ /home/bo/perl5/bin/rename -V
/home/bo/perl5/bin/rename using File::Rename version 2.01

~/git/kiri/bin$ file /home/bo/perl5/bin/rename
/home/bo/perl5/bin/rename: Perl script text executable
```

> Did you install the dependencies with this command?

Yep. I did that, but tried again to check for updates and nothing looked to have updated or downloaded (everything was already newest version). The only thing that I didn't do that might be noteworthy is it claims "Now run 'opam upgrade' to apply any package updates." I have never run that. Do you? Anyways, I did it, and th only upgrades were for the `dune`, `dune-configurator`, and `ca-certs-nss` packages. After which I ran `~/git/kiri/bin/kiri -V -r --last 2` with the same issue persisting.

topherbuckley, 2023-08-10:
Figured it out! See [this PR](https://github.com/leoheck/kiri/pull/93) for the fix. 

Take a 2nd look at the error made by rename:

>Can't rename /media/christopher/HDD/git/**robot-pcb**/kicad/robot/.kiri/885e8c6/pcb/**robot-46.svg** /media/christopher/HDD/git/**layer-pcb**/kicad/robot/.kiri/885e8c6/pcb/**robot-46.svg**: No such file or directory

The regex was matching another part of my path (i.e. robot-pcb) and substituting that rather than the filename at the end of the path. I added the end-of-line regex and the digits to make the matching robust against this issue.