leoheck/kiri #90: realpath and xargs use GNU only extensions that don't work on MacOS

mhavelka77, 2023-07-31:
In several places in the project, there is the binary `realpath` with the `--relative-to` option used.
For example in the `kiri` script, line 2161:
```bash
# Initial Kicad project path
	local project_path_1="$(grealpath --relative-to="${KICAD_PROJ_ABSOLUTE_PATH}" "${OUTPUT_DIR_PATH}/${commit_hash_1}/${kicad_proj_name}.${kicad_pro_extension}")"
	local project_path_2="$(grealpath --relative-to="${KICAD_PROJ_ABSOLUTE_PATH}" "${OUTPUT_DIR_PATH}/${commit_hash_2}/${kicad_proj_name}.${kicad_pro_extension}")"
```

Similarly, the script also uses the `xargs` binary with the `-d` option for delimiter specification.
Line 177:
```bash
local page_filename=$(echo "${page_filename_path}" | gxargs -d '\n' basename)
```

Those binaries are however not the same on MacOS and unfortunately cause the following errors:

realpath:
```
realpath: illegal option -- -
```

xargs:
```
xargs: illegal option -- d
```

Even though the correct binaries are installed with the setup script via homebrew, they are called `grealpath` and `gxargs`. Temporary solution is thus manually changing all occurences of `realpath` and `xargs` to `grealpath` and `gxargs` in the codebase.

The `realpath` issue has already been discussed in #67

leoheck, 2023-07-31:
Thanks for opening this new issue, this way is a bit better. I've pushed a fix, could you try it? And also thanks for testing/using Kiri with the MacOS.

mhavelka77, 2023-07-31:
Confirmed, its working flawlessly now after following the installation guide. Thank you! 

Closing the issue.

closed 2023-07-31 by mhavelka77

leoheck, 2023-07-31:
Nice, thank you.