leoheck/kiri #91: Mouseclicks do not update view

bobatsar, 2023-08-04:
Some things seem to be broken in the current version.

When selecting a sheet on the right side with the mouse, the view does not update.
If I use the keyboard shortcuts, everything works fine (selecting revisions, sheets, layers, ...).

Also clicking on the "Info" button does nothing. Here also the `i` shortcut does not work.

I am using kiri on Ubuntu 23.04, Kicad 7 with chrome
But anyhow, thanks for the great tool!

leoheck, 2023-08-04:
Hi @bobatsar, keyboard usage is the main goal, actually as you could see. 
But thanks for reporting it, I am going to check if I can improve these things somehow.

leoheck, 2023-08-04:
These things should be working. They are not working for me too. I also use chrome. Which version do you have?

bobatsar, 2023-08-04:
I have Chrome 115.0.5790.170 and tested also with Firefox 115.0.2 (64-Bit)
Both do not work.

leoheck, 2023-08-05:
Hi @bobatsar I found out the issue with the `i` was introduced some time ago. And the issue where it is not possible to change the schematic page by clicking it on the list was never addressed by me. 

But I think I could fix those two things now.

Could you update your kiri to check if the problem still persists for you?

closed 2023-08-09 by leoheck

bobatsar, 2023-08-09:
@leoheck I just checked it and both issues are now fixed. Thanks!