leoheck/kiri #94: Layout diff gets cropped

FPiorski, 2023-08-23:
I start my PCBs at (0, 0) and go right and up, so out of the drawing sheet - I got that from someone online, much cleaner than starting at a random point inside the drawing sheet, alas, it doesn't work well in KiRI as of right now.

I checked and both kidiff and kicad-cli have no problems with generating the correct SVGs, yet I can't see anything in KiRI. Through trial and error I found that if I resize the browser window (change the window aspect ratio from the almost-full-screen, almost-16:9 to something with the same height but smaller width (so make it more vertical)) then I'm able to see at least part of the diff (see pic attached). If I make the window vertical enough then I can see enough of the top right quarter of the (KiCad) coords, but still no top left (had some dimension markings on F.Fab extending into the negatives, that's how I noticed).

Anyways, after looking around for a bit I think it might be a problem in svg-pan-zoom applying a viewbox based on the svg element's size (derived from the browser window size), but I couldn't really find an easy workaround - not well versed in this stuff at all.

[image]

leoheck, 2023-08-23:
Hmm, this is a weird thing to do. I never saw anyone doing this. But it should not be a problem. 

By designing this way the board is going to be outside of the sheet which is a bit bad since Kiri is plotting the sheet (and things inside of it). There is also a flag that disables the page frame `-f` or `--page-frame`. Could you check if this works for you? I am not sure if this will work and it may have bugs since it is something that was added a long time ago. But it is worth trying, at least. 

I am curious why do you need to work this way? Do you know Kicad has a feature that you can hit the Spacebar, and it will mark the point under the cursor as `0,0`. Then, if you move the cursor it will show the relative distances. These are the numbers you can see right in the middle of the status bar with the prefixed `dx` and `dy`.

If you share your project or a similar project with me I can verify this behavior on my computer and try to do something regarding it, eventually.

leoheck, 2023-08-23:
I was curious about your issue, I could see it happening.
Yeah, the issue is that you are placing the design outside of the sheet.
And no other svg viewer like this.

As you can see, inkscape can see my design.
[image]

However gnome-eye does not show out of the sheet borders
[image]

The same happens with browsers as Firefox, here:
[image]

And google-chrome.. 
[image]

I will try to implement something that you can use to tweak your files I am just not sure how long it will take since I tried something similar in the past when kicad was generating svgs with issues. 

Now, honestly, you should design inside the sheet not outside of the sheet haha. What you are doing is like taking notes on the table even if you have the notebook in front of you.

FPiorski, 2023-08-23:
I agree that it's an edge case, so all in all I'm happy with my workaround of changing the browser window size to be able to see the diff - to that end, this issue can be treated more as an information for anyone trying to figure out why stuff's not working for them as opposed to an actual bug report, I know the first place I look when something's not working right is GitHub issues so maybe others do too. So with that in mind, feel free to ignore the rest of this comment.

Playing the devil's advocate for a bit, the options as I see them are:
1. start at a random place inside the drawing sheet,
2. start at a random place and place the grid origin where you start ('S'),
3. start at (0,0) and go right and up,
4. start at (0,0) and go right and down.

I started with option 1, but it isn't really that neat because you're starting in an arbitrary spot.
Option 2 helps, you have a permanent offset from (0,0) for your grid, so you can place the grid origin in some important location on your arbitrarily located PCB and that helps to keep things aligned when switching between grid sizes. You're still stuck with the actual positions of footprints and tracks and all the other objects being referenced to (0,0) instead of the placed grid origin, which is the correct behavior if you think about it, but it means you're stuck with doing relative moves with Shift+M instead of editing the element coordinates themselves because those have an arbitrary offset that isn't easy to work around in your head.
So, with option 2 going with Shift+M is your only sane choice, but with option 3, you can either do Shift+M or just edit the coordinates themselves, since the PCB starts at (0,0) and there's no arbitrary offset - you have two choices and can choose the one that makes more sense in context, instead of being stuck with only one. The only downside of option 3 is that you're going outside of the drawing sheet, so technically it's not what the devs intended, but it works just fine. Up until now I wasn't aware of a situation where this approach wouldn't "just work", but there we have it - KiRI - and after thinking about it I fully agree that it makes sense for it to not work, the PCB is off the drawing sheet after all.
Right, so option 4 should be the best of both worlds and it would be, if not for the fact that the drawing sheet obscures the PCB. You can hide the drawing sheet, but that setting is kept in a .kicad_prl, which is not something you'd want to keep in VCS, plus I've had problems with those appearance settings not loading right recently.
So yeah, I'm now considering switching to option 4 - right now I can just resize the window and get KiRI to cooperate, but since the PCB being inside the drawing sheet is by all means a sane assumption, sticking with option 3 might come back to bite me in the future when trying to get my projects to work with some other awesome third-party project.
Oh, and for reference, I switched from starting in an arbitrary place to starting at (0,0) after looking at one of MicroType Engineering's KiCad projects, so while it is an edge case, if other people got inspired too, it might not be as rare as one might think.

Phew, sorry for the very long comment, but you did ask :smiley: 
>I am curious why do you need to work this way?

leoheck, 2023-08-23:
Dude this is crazy af. You are over complicating something really really really simple.

Back to kiri, I think it is possible to tweak the svg pan and zoom to start showing the viewport before of the origin. Then the workaround can be an easy change in the index.html or even a cli parameter setting the canvas.

Something related with what you can find here:
https://vanseodesign.com/web-design/svg-viewport/

You can help tweaking the generated html so it changes the viewport to where you need... if you achieve that you can show me how and I can add a parameter for you to customize this when launching kiri.

leoheck, 2023-08-24:
I found out that you can tweak this by adding this viewbox property on the svg in `kiri.js` in the highlighted lines.

[image]

The syntax should be this one

[image]

Now you will have to play a bit to check the right values for you. I could not figure yet good values to put there automatically, but at least I could see things outside of the sheet

[image]

leoheck, 2023-08-24:
This is the best I could find to help you, now.
[image]

FPiorski, 2023-08-24:
I played around for a bit and couldn't really find a way to automatically get the required viewbox for a given svg file, so I think I'm just going to add a static "viewbox -1000 -1000 1000 1000" or something in my install of kiri.js and call it a day, thanks for all the help!

leoheck, 2023-08-24:
Ah, that would be interesting too, but I believe the svg file does not have the correct viewbox, at least the xmin/ymin

If I find a way to get the svg width and height automatically, I could create a command line parameter to change this xmin/ymin

leoheck, 2023-08-24:
This is the svg generated by the PCBNew, I will try to get those settings with javascript
[image]

leoheck, 2023-08-24:
saving this for me here
https://typeofnan.dev/how-to-perfectly-fit-an-svg-to-its-contents-using-javascript/

FPiorski, 2023-08-25:
I checked the viewbox in svg files generated both with kidiff and kicad-cli and they both ignore the out-of-bounds PCB, so I think that line is just statically generated by looking at the sheet size.
Yeah, that window resize thing is weird, that's how I got the screenshot in the first post.
I tried running the getBBox() method on a couple of HTML elements in the javascript console but it didn't return useful data for any of them (I tried IDs of svg-id, img-1, img-2, and even diff-xlink-1 and diff-xlink-2), that's what I mean by "played around for a bit" in the last comment, though I may just have been doing something wrong.

leoheck, 2023-08-25:
yeah, I am playing around with these things right now, too.. but nothing seems to work.. if the browser is a small window something makes it work and we can see things out of the sheet, but if the browser is a bit bigger maximized then it doesn't work

leoheck, 2023-08-25:
an alternative to this would be post-process svg files with a cli tool... I used to fix svgs in the past but I could not find my previous script yet.

FPiorski, 2023-08-25:
Ok, trying to run getBBox() on any form of `<img>` tag doesn't work, but opening up just the SVG file in chrome and running getBBox() on that returned the actual bounding box (with negative x and y values and all).
Apparently, there's no way to get to the underlying SVG element from JS when using `<img>`, but it *is* possible to do so when using an `<object>` tag. So I added a temporary new object element in createNewEmbed() and appended it to diff-container (or wherever, just to get it in the HTML). I couldn't get it to work when I had this new object's style set to "display:none", but when I didn't, running `document.getElementById("svg-temp1").contentDocument.getElementsByTagName('svg')[0].getBBox()` from the javascript console returned the proper bounding box.
Yeah, it's far from actually getting the bounding box in code, but it's the first step, plus setting the viewbox based on the svg bounding box would not only fix my issue but would also result in autofilling the screen with the PCB diff, which would be a cool feature.

leoheck, 2023-08-25:
That's a great progress, I will try to replicate here later. Meanwhile, if you hit `f` you can see the whole diff area fullscreen.

leoheck, 2023-08-28:
@FPiorski I am starting to think this whole issue is being created by Kicad.

Even if there are shapes out of the sheet, it does not update the viewBox correctly. 
Its viewBox always starts with 0,0

This is for instance the svg viewport (width and height) and the viewBox of the layer-31.svg of my test project.

```html
<svg
  xmlns:svg="http://www.w3.org/2000/svg"
  xmlns="http://www.w3.org/2000/svg"
  xmlns:xlink="http://www.w3.org/1999/xlink"
  version="1.1"
  width="297.002mm" height="210.007mm" viewBox="0.000 0.000 297.002 210.007">
```

```html
  width="297.002mm" height="237.217mm" viewBox="0.000 -27.210 297.002 237.217">
```

leoheck, 2023-08-28:
I have a workaround for you (`svg_tweaks` script) where you can manually set the initial x and y of your svg instead.
You can find it currently on this https://github.com/leoheck/kiri/tree/toogle-new-old branch

This is a suggestion of use, right now:
```bash
x=-50; y=-100; kiri -r -l && find .kiri -name "*.svg" -exec svg_tweaks {} ${x} ${y} \; && kiri
```

You will have to set x and y with the coordinates that have your design inside, then it will run kiri, without launching the webserver, then it will fix the svgs, and then it will launch kiri webserver with the fixed svgs.

I will have to think about how to integrate this inside kiri by having a flag to set x and y directly, later.

leoheck, 2023-08-28:
Ah, ignore the last message, I was able to create a quick setup for this.
This is still in the same branch (https://github.com/leoheck/kiri/tree/toogle-new-old)

```
-sxy|--sch_xy x,y  Set x,y of svg viewBox for Schematics (workaround for problematic SVG's of Kicad)
-pxy|--pcb_xy x,y  Set x,y of svg viewBox for PCBs (workaround for problematic SVG's of Kicad)
 ```

It would be nice to hear from you if this helps you to visualize your pcbs.

Here is a quick example
```bash
kiri -r -pxy -100,-100
```

leoheck, 2023-08-28:
I am going to close this issue since this solution is good enough, at least I think it is. 
Feel free to open it again if there are extra things to discuss.

I also think that an automatic solution for tweaking each svg independently would not be good since the files could be misaligned eventually. This could also take a lot of time to find the leftmost x,y for each svg with the tools I know.

closed 2023-08-28 by leoheck

leoheck, 2023-08-28:
Ah, just let me know if this solution works for you, then I will merge this branch.

leoheck, 2023-08-30:
Hey, this feature was merged in the main already.