leoheck/kiri #95: Update install_plugin.sh for KiCad 7.0

FPiorski, 2023-08-23:
I commented on #71 ("kicad 7?") about the `install_plugin.sh` file needing to be updated for KiCad 7.0, but with the context of all the other discussion under that issue, the comment relating to KiCad 7.0 compatibility could be misunderstood, so, as requested, I'm opening up another issue:

While I didn't really test running KiRI through the plugin button yet, I can already report that for it to even show up in KiCad 7.0, it needs to be copied to the correct folder, but the `install_plugin.sh` script ran by the KiRI installer script only supports KiCad 5.1 and 6.0 - for 6.0 the plugin path is hardcoded:
```bash
	if [[ -d "$HOME/.local/share/kicad/6.0/" ]]; then (...)
```
The path only differs by the version number and everything else is the same, so all that is needed to fix it is to copy a couple of lines and replace "6.0" by "7.0", but until then the plugin won't install in KiCad 7.0.

closed 2023-08-24 by leoheck

leoheck, 2023-08-24:
Interesting, thanks for the feedback. I never used `.bash_profile`.  What is your Operating System and version, by the way?

FPiorski, 2023-08-25:
Debian 12.1 with no DE, just Xorg + i3wm on top of that, which might have something to do with it or not. KiCad 7.0.7 compiled from source.