leoheck/kiri #99: svg from plotgitsch in wrong directory

chrisnoisel, 2023-09-19:
Hello,
I'm running Kiri with old schematics that require plotgitsch. During the launch I've dozen of windows popping on my screen showing each svg (which seem valid) for *each* project commit (it's ... a lot). After the processing is done, every svg is reported as 404 in the web interface. In the project directory, svg are left in the middle of the schematics (with prefix from_* or to_*) instead of the .kiri directory

**My setup**
I run Arch Linux, Kiri is installed from AUR : https://aur.archlinux.org/packages/kiri-git
plotgitsch wasn't properly compiled. So I built and installed it separately from https://github.com/jnavila/plotkicadsch 
I've got Kicad 7, I've still to figure out why python can't find the pcbnew import but it seems unrelated to the issue, no other error reported with "kiri -V -r [...]"

Do you know what may be missing ? You mentioned in an other issue that you have forked plotgitsch but it doesn't show in the .gitmodules so I'm not sure it still true. 

Thanks

leoheck, 2023-09-19:
Hello @chrisnoisel 

I am not responsible for this AUR repo so I cannot add much here. 

I recommend you install it with the instructions found in the `README.md`. The installation is pretty straightforward. You just have to run 2 scripts (copy and paste). One is going to install the basic dependencies and the second one will install Kiri (with `Kidiff` and `plotgithsh`). If you do that, you can share the issues with me, then I will fix the installer if needed. However, if you prefer to fix yourself with the AUR package, feel I won't be able to update my scripts to share them with others that use Arch and have the same issues you have.

> During the launch I've dozen of windows popping on my screen showing each .sch svg (they look ok) for each project commit (it's ... a lot)

This seems to be something broken on kiri however I could not reproduce it by my side.. kiri should not show svgs while running (unless maybe, you are using the wrong version of plotgitsh or maybe my fix script is not in front in the PATH of one that plotgitsh runs). 

> I run Arch Linux, Kiri is installed from AUR : https://aur.archlinux.org/packages/kiri-git plotgitsch wasn't properly compiled. So I built and installed it separately from https://github.com/jnavila/plotkicadsch

Alright, so schematics are good to go, right?

For layout, Kiri uses Kidiff, which uses Python module PCBNew. This module comes with Kicad. Now, Kicad 7 may have changed something for Arch or the AUR maintainer could have done that too.  So you have to play a bit with `kidiff` to check if you can tune it to make it work. 

The latest kiri with kicad 7 has the option to plot the layout using kicad-cli. It may work for you. But it is not the default since it takes ages. But if you want to give it a try before of fixing kidiff/pcbnew module

```
     -k|--kicad-cli   Force kicad-cli to plot layout
```

To test everything, you can go inside your kicad repo and run kiri with `-V` flag. So it will show every command it uses to generate each step. Then you can use those commands independently to try to fix each tool for each step.

For instance:
```bash
kiri -r -V
```

This will remove old runs and show the commands.

For instance... 

[image]

Then you are going to see the command in Yellow (here) and the directory from which it runs that command. 
Then you can rerun this command inside the given directory to evaluate if the tools are working.

leoheck, 2023-09-19:
> You mentioned in an other issue that you have forked plotgitsch but it doesn't show in the .gitmodules so I'm not sure it still true.

I forked it in the past, but I think I am not using my version anymore.

Plotgitsh uses this `git-imgdiff` script, so I have a version that should be in front of it in the PATH.
Check from where this script comes with `which git-imgdiff`
It should come from `.../kiri/bin/` folder.

chrisnoisel, 2023-09-19:
Hi @leoheck 

> I am not responsible for this AUR repo so I cannot add much here.
Yeah don't worry I'm just here to have some clues as the project has multiple layers of dependencies, it's not easy to track everything. If I can pinpoint the problem I might contribute to the AUR later.

> Alright, so schematics are good to go, right?
It's the other way around, sorry if I wasn't clear enough. PCB layouts are good (both with --kicad and Kidiff, I've fixed the import pcbnew). The problem are the schematics.

> Check from where this script comes with which git-imgdiff
> It should come from .../kiri/bin/ folder.
And that's it ! Displaying was the initial behavior of git-imgdiff I didn't get that.

It seem to work now, thank you !

I've still to think about how I'd like to package that. Ideally I'd rather not override the git-imgdiff in the .bashrc from a perfectly fine standalone plotgitsch install, I guess that would be patched at runtime, what do you think ?

leoheck, 2023-09-19:
Good! 

> I guess that would be patched at runtime, what do you think?

This script is used inside `plotgitsch` tool. When I forked it I could replace it easily. But then everything would be redirected to my fork, and I would like to give the whole traffic to the original tool. 

I saw that I could override this ` git-imgdiff` with my custom script to make kiri happy =)

How would you suggest to patch this script at the runtime?
It has to be in the path so plotgitsch finds it.
Ah, I could source it at the runtime if this is what you are suggesting since it is inside kiri repo.
But then I would lose the ability to run plotgitsch commands in the command line easily, unless I add this as an instruction to each line.

chrisnoisel, 2023-09-19:
To me, if you don't use a specific fork, `plotgitsch` should be available in its original form from the command line since it's expected to follow its documentation/behavior etc. or that could be considered a bug, or disturb other programs relying on it.

Altering the program in command line should be an optional conscious decision from the user (like using a virtual env in python or LD_PRELOAD), that is : willingly setting up a bashrc or having a command modifying the PATH at will etc. 
I think Kiri could silently do something like that : internally appending the PATH with its own `git-imgdiff` first during runtime or plotgitsch call.

(Other solution : plotgitsch could provide an argument to set a path to a custom `git-imgdiff` script, that would be a clean solution)

leoheck, 2023-09-19:
>To me, if you don't use a specific fork, plotgitsch should be available in its original form from the command line since it's expected to follow its documentation/behavior etc. or that could be considered a bug, or disturb other programs relying on it.

Agree, but no one will use it if they are using kiri, honestly. I never heard anyone complaining. But I totally agree with you. this is why I chose to use the original version. You can override the path with the kiri's git-imgdiff whenever you want. 

As a workaround for you... you can remove the custom git-imgdiff from kiri/bin and put it somewhere else. And then if you want to run kiri you have to preload the PATH with this new place where the custom git-imgdiff is. This way you can have a customizable environment with the original plotgitsch. 

> (Other solution : plotgitsch could provide an argument to set a path to a custom git-imgdiff script, that would be a clean solution)

I tried many times, but the dev does not understand me or he does not want to help me. 

Then I forked it once, but after some time I chose to not fork it anymore. Forking it won't help anything since Kiri users are going to use my version o plotgitsch anyway, which can be a totally different app. Also, modifying plotgitsch is hard since that language (or maybe that code in particular or both) is a bit complex for me to understand. I would love to fork it and add this flag that adds an extra ability to change the script or even an environment variable, when set, to use a different tool to replace that script. If you feel that you can help me change it, I totally agree to do that. But I have failed many times. 

Another easy way out of this is to rename the plotgitsch/plotkicadsch to something else, whithing this fork.. then it will be a different tool, then the git-imagediff could be renamed as well. So we can have both repos concurrently installed. This could be another option and could be easier to make it work.

chrisnoisel, 2023-09-19:
> Another easy way out of this is to rename the plotgitsch/plotkicadsch to something else

I think that is more work than having Kiri switching the git-imgdiff when needed. It allows you to keep your bashrc config too.

Since I want to have Kiri managed by pacman, I'll probably create dependency packages instead of submodules with an install script. I'll see.

For now my issue is resolved. Thanks for your time !

closed 2023-09-19 by chrisnoisel