leoheck/kiri #102: Installing kiri on Ubuntu (Command 'kiri' not found, did you mean:)

radiolyc, 2023-10-06:
Hi! I'm trying to install **kiri** on Ubuntu system, but it doesn't work for me.
1) Firstly, I've just opened terminal and installed all dependencies:
`bash -c "$(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_dependencies.sh)"`
[image]
2) Then I installed **kiri** itself
`bash -c "INSTALL_KIRI_REMOTELLY=1; \
    $(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_kiri.sh)"`
[image]
3) But when I'm trying to call:
`kiri -h`
It says, that it doesn't exist.
[image]
I tried do it several times, but with no success. Maybe I miss something?

leoheck, 2023-10-06:
hi @radiolyc after installing It you have to customize your terminal.

Copy and paste these lines 

[image]

Or add them to the `~/.bashrc` so it will be available next time when you launch the terminal.

radiolyc, 2023-10-07:
@leoheck Thanks, now it works!
[image]
[image]

closed 2023-10-07 by radiolyc