leoheck/kiri #103: feature request: highlight pages and layers that have changes

milesvp, 2023-11-10:
Hey there! I just found Kiri and it's amazing. I'm starting to use it in my work flow, and there's one pretty big feature that's missing. When I view changes in the browser, I have no way of telling where the changes occur. I've been taking advantage of git to collaborate with KiCad, and this tool saves a ton of time, but some small changes can be easy to miss, it would be awesome to have a visual way to see where the changes occur so as not to miss minor but important things like, silkscreen changes on a board with 30 layers to look through. 

Also, forgive me if this isn't a good avenue to request features.

leoheck, 2023-11-10:
Hello @milesvp, thanks for your input.

Implementing the feature you suggested would be a valuable enhancement for Kiri. However, I'm currently facing a challenge in distinguishing relevant changes in the layout. For instance, merely removing or disabling layers without changes becomes intricate due to variations in attributes like UUIDs, which might not necessarily impact the visual layout.

Even when the images appear identical, alterations within the kicad_pcb can occur due to non-critical attribute modifications. Managing these nuances proves to be quite complex. Additionally, experimenting with a more effective color scheme to highlight differences poses its own set of challenges, and I lack a clear understanding of how to approach this optimally.

I am eager to elevate Kiri further, but I'm uncertain about the best strategy. If you have any insights or ideas to share that could guide me in refining the difference-highlighting feature or improving the color scheme, I would greatly appreciate your assistance.

milesvp, 2023-11-10:
Yeah, I was afraid you'd say that. It can be a challenge manually looking at diffs in git to see what might be a delta. I was hoping that maybe you already had this data since the layers/pages show a visual diff, and it was just a UI improvement. It occurs to me as I write this, that may just rendering magic in plotgitsch. I may have to poke around a little and see if there's any way to make this information jump out. If I can I'll definitely let you know.

leoheck, 2023-11-11:
The way how kiri works is this: it converts the schematic and the layout to a image (an svg image) the  after that it uses a tool to overlap those images with a specific color, and then parts that overlap are going to have a color that is a mixture of 2 colors. and parts that do not overlap are going to be colored according considering if they are new or old. I don't have an output saying that there is a change or where. It doesn't use git to do that. The git is used to access the versions versions of the flies.

However thinking here, I could manage to do what you are describing by diffing the exported images since they are .svgs but ignoring some parts of the file that have strings that always change. I will just have to study two generated images that don't have changes to understand if this is possible.

leoheck, 2023-11-15:
Perphaps there is a chance of doing this.

assoc-board/.kiri/5be11dd/_KIRI_/pcb on  main [$] …
➜ echo "Running path $(pwd)\n"
for i in *.svg; do
        id=$(echo $i | sed "s/layer-\(..\).svg/\1/g")
        diff -q -s \
                 layer-${id}.svg ../../../7aebd2e/_KIRI_/pcb/layer-${id}.svg
done

Running path /home/lheck/Documents/KICAD_PROJECTS/assoc-board/.kiri/5be11dd/_KIRI_/pcb

Files layer-00.svg and ../../../7aebd2e/_KIRI_/pcb/layer-00.svg differ
Files layer-01.svg and ../../../7aebd2e/_KIRI_/pcb/layer-01.svg differ
Files layer-02.svg and ../../../7aebd2e/_KIRI_/pcb/layer-02.svg differ
Files layer-31.svg and ../../../7aebd2e/_KIRI_/pcb/layer-31.svg differ
Files layer-32.svg and ../../../7aebd2e/_KIRI_/pcb/layer-32.svg are identical
Files layer-33.svg and ../../../7aebd2e/_KIRI_/pcb/layer-33.svg are identical
Files layer-34.svg and ../../../7aebd2e/_KIRI_/pcb/layer-34.svg differ
Files layer-35.svg and ../../../7aebd2e/_KIRI_/pcb/layer-35.svg are identical
Files layer-36.svg and ../../../7aebd2e/_KIRI_/pcb/layer-36.svg differ
Files layer-37.svg and ../../../7aebd2e/_KIRI_/pcb/layer-37.svg differ
Files layer-38.svg and ../../../7aebd2e/_KIRI_/pcb/layer-38.svg differ
Files layer-39.svg and ../../../7aebd2e/_KIRI_/pcb/layer-39.svg differ
Files layer-40.svg and ../../../7aebd2e/_KIRI_/pcb/layer-40.svg are identical
Files layer-41.svg and ../../../7aebd2e/_KIRI_/pcb/layer-41.svg are identical
Files layer-42.svg and ../../../7aebd2e/_KIRI_/pcb/layer-42.svg are identical
Files layer-43.svg and ../../../7aebd2e/_KIRI_/pcb/layer-43.svg are identical
Files layer-44.svg and ../../../7aebd2e/_KIRI_/pcb/layer-44.svg are identical
Files layer-45.svg and ../../../7aebd2e/_KIRI_/pcb/layer-45.svg are identical
Files layer-46.svg and ../../../7aebd2e/_KIRI_/pcb/layer-46.svg differ
Files layer-47.svg and ../../../7aebd2e/_KIRI_/pcb/layer-47.svg differ
Files layer-48.svg and ../../../7aebd2e/_KIRI_/pcb/layer-48.svg differ
Files layer-49.svg and ../../../7aebd2e/_KIRI_/pcb/layer-49.svg differ

leoheck, 2023-11-29:
Hi @milesvp I was experimenting with optimizing the svgs files for web. This is really good for performance while browsing the results. Then I saw the produced files are better for comparing and telling if they match or not. I am experimenting with making an improved version that hides layers that are equal. This is going to improve the overall experience.

You can also use/test it if you upgrade Kiri with the last changes.
You will have to have `svgo` in the path
And after running kiri, you can replace the `.kiri/web/kiri.js` with this new one https://www.dropbox.com/scl/fi/dbwhrxvt4okx9pzznks4v/kiri.js_test?rlkey=21lnj19ecsgz7zq2o6qnnhtyy&dl=0
You should also, rename the downloaded file from `kiri.js_test` to `kiri.js`

leoheck, 2023-11-29:
Note that this is not perfect, some layers can be removed. Some others without "visible" changes are still there. 

https://github.com/leoheck/kiri/assets/1277920/24e26d35-df14-42ae-8678-314e37473ae7

Nudelsalad, 2023-12-10:
Hi @leoheck,
would this be worth merging it into the master branch? Or what is the current state?
Btw did you do some further testing with svg optimizer from pip?

leoheck, 2023-12-11:
Hey @Nudelsalad, I am about to merge it. 

However, I did not feel the scour was making any improvement on the navigation as I felt with the svgo and the svgcleaner. Do you feel it works better for you? Or at least better than not optimizing anything? I am asking because this adds a bit of time when generating the outputs. 

But I have to update the installer script and also I would test it a bit more since the javascript was updated a bit and since it is a bit complex, those changes may break something. 

Also, I am not going to merge the color style of the diffs, since it is not quite good for me yet. That color matrix is hard haha. 

I will give it a look later today, cool?

Nudelsalad, 2023-12-12:
Thanks for asking, I felt that svgcleaner worked better than scour in terms of optimization and execution time. 
No problem

closed 2024-07-11 by leoheck