leoheck/kiri #104: Issues reported on KiBot integration

leoheck, 2023-11-28:
This discussion is happening here https://github.com/INTI-CMNB/KiBot/issues/391
I am just summarizing here the issues that are being listed there.

- dos2unix is really needed?
- when plotkicadsch is missing, then Xdotool is used. If it fails the keyboard is left changed to US.
- It would be nice if --kicad-cli could force the use of kicad-cli even for KiCad 5 files.
- Layout does not have changes when using this command `kiri -D --kicad-cli -l`
- Just a single instance of the schematics pages is being displayed, and this may hide changes (refdes, values maybe).
- Performance issues with kiri output on large projects with large schematics.

leoheck, 2023-12-11:
yeah but I cannot recordist this. are you using mouse?

Nudelsalad, 2023-12-12:
Yes I am using mouse. It appears to be only when hovering over the third item in the list. Sry that my mouse is not visible, but the flickering was always when hovering over the 3rd item below the schematic and pcb symbol

[Screencast from 12.12.2023 06:14:21.webm](https://github.com/leoheck/kiri/assets/64224137/9819204d-0c65-44ee-a4a1-2d5a3d06d5d6)

when selecting the third one, it is also flickering when hovering over the third one

closed 2024-07-11 by leoheck