leoheck/kiri #108: Hierarchical Schematic Sheets

MartinMechatronics, 2024-02-15:
It doesn't seem to be tracking hierarchical sheets correctly.  It only displays the root sheet.  Is this just a know limitation or is there something I am doing wrong?  Is there a best practices so the changes are displayed correctly?

***Hardware and OS*********************
MacBook Air
Apple M2
macOS Sonoma Version 14.2.1 (23C71)

*** KiCAD Version***********************
Application: KiCad Schematic Editor arm64 on arm64

Version: 7.0.10, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.13.0
	HarfBuzz 7.3.0
	FontConfig 2.14.2
	libcurl/8.4.0 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.12 nghttp2/1.55.1

Platform: macOS Sonoma Version 14.2.1 (Build 23C71), 64 bit, Little endian, wxMac

Build Info:
	Date: Dec 29 2023 04:31:34
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.82.0
	OCC: 7.7.0
	Curl: 7.77.0
	ngspice: 41
	Compiler: Clang 13.0.0 with C++ ABI 1002

Build settings:
	KICAD_SPICE=ON

*** Kiri Version*************************
kiri 7a938be 
    kicad 7.0.10
    plotgitsch v0.9.0-8-ge5f94e4
    kidiff be291c6 

*** Git Version*************************
git version 2.43.2

leoheck, 2024-02-15:
Hi @MartinMechatronics, it was working fine on my tests, but depending on your setup it could be a bug since this is a very confusing feature of Kicad. Also, if I am not mistaken someone made changes on this matter, recently so it can be also a new bug.

It looks like you are using the latest stable Kicad. Can you share a sample project where I can see the issue you are observing? Then I will be able tackle the problem.

MartinMechatronics, 2024-02-15:
Hi @leoheck. It's good to know it is supposed to work, mostly likely it's something I am doing wrong in how I have my projects structured and tracking them in git before I started using Kiri.  The projects that are a problem are more complicated and I can't share.  I will try creating a new project that uses hierarchical sheets and see how it tracks changes.  It is a little confusing of a feature and I'm not sure if I am using it as intended.  It didn't see it mentioned in your documentation and I didn't want to struggle to make something work that was never intended to.  Bravo on your work on this project, I think it's great. As a side note, it runs much better on Chrome than Safari. Thank's for your feedback, I will let you know how it works out.

leoheck, 2024-02-16:
>  I will try creating a new project that uses hierarchical sheets and see how it tracks changes.

Yeah, good, that's exactly what we need. Try to replicate your complicated project, at least considering the hierarchal sheets and files/folders organization.

My goal with Kiri is to support what I can, but sometimes users get creative haha. I will try my best if I have a sample project that mimics your design, but sometimes Kicad is just too much complicated internally for me to keep Kiri working on all scenarios known.

ahalekelly, 2024-03-13:
@leoheck I'm having this issue with my project too, also on MacOS 14.2.1 and Apple Silicon but on Kicad 8.0. I tried with a couple other projects on Github, and was able to reproduce the issue with [this project](https://github.com/Teslafly/Moxie-Drive-TO220-x24/tree/main/moxiedrive_24_power_board%20V1), but hierarchical schematics work with [this project](https://github.com/vedderb/bldc-hardware/tree/master) which is from a much older version of Kicad.

leoheck, 2024-03-13:
Hey @ahalekelly thanks for the sample projects.

leoheck, 2024-04-04:
@ahalekelly,

The project [Moxie-Drive-TO220-x24/moxiedrive_24_power_board](https://github.com/Teslafly/Moxie-Drive-TO220-x24/tree/main/moxiedrive_24_power_board%20V1) works for me unless I am failing to see something. 

These are the schematic files I could find inside this project.
[image]

Here is Kiri generated page
[image]

=========

The project [bldc-hardware](https://github.com/vedderb/bldc-hardware/tree/master) uses with Kicad 4. Kiri supports Kicad >= 5 by default. However, I could use this project to backport it. But right now it is not kind of supported.

leoheck, 2024-04-04:
Ok even better, considering those 2 repositories.

Could you provide the output of the following commands so I can check if this part is working for you as it is working for me?

For this project [Moxie-Drive-TO220-x24/moxiedrive_24_power_board](https://github.com/Teslafly/Moxie-Drive-TO220-x24/tree/main/moxiedrive_24_power_board%20V1), that uses Kicad 7.

```bash 
Moxie-Drive-TO220-x24/moxiedrive_24_power_board V1 on  main [?] …
➜ find_kicad6_sch_hier 24_power_board.kicad_sch 
24_power_board|24_power_board.kicad_sch||24_power_board|24_power_board
Power phase|Power phase.kicad_sch||Phase A|24_power_board-Phase A
shunt_isense|shunt_isense.kicad_sch||Isense_Backup_shunts|24_power_board-Phase A-Isense_Backup_shunts
50v_to_12v_1A|50v_to_12v_1A.kicad_sch||50v to12v@1A|24_power_board-50v to12v@1A
inrush_lim|inrush_lim.kicad_sch||Inrush Limiter|24_power_board-Inrush Limiter
Power phase|Power phase.kicad_sch||Phase B|24_power_board-Phase B
shunt_isense|shunt_isense.kicad_sch||Isense_Backup_shunts|24_power_board-Phase B-Isense_Backup_shunts
Power phase|Power phase.kicad_sch||Phase C|24_power_board-Phase C
shunt_isense|shunt_isense.kicad_sch||Isense_Backup_shunts|24_power_board-Phase C-Isense_Backup_shunts
```

===================

For this project [bldc-hardware](https://github.com/vedderb/bldc-hardware/tree/master), that uses Kicad 4,
this is the script that parses the hierarchy, could you run it, please?

```bash 
bldc-hardware/design on  master [?] …
➜ find_kicad5_sch_hier BLDC_4.sch              
BLDC_4|BLDC_4.sch|-|BLDC_4|BLDC_4
CAN|CAN.sch|-|2F|BLDC_4-2F
temp|temp.sch|-|5E|BLDC_4-5E
mosfets|mosfets.sch|-|8E|BLDC_4-8E
hall_filters|hall_filters.sch|-|C6|BLDC_4-C6
Power|Power.sch|-|D1|BLDC_4-D1
```

leoheck, 2024-04-05:
Hey guys, @MartinMechatronics @ahalekelly 

I believe that I could fix this, at least for the given sample projects. It was tested on a Mac with Kicad 7.
Since I am not using this Mac for a while I have to upgrade it to the lastest OS (Ventura or something) and also the Kicad 8 to make sure the fixes it still work. 

Please, could you also check if the latest pushed commits are working for you too?

leoheck, 2024-04-05:
The `kiri -v` may also be fixed, please, test this for me too.

leoheck, 2024-04-05:
Alright, I could test it again. It now works with `kidiff` and also with `kicad-cli` (using `-k` flag) 
I am going to close this issue. 
Feel free to reopen it if the problem persists. 
Let me know it if works too, cheers!

closed 2024-04-05 by leoheck